esphome/feature-requests #2790: midea_ac.follow_me to send temperature updates in Fahrenheit instead of celsius

dbaq, 2024-06-29:
**Describe the problem you have/What new integration you would like**

The climate component for Midea works splendidly. I've also added an IR transmitter to use the "follow me" feature (described here https://esphome.io/components/climate/midea#additional-control-options-using-ir-commands).

Everything works well but when the temperature sensor has a new value and `midea_ac.follow_me` is being called, the display on the minisplit unit switches to celsius. Using the remote that comes with the minisplit reverts the display back to fahrenheit but it will go back to celsius again the next time the temperature is updated from esphome (i.e. 2min later). 

I am hoping `midea_ac.follow_me` could accept a third parameter (`use_fahrenheit:true`) to send the temperature via IR in fahrenheit. 

Here is my config currently, note how I have to convert the farenheit temp sensor to celsius as well.

```
  - platform: homeassistant
    name: Room Temperature Sensor
    entity_id: sensor.primary_bedroom_sensor_temperature   # Sensor from HASS
    internal: true
    filters:
      - throttle: 10s
      - heartbeat: 2min             # Maximum interval between updates.
      - debounce: 1s
      - lambda: return (x - 32.0) * (5.0/9.0);
    on_value:
      midea_ac.follow_me:
        temperature: !lambda "return x;"
        beeper: false    
```

**Please describe your use case for this integration and alternatives you've tried:**

- People in the USA are not used to Celsius 🙃 
- I tried to use the minisplit remote to force an override but it does not stick.

**Additional context**

Other people are encountering the same issue:
- https://community.home-assistant.io/t/midea-branded-ac-s-with-esphome-no-cloud/265236/1082
- https://community.home-assistant.io/t/midea-branded-ac-s-with-esphome-no-cloud/265236/1102

Thank you!

DjordjeMandic, 2024-11-13:
Can you do pronto capture of follow me signal from remote when display is set in F?

dbaq, 2024-11-13:
I am not familiar with that but happy to give it a try. Can you point me to any kind of docs by any chance?

DjordjeMandic, 2024-11-13:
@dbaq Could you test [#7762 pull request](https://github.com/esphome/esphome/pull/7762) ?

dbaq, 2024-11-14:
@DjordjeMandic you rock. I will give it a try.

dbaq, 2024-11-14:
works wonderfully!!! https://github.com/esphome/esphome/pull/7762#issuecomment-2477188926

closed 2024-12-18 by jesserockz