esphome/feature-requests #2792: Add ESP32-H2 management

Aratist, 2024-06-30:
**Describe the problem you have/What new integration you would like**

Hi, 
I would like that ESP-Home manage natively ESP32-H2 device ans use Zigbee with it (use ESP32 H2 with Wifi, BT & Zigbee)
**Please describe your use case for this integration and alternatives you've tried:**

Curently, we have to use a workaround to use ESP32-H2 and it not friendly user and it is difficult for noob
**Additional context**

I love your job , I'm an embedeed software developer, I want to dev with you as contributor)

matthew73210, 2024-07-09:
Unless i'm mistaken, the H2 can't do wifi

lboue, 2024-07-11:
@Aratist 
You can check this page: [ESP Product Selector](https://products.espressif.com/#/product-selector?language=en&names=)

[image]

WebSpider, 2024-08-07:
H2 does not do WiFi, it always needs an additional chip (ESP-C, for instance) to do WiFi.

It can do BT, so provisioning via improv and possibly communications to HA over BT are possible, but do require work.

randybb, 2024-08-07:
An official dev board has been just released https://www.cnx-software.com/2024/08/07/esp32-p4-function-ev-board-development-board-launched-for-55-with-7-inch-display-and-camera-module/
docs are here: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/user_guide.html

WebSpider, 2024-08-07:
That's the P4, not the H2.

H2 can be found here: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32h2/esp32-h2-devkitm-1/index.html

Aratist, 2025-01-31:
Hi,
Thank you, for your help, I'll try it

Hedda, 2025-04-13:
Duplicate of existing feature request, see -> https://github.com/esphome/feature-requests/issues/1397