esphome/feature-requests #2801: Add support for second CAN-Bus on ESP32-C6

Adminius, 2024-07-09:
**Describe the problem you have/What new integration you would like**
ESP32-C6 has hardware for two CAN-busses. ESPHome is not cabable to use it

**Please describe your use case for this integration and alternatives you've tried:**
Any dual CAN-bus application. SPI-based CAN transceiver could be an alternative but requeires more wirerig and special hardware.

**Additional context**
for second CAN-Bus **_twai_receive_v2_** has to be used:
https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32c6/api-reference/peripherals/twai.html#install-multiple-twai-instances

Adminius, 2024-07-09:
Some examples for the new API:
https://github.com/MagnusThome/RejsaCAN-ESP32/blob/main/Code%20Examples/RejsaCAN%20v6.x%20-%20ESP32-C6%20-%20dual%20CAN%20-%20Self%20tester/main/main.c

taoloo897, 2024-09-16:
I tried to compile this (attached file dual_TWAI) but a lot of errors(errors.txt) in arduino ide 2.3.2  with esp32 3.0.4 version.
[dual_TWAI.txt](https://github.com/user-attachments/files/17011871/dual_TWAI.txt)
[errors.txt](https://github.com/user-attachments/files/17011888/errors.txt)

Adminius, 2024-09-16:
Arduino IDE is still on on the old ESP IDF. 5.2+ ist necessary.
I'm using this:
https://github.com/esphome/esphome/pull/7135
with this:
https://github.com/Adminius/mitpylon/blob/main/mitpylon.yaml

every day without problems

taoloo897, 2024-09-16:
Thanks for answer but i'm a little confuse. You don;t use arduino ide if i understood well? Do you use visual studio code with esphome? Thanks.

Adminius, 2024-09-16:
I'm usind HomeAssistant Add-On for ESPHome