esphome/feature-requests #2803: Add support ST77916 RGB QSPI display

xiasi0, 2024-07-13:
[4_LCD.pdf](https://github.com/user-attachments/files/16201878/4_LCD.pdf)
Schematic diagram and demo

bitbank2, 2024-07-18:
I added support for this display to my bb_spi_lcd library. What functions does ESPHome need from the display library? Just pushing pixels? text? boxes?

xiasi0, 2024-07-19:
[url](https://community.home-assistant.io/t/guition-4-480x480-esp32-s3-4848s040-smart-display-with-lvgl/729271)
This is equipment from the same brand
[url](https://esphome.io/components/display/qspi_amoled.html)
ESPHome should need to initialize this display.
ESPHome currently only supports qspi-amoled. qspi-lcd is not yet supported.

xiasi0, 2024-07-19:
> [4848S040](https://community.home-assistant.io/t/guition-4-480x480-esp32-s3-4848s040-smart-display-with-lvgl/729271) This is equipment from the same brand [Quad SPI AMOLED Displays](https://esphome.io/components/display/qspi_amoled.html) ESPHome should need to initialize this display. ESPHome currently only supports qspi-amoled. qspi-lcd is not yet supported.

xiasi0, 2024-07-19:
> I added support for this display to my bb_spi_lcd library. What functions does ESPHome need from the display library? Just pushing pixels? text? boxes?

[[4848S040](https://community.home-assistant.io/t/guition-4-480x480-esp32-s3-4848s040-smart-display-with-lvgl/729271)](https://community.home-assistant.io/t/guition-4-480x480-esp32-s3-4848s040-smart-display-with-lvgl/729271)
This is equipment from the same brand
[Quad SPI AMOLED Displays](https://esphome.io/components/display/qspi_amoled.html)
ESPHome should need to initialize this display.
ESPHome currently only supports qspi-amoled. qspi-lcd is not yet supported.

bitbank2, 2024-07-19:
QSPI amoled and QSPI LCD are identical interfaces. The only difference will be the initialization sequence. I support both types in my bb_spi_lcd library.

xiasi0, 2024-07-20:
> QSPI amoled and QSPI LCD are identical interfaces. The only difference will be the initialization sequence. I support both types in my bb_spi_lcd library.

Yes, currently ESPHome can initialize two display models, RM67162 or RM690B0。 I don't know how to reference your display library in ESPHome. Hope you can provide tips and assistance

xiasi0, 2024-07-20:
> QSPI amoled and QSPI LCD are identical interfaces. The only difference will be the initialization sequence. I support both types in my bb_spi_lcd library.

```
display:
  - platform: qspi_amoled
    model: RM690B0
    data_rate: 80MHz
    spi_mode: mode0
    dimensions:
      width: 360
      height: 360
      offset_width: 16
    color_order: rgb
    invert_colors: false
    brightness: 255
    cs_pin: 10
    reset_pin: 47
    update_interval: never
    auto_clear_enabled: false
    id: my_display
```
I referenced the wrong display model, resulting in a complete display error

bitbank2, 2024-07-20:
I don't use ESPHome and don't have spare time to work on it. I mentioned my code because it's an Arduino library that supports a ton of LCD displays in a much simpler way compared to libraries like TFT_eSPI. If there is funding available for this effort, I can provide more assistance.

xiasi0, 2024-07-21:
> I'm making a simpler version of my library for LVGL and ESP-IDF. Perhaps that will be easier to adapt to ESPHome? Here's my AnimatedGIF library + my bb_spi_lcd driving the JC3636W518:
> 
> https://youtube.com/shorts/D-XRrlbDBuU?feature=share

ESPHome will be even better thanks to your contribution. Thank you

agillis, 2024-07-23:
It does look like the current drive support quad SPI

https://esphome.io/components/spi.html

Now we just need a working config

xiasi0, 2024-07-24:
> ST77916

Yes, looking forward to support

xiasi0, 2024-07-24:
> It does look like the current driver supports quad SPI
> 
> https://esphome.io/components/spi.html
> 
> Now we just need ST77916 support

Yes, looking forward to support

5orwt9, 2024-09-24:
come here to share your opinions

@agillis @bitbank2 @xiasi0 @glmnet @fabaff 

https://github.com/Bodmer/TFT_eSPI/discussions/3477#discussioncomment-10739750

clydebarrow, 2024-10-30:
Should be easy enough to support now with the `qspi_dbi` driver and a custom init sequence. Not released yet, but available in the dev branch. Docs here:

https://deploy-preview-4335--esphome.netlify.app/components/display/qspi_dbi.html

clydebarrow, 2024-10-30:
config here: 

https://github.com/clowrey/esphome-guition-360x360-jc3636w518/blob/main/guition-360x360-jc3636w518-lvgl.yaml

closed 2024-10-30 by clydebarrow