esphome/feature-requests #2804: jsn_sr04t component: RCWL-1655 compatibility

fraxinas, 2024-07-14:
**Describe the problem you have/What new integration you would like**
I bought a JSN_SR04T sensor module from ebay, thinking it would work with the existing implementation in esphome. Turned out this is a slightly different flavor of controller which didn't work with the existing sensor component.

**Please describe your use case for this integration and alternatives you've tried:**
As it turns out the module is a "RCWL-1655", which requires a different request format and delivers the measurement in 24 instead of 16 bits and without a checksum
https://makerhero.com/img/files/download/RCWL-1655-Datasheet.pdf

**Additional context**
I've implemented support for this flavor of the distance sensor here: 
https://github.com/esphome/esphome/compare/dev...fraxinas:esphome:jsn_sr04t_RCWL-1655

jaredcat, 2024-09-16:
@fraxinas I also bought a sensor and its RCWL-1655. I'm new to esphome and trying to get this to work. Could you provide me an example to get this working? Even after importing the changes as a custom component, I can't seem get any readings.

Also, Any chance you could address the comments in the PR?

TrevorSquillario, 2024-09-18:
This is the code I used to get my RCWL-1655 working. You will have to modify the last 2 calculations for your container height (in meters). I used the `ultrasonic` platform since the default mode of the RCWL-1655 will work like the legacy ultrasonic sensor. I remember having to use a static IP as well, couldn't get DHCP working.  This is a workaround until the `jsn_sr04t` module is updated to support the RCWL-1655. Using the `ultrasonic` module is less power efficient from my understanding but it works! I'm using this to monitor the water level of a 55 gal barrel. 

RCWL-1655 Pinout (White connector facing towards you, left to right)
```
5V
RX Trig
TX Echo
GND
```

ESP8266 Wiring
```
Connect the GND from the ESP8266 to the GND pin of the module
Connect the Vin pin from the ESP8266 to the 5V pin
Connect the D5 (GPIO14) pin from the ESP8266 to the TX (Echo) pin on the module
Connect the D6 (GPIO12) pin from the ESP8266 to the RX (Trig) pin on the module
```

```
esphome:
  name: reservoir-55gal
  friendly_name: reservoir_55gal

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.70
    gateway: 192.168.0.1
    subnet: 255.255.255.0

# Source: https://markus-haack.com/watertank-esphome/

sensor:
  # Wifi signal sensor.
  - platform: wifi_signal
    id: wifi_signal_percent
    name: Wifi Signal
    update_interval: 600s
    unit_of_measurement: '%'
    filters:
    - lambda: |-
        if (x <= -100) {
          return 0;
        } else {
          if (x >= -50) {
            return 100;
          } else {
            return 2 * (x + 100);
          }
        }

  # Templates for calculated liter & percent
  - platform: template
    name: Gallon
    id: gallon
    icon: 'mdi:water'
    unit_of_measurement: 'gal'
    accuracy_decimals: 1

  - platform: template
    name: Percent
    id: percent
    icon: 'mdi:water-percent'
    unit_of_measurement: '%'

  # The actual distance sensor
  - platform: ultrasonic
    trigger_pin: GPIO12
    echo_pin: GPIO14
    name: Distance
    id: distance
    update_interval: 60s
    pulse_time: 50us
    filters:
    - filter_out: nan
    - median:
        window_size: 7
        send_every: 4
        send_first_at: 3
    #- calibrate_linear: # Map values
    #   - 0.23 -> 1.86 # 
    #   - 2.41 -> 0.0 # Distance reading from fixed sensor position to empty tank
    on_value:
      then:
        # Empty volume in liters
        # V = π * radius² * height
        - sensor.template.publish:
            id: gallon
            state: !lambda |-
              auto m = (3.141592653589 * pow(0.28, 2) * .85) * 1000;
              auto v = (3.141592653589 * pow(0.28, 2) * x) * 1000;
              return (m - v) * 0.2641720524;

        # p = π * radius² * height * 1000 / max_litre * 100
        - sensor.template.publish:
            id: percent
            state: !lambda |-
              auto m = (3.141592653589 * pow(0.28, 2) * .85) * 1000;
              auto v = (3.141592653589 * pow(0.28, 2) * x) * 1000;
              auto p = (m - v) / 208.198 * 100;
              return p;
```