esphome/feature-requests #2815: Add the ability to monitor the condition of the air conditioner. Incorrect operation after power off IR climate control remote control

TokarevSergey, 2024-07-19:
Good afternoon.
Add the ability to monitor the condition of the air conditioner. Incorrect operation after power off IR climate control remote control.
I have an air conditioner that works with IR Remote Climate integration
climate:
  - platform: whirlpool
    name: "Кондиционер спальня"
    model: DG11J1-91
    id: "konditsioner_spalnya"
    sensor: current_temperature
    receiver_id: "s08_receiver"
    transmitter_id: "s08_transmitter"
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C
If you turn off the electricity during operation, then after turning on the electricity, the air conditioner will go into off mode, and the integration will remain in the on position. Please add the ability to monitor the condition of the air conditioner. Thank you.

randybb, 2024-07-19:
How you want to monitor it via IR? It is just one way and you basically expects that the device has the same state. Only thing that you can do is periodically send all commands so your AC will match settings.

nagyrobi, 2024-07-19:
Add a power sensor to monitor the presence of ekectricity. Set an atomation so when it's gone, turn the climate off in ESPHome. It will send the power off IR commands but that wouldn't matter, as the device itself has the power cut off.

Btw: it's a bad practice to often cut off electricity of an air conditioner while in operation. It will be damaged. It needs proper shutdown with power off from IR, and wait a few minutes for the compressor to cool down before cutting electricity.

TokarevSergey, 2024-07-22:
> Добавьте датчик мощности для контроля наличия электричества. Настройте атомацию так, чтобы при ее отсутствии выключать климат в ESPHome. Он будет отправлять ИК-команды на отключение питания, но это не будет иметь значения, так как само устройство отключится от питания.
> 
> Кстати: часто отключать электричество от работающего кондиционера — плохая практика. Он будет поврежден. Его нужно правильно выключить, отключив питание от ИК, и подождать несколько минут, чтобы компрессор остыл, прежде чем отключать электричество.

Я понимаю что не правильно но сложилась так ситуация.

TokarevSergey, 2024-07-22:
> Добавьте датчик мощности для контроля потребления электричества. Настройте атомацию так, чтобы при ее отсутствии выключать климат в ESPHome. Он отправит ИК-команды на отключение питания, но это не будет иметь значения, так как само устройство отключится от питания.
> 
> Кстати: часто отключать электричество от работающего кондиционера — плохая практика. Он будет поврежден. Его правильно нужно выключить, отключив питание от ИК, и потребовалось несколько минут, чтобы компрессор остыл, прежде чем отключить электричество.

Perhaps run wiring from the esp house's GPIO to power the 5V air conditioner and hang a small battery on the esp house, allowing it to work for the first time. How can I register that if the signal is lost at the GPIO pin, then send a signal to turn it on?