esphome/feature-requests #2816: Link changelog to releases

serl, 2024-07-19:
**Describe the problem you have/What new integration you would like**

I'd like to have a direct link to the changelog in the ESPHome website (for example [2024.7.0](https://esphome.io/changelog/2024.7.0.html)), in order to review the possible necessary upgrading manual steps before upgrading the addon from Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**

It took me some time to find that there's a changelog page in the ESPHome website, which often contains guidelines on upgrading and it's way more readable than the full list of changes (which is actually included in that page as well).
This is very important information and I think it should be more accessible to the users.

**Additional context**

Actually, I see that the changelog page of the website is composed of a more textual part, followed by the release description in Github (and the addon takes that same release description).
So I'd say that in a perfect world, all the information should be in the release description in Github, so it would be more or less automatically slapped everywhere for maximum reach. But adding a link on the top of the addon's changelog to `https://esphome.io/changelog/XXXX.Y.Z.html` would be already a huge improvement for me.

randybb, 2024-07-19:
basically duplicate of https://github.com/esphome/feature-requests/issues/2623

serl, 2024-07-19:
> basically duplicate of #2623

I mean it's about spreading the changelog as well, but not really a duplicate IMO. #2623 is about having a link to the changelog in the dashboard. This is about having a link to the extra text which appears before the changelog in the release info

serl, 2024-07-19:
Adding an extra chip, since we're here... I see that for Core, there's simply a link ("Read release announcement") pointing `https://www.home-assistant.io/latest-release-notes/`:
[image]

Would be possible to have the same for ESPHome?
[image]

serl, 2025-06-22:
Closing, as 2025.6.0 had already all of this