esphome/feature-requests #2826: Remko RKL490 to RKL495, TROTEC, Novamatic, Rexair, Fischer, freecom etc. AC remote IR code - Need help

sorryusernameisalreadytaken, 2024-08-01:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I only borrowed the remote control from a friend. I still have it for 2 days. I would be very grateful for any tips on IR code dumps.

I can't believe that a replacement remote control would cost €70!

nagyrobi, 2024-08-01:
Please join the Discord support server for faster response

sorryusernameisalreadytaken, 2024-08-01:
Closed! I am having a flu and it was really hard to work on that issue.... Puh.

Discord. really, why is everybody using a closed sourced community.

nagyrobi, 2024-08-01:
I'm sorry but the situation is that it's very unlikely that you'll get any answer here within the 2 days timeframe your friend imposed to you. Discord is mostly instant if you are polite with the community and give details of your question in a thread.
I want the best for you, believe me.

sorryusernameisalreadytaken, 2024-08-01:
Thanks. Yeah I know, not the best env.

But why clsoing this issue. In the https://esphome.io/components/climate/climate_ir.html is written:
"There is a growing list of compatible units. If your unit is not listed below you should submit a feature request (see FAQ)."

sorryusernameisalreadytaken, 2024-08-02:
I guess this should help me a lot:
https://github.com/triluch/esphome-climate-ir-fral

typhoe, 2025-07-05:
@sorryusernameisalreadytaken Hi, did you manage to add the remote to esphome ?
I also have an AR-715 (used with a REMKO RKL491DC)
Using esphome in `dump: all` mode, I only see codes in `pronto` format so I could use these codes to send only the commands I need but if there exists proper support, it would be better.
Thank you