esphome/feature-requests #2831: Add support for GC9D01N screen to ili9xxx

styphonthal, 2024-08-02:
**Describe the problem you have/What new integration you would like**

Add the GC9D01N screen to the ili9xxx esphome component. 

**Please describe your use case for this integration and alternatives you've tried:**

I’ve tried Model CUSTOM, st7789v and GC9A01A without any success.
This screen is used on the lilygo touch bar, 0.99 inch screen, and 0.75 inch round display among others

**Additional context**

Lilygo github for screen is here: https://github.com/Xinyuan-LilyGO/T-0.99TFT/blob/main/src/TFT_GC9D01N.h
There is also a datasheet/manual from the manufacturer here: https://github.com/Xinyuan-LilyGO/T-0.99TFT/blob/main/GC9D01N%20DataSheet%20V1.1.pdf

styphonthal, 2024-09-08:
The circuitpython example above has an init sequence, but I don't know how to change it to be used in esphome

makerwolf, 2025-03-16:
Has anyone ever gotten this display to work with esphome?

styphonthal, 2025-03-20:
> Has anyone ever gotten this display to work with esphome?

https://community.home-assistant.io/t/does-gc9d01n-lcd-screen-work-in-esphome/754282/11

closed 2025-03-20 by styphonthal