esphome/feature-requests #2835: Add sniff option to modbus component

peterv99, 2024-08-06:
**Describe the problem you have/What new integration you would like**
Heatpumps from Pumo-Radson/Mirai/Emmetti use a modbus based controller called [Febos Crono](https://emmeti.com/en/products/air-conditioning/residential-inverter-air-to-water-heat-pumps/air-to-water-heat-pump-mirai-smi-febos-4-0/febos-crono). The Febos Crono is a glorified room thermostat, optionally with wifi to connect to a closed vendor cloud environment that hosts a webapp. The Febos uses a 2-wire RS485 RTU modbus to talk to the heat pump and to an optional power integration module ([Febos Energy](https://emmeti.com/en/products/air-conditioning/residential-inverter-air-to-water-heat-pumps/air-to-water-heat-pump-mirai-smi-febos-4-0/febos-energy))

The Crono is the modbus client, it completely controls the heat pump settings such as schedule, temperatures, when to heat sanitary hot water etc. Both the heat pump and the optional energy monitor serve as servers. Modbus addresses are fixed, the heat pump has address 3 and the energy monitor has address 2, and the Febos Crono (client) has address 1. 

**Please describe your use case for this integration and alternatives you've tried:**
I want to monitor and log the performance of the heat pump by sniffing the modbus communications. The current client already loops through all relevant registers and issues reads. Modbus RTU does not support multiple clients, and I want to monitor the modbus read requests and corresponding responses without interfering with communications. The current modbus component does not support sniffing the bus, it expects to either be a server or a client.

I like to be able to integrate the modbus sniffer in esphome/home assistant, so it would be easy for others to use/implement. I could of course sniff the bus (have a working rpi with a level shifter that monitors the modbus now and could publish e.g. to MQTT) but that feels like an unnecessary hack given the integration potential with esphome.

**Additional context**

Custom component that does something similar but does not use the modbus component: https://github.com/htvekov/solivia_esphome

cleontin, 2025-05-15:
I have a modified version of the modbus component that can parse and log all messages seen. I can share it for one-time use if still needed.

I'm looking at how this can be integrated with the current version of modbus without breaking compatibility with other components that use it. Any help with this topic would be greatly appreciated.

peterv99, 2025-05-15:
A copy of the modified version would be helpful. Willing to help integrating though I am not a real coder.

peterv99, 2025-11-12:
Yep, cool, will look into it!

closed 2025-11-12 by peterv99