esphome/feature-requests #2836: support for power_supply on cww light

gerasimone, 2024-08-06:
**Describe the problem you have/What new integration you would like**
CWWW light use custom powerful psu, but the option was not accpted for light.cwww

**Please describe your use case for this integration and alternatives you've tried:**
tody we need to use on_turn_on and on_turn_off, but with multiple conditions if more than a led strip is used

**Additional context**

randybb, 2024-08-06:
It is not under light, but under the physical output as every "color" can be controlled from a different psu.

In my config it is like this:
```yaml
light:
  - platform: cwww
    name: "${name} Light 1"
    id: light_1
    default_transition_length: 200ms
    restore_mode: ALWAYS_ON
    cold_white: pwm_cw_1
    warm_white: pwm_ww_1
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K

output:
  - platform: pca9685
    pca9685_id: 'pca9685_hub1'
    id: 'pwm_cw_1'
    channel: 0
    zero_means_zero: true
    power_supply: power_supply1
  - platform: pca9685
    pca9685_id: 'pca9685_hub1'
    id: 'pwm_ww_1'
    channel: 1
    zero_means_zero: true
    power_supply: power_supply1
```

gerasimone, 2024-08-06:
Thank you @randybb , it works