esphome/feature-requests #2837: RP240 Platform - Raspberry Pi Pico 2 with RP2350 microcontroller support

Hedda, 2024-08-09:
**Describe the problem you have/What new integration you would like**

Please consider adding platform support for Raspberry Pi Pico 2 with RP2350 MCU support, (RP2350 is successor/next-generation of RPi RP2040):

https://www.raspberrypi.com/news/raspberry-pi-pico-2-our-new-5-microcontroller-board-on-sale-now/

https://www.raspberrypi.com/products/raspberry-pi-pico-2/

https://github.com/raspberrypi/pico-sdk-rp2350

https://github.com/raspberrypi/pico-examples-rp2350

https://www.cnx-software.com/2024/08/08/raspberry-pi-pico-2-raspberry-pi-rp2350-dual-core-risc-v-or-arm-cortex-m33-microcontroller/

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Hedda, 2024-08-09:
For reference, here is a related feature request to the upstream Raspberry Pi development platform for [PlatformIO](https://platformio.org/) that is needed:

* https://github.com/maxgerhardt/platform-raspberrypi/issues/69

Hedda, 2024-11-26:
> The W version is now released https://www.raspberrypi.com/news/raspberry-pi-pico-2-w-on-sale-now/

Nice specifications, and those specs with the same Wi-Fi and Bluetooth chip will of couse also soon be used to create RP2350 clones and other boards based on RP2350:

- https://www.cnx-software.com/2024/11/25/7-raspberry-pi-pico-2-w-board-2-4-ghz-wifi-4-bluetooth-5-2-wireless-module/

danielkent-net, 2024-12-05:
I got some Pico 2Ws today and I started looking at the esphome code a bit. I didn't get very far but I found a few things that might be helpful for anyone else that wants to try advancing this at all:
- `arduino_pico` version 4.3 adds support for the 2W, and the 4.3.1 release fixes a few bugs. There needs to be a corresponding version bump in version in a few places.
- Similarly, git commit 39b9039 on maxgerhardt's PlatformIO fork is needed for compatibility with the above version of `arduino_pico`.

These changes aren't the only things needed to build RP2350 compatible binaries; with the changes I made (https://github.com/danielkent-net/esphome/commit/6370f11c0f0c7a2dcc0f7e320ee6ff4d0b88ec0e) I think it's still outputting an RP2040 uf2. It does compile _something_, for what it's worth. Granted, I'm only testing with a very minimal example that just blinks the LED.

I also don't know if anyone's considered whether RP2350 support should live in its own platform, or be part of the `rp2040` platform (and subsequently keep `rp2040_pwm` and other related platforms as-is). My fork keeps it in the same repository since I saw a lot of code that would otherwise need to be duplicated in the new platform (or put in some sort of common platform that both reference). I'm not too familiar with this part of esphome code, so if anyone has a better idea and wants to chime in that would be helpful.

Hedda, 2024-12-05:
> I also don't know if anyone's considered whether RP2350 support should live in its own platform, or be part of the `rp2040` platform (and subsequently keep `rp2040_pwm` and other related platforms as-is).

Might also simlpler to add new platforms if this new pull request is merged:

* https://github.com/esphome/esphome/pull/7715

danielkent-net, 2024-12-07:
> Might also be simpler to add new platforms like this in the future if all these new pull requests by @tomaszduda23 are merged?

I'm not sure it's strictly necessary, but it might help down the line with further Pico releases.

---

Doing some more experimentation, my changes do in fact produce RP2350-compatible binaries, but GPIOs don't seem to work. The output looks like this:
```
[I][app:100]: ESPHome version 2024.12.0-dev compiled on Dec  7 2024, 13:45:29
[C][logger:185]: Logger:
[C][logger:186]:   Level: DEBUG
[C][logger:188]:   Log Baud Rate: 115200
[C][logger:189]:   Hardware UART: USB_CDC
[C][gpio.output:010]: GPIO Binary Output:
[C][gpio.output:011]:   Pin: GPIO32
```

If I compile with `VERY_VERBOSE` debug, I see that it looks like it's trying to toggle the GPIO:

```
[I][app:100]: ESPHome version 2024.12.0-dev compiled on Dec  7 2024, 14:04:46
[C][logger:185]: Logger:
[C][logger:186]:   Level: VERY_VERBOSE
[C][logger:188]:   Log Baud Rate: 115200
[C][logger:189]:   Hardware UART: USB_CDC
[C][gpio.output:010]: GPIO Binary Output:
[C][gpio.output:011]:   Pin: GPIO32
[VV][scheduler:225]: Running timeout '' with interval=250 last_execution=1966 (now=2216)
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=1962 (now=2462)
[VV][scheduler:032]: set_timeout(name='', timeout=250)
[VV][scheduler:225]: Running timeout '' with interval=250 last_execution=2462 (now=2712)
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=2462 (now=2963)
[VV][scheduler:032]: set_timeout(name='', timeout=250)
[VV][scheduler:225]: Running timeout '' with interval=250 last_execution=2963 (now=3213)
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=2962 (now=3465)
[VV][scheduler:032]: set_timeout(name='', timeout=250)
[VV][scheduler:225]: Running timeout '' with interval=250 last_execution=3466 (now=3716)
```

The same behavior is also exhibited when compiling for RP2040 (switching board to pipicow in the example yaml), so this issue isn't specific to the RP2350.

I know the LED pin on the Pico W/2W is tied to the wireless chipset and isn't a true GPIO; my next attempt might be to toggle a different GPIO and check the output on an oscilloscope, in case that's part of the problem.

JanPetterMG, 2024-12-16:
Wasn't able to lit the internal LED me either, but connecting an external one via GPIO and getting it blinking was hassle free. I've also got Wi-Fi (with OTA), SPI and the internal temperature sensor working.

Haven't experimented too much yet, but it was quite easy to get started, just put this in the YAML:
````
rp2040:
  board: rpipico2w
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git#develop
    version: dev
````

Hedda, 2025-03-18:
No news on the ESPHome side, but FYI, Raspberry Pi has just announced general availability for the RP2350A and RP2350B in bulk, budles, or single pieces:

- https://www.raspberrypi.com/news/rp2350-now-available-to-buy-a-high-performance-secure-microcontroller-for-your-next-project/

Before this the RP2350 microcontroller unit itself has been very hard to source for hobbyist and low-volume projects that just want the MCU and not a complete development board.

- https://www.cnx-software.com/2025/03/18/buy-raspberry-pi-rp2350-mcu-rp2354a-and-rp2354b-variants/?amp=1

PS: They have also pre-announced new RP2354A and RP2354B microcontrollers variants with 2MB of stacked flash memory on top of the microcontroller chip to allow more compact designs.

Hedda, 2025-03-18:
> Pico-SDK 2.0.0 supports this new RP2350 microcontroller chip:
> 
> https://github.com/raspberrypi/pico-sdk/releases
> 
> For reference, here are a few related feature requests to upstream dependencies that are needed to be implement first before they can be used by ESPHome for all Raspberry Pi Picos:
> 
> * [[REQUEST] Raspberry Pi RP2350 and Raspberry Pi Pico 2 support? platformio/platform-raspberrypi#67](https://github.com/platformio/platform-raspberrypi/issues/67)
> * [[REQUEST] Raspberry Pi RP2350 and Raspberry Pi Pico 2 support? maxgerhardt/platform-raspberrypi#69](https://github.com/maxgerhardt/platform-raspberrypi/issues/69)
> * [RP2350 Support earlephilhower/arduino-pico#2325](https://github.com/earlephilhower/arduino-pico/issues/2325)
>   
>   * [New RP microcontroller coming earlephilhower/arduino-pico#2226](https://github.com/earlephilhower/arduino-pico/discussions/2226)
> 
> FYI, [PontusO](https://github.com/PontusO) has forked [earlephilhower](https://github.com/earlephilhower)’s arduino-pico repo to a new arduino-pico-rp2350 repo where he got RP2340 working with an older version of the pico-sdk:
> 
> https://github.com/PontusO/arduino-pico-rp2350
> 
> > _An arduino-pico port is on it's way (https://github.com/PontusO/arduino-pico-rp2350). Unfortunatly there were a ton of PR's on the pico-sdk the past few days so it is not ready for prime time yet. **It works fairly well in its current state though but this is using a 2 month old version of the pico-sdk**._
> 
> _Originally posted by [PontusO](https://github.com/PontusO) in [earlephilhower/arduino-pico#2226 (comment)](https://github.com/earlephilhower/arduino-pico/discussions/2226#discussioncomment-10278437)_
> 
> We do not know yet if he will leave that new arduino-pico-rp2350 repo as a a separate fork dedicated only to the RP2350 or if he or others will submit pull requests with changes from it to earlephilhower’s arduino-pico repo and use that as upstream for all picos, including both RP2040 and RP2350 in the same repository.
> 
> https://github.com/PontusO) in [earlephilhower/arduino-pico#2226](https://github.com/earlephilhower/arduino-pico/discussions/2226)

Be aware that while you will probably never get a official answer from PlatformIO, you should know that PlatformIO's CEO have more or less made it clear that they will not allow their upstream platform-raspberrypi to keep being developed in their repository/repositories as long as Raspberry Pi Foundation or Raspberry Pi Ltd is not officially paying PlatformIO to maintain it there, and infer that the decision to discontinue support for upstream PlatformIO platforms are entirly up to the companies paying PlatformIO to maintain each platform.

The alternative options for now is to either use a fork or use a Raspberry Pi Pico Arduino core without PlatformIO. New forks of PlatformIO's platform libraries have therefore been created and maintained separate due to the lack of ongoing development in PlatformIO upstream platform library. Anyone in the community can use and contribute to those, see example:

* https://github.com/maxgerhardt/platform-raspberrypi

FYI, it looks like we can see practically the same situation between PlatformIO and Espressif (platform-espressif32 for ESP32) which this has gone on for longer:

* https://github.com/platformio/platform-espressif32/issues/1225

PlatformIO is also no longer taking community contributions to improve its platform-raspberrypi platform library in their upstream repository/repositories, which means that upstream development of PlatformIO's platform library for Raspberry Pi Pico(s) have stalled or put on hold for now. Because of this users feel forced to move to forks or alternative Arduino-Pico core without PlatformIO in order to use the new RP2350 MCU/SoC. 

* https://github.com/maxgerhardt/platform-raspberrypi/issues/73

* https://github.com/earlephilhower/arduino-pico
  * https://arduino-pico.readthedocs.io/en/latest/

 For additional information, please refer to these GitHub links where the discussions there are self-explanatory, allowing you to draw your own conclusions:

* https://github.com/espressif/arduino-esp32/discussions/10039
* https://github.com/platformio/platform-espressif32/issues/1225
* https://github.com/espressif/arduino-esp32/pull/8606

PS: I understand PlatformIO is a company that need income but I do not see how this practice will benefit them in the long run.

Strat00s, 2025-03-21:
TLDR: it's always about money...

JanPetterMG, 2025-06-21:
- ADC is now functional (easy [workaround](https://github.com/esphome/issues/issues/6781#issuecomment-2993305848)).
- 2025.5.2 seemingly have introduced a typo (still present in the latest dev from 4hr ago), but 2025.5.1 still builds fine.

Other than the on-board LED, official support and of course money, is there anything that's not yet supported on the 2/2W ?

wroadd, 2026-01-17:
additiaonal trend information:
Lot of industrial relay boards which have an ESP32 version, now have RPi2350 version, too and they are much more cheaper then ESP32 versions.
Also wireless thermostats are using RPi2350 (Thermogea, Alledio, etc).