esphome/feature-requests #2842: USB UAC Support

netham45, 2024-08-11:
**Describe the problem you have/What new integration you would like**

I would like support for USB UAC audio devices on ESP32S2/ESP32S3.

**Please describe your use case for this integration and alternatives you've tried:**

esphome has no feature for USB audio as far as I know, however it could improve the number of quality audio output options available along with their ease of use. I am using a standalone project for it at the moment. I'm using an esp32s3 with a usb dongle dac for my own esp-idf project and it's been working well.

**Additional context**

https://github.com/esp-arduino-libs/ESP32_USB_Stream/ - Espressif Arduino library
https://components.espressif.com/components/espressif/usb_host_uac/versions/1.0.0 - esp-idf library

nagyrobi, 2024-08-12:
_Note: For physical device with both microphone and speaker, the driver will treat it as two separate logic devices._

schildbach, 2026-09-12:
Just a small heads up: the non-ESPHome project [Airplay-ESP32](https://github.com/rbouteiller/airplay-esp32) has recently [implemented USB UAC for ESP32-S3](https://rbouteiller.github.io/airplay-esp32/features/usb-audio/).

schildbach, 2026-10-04:
Also worth mentioning: the [DSPi project](https://github.com/WeebLabs/DSPi) can do UAC on RP2040/2350. It's USB only though, as there is no support for any kind of network.

Hedda, 2026-10-05:
> worth mentioning: the DSPi project can do UAC on RP2040/RP2350. It's USB only though, as there is no support for any kind of network.
> 
> * https://github.com/WeebLabs/DSPi

Combining Raspberry Pi Pico with DSPi firmware as USB sound card with a ESP32 running ESPHome with USB UAC support sounds awesome!

* https://www.audiosciencereview.com/forum/index.php?threads/introducing-dspi-a-powerful-user-friendly-and-open-source-dsp-for-less-than-a-cup-of-coffee.69343/

DSPi firmware project also has optional support to use PCM1808 ADC module on Pico 2 (I2S Input) for analogue line-in level input as well:

  * https://github.com/WeebLabs/DSPi

closed 2026-10-19 by bdraco