esphome/feature-requests #2846: IR dump output format

Miq1, 2024-08-14:
**Describe the problem you have/What new integration you would like**

The dumped output from ``remote_receiver`` always needs editing before being usable in another YAML:
```
[15:41:49][I][remote.pronto:234]: Received Pronto: data=
[15:41:49][I][remote.pronto:236]: 0000 006D 001F 0000 0068 0021 0012 0010 0012 0010 0012 0021 0012 0021 0034 0021 0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0023 0021 0023 0021 0012 0010 0023 0021 0023 0021 0012 0010 0023 0021 0012 0010 
[15:41:49][I][remote.pronto:236]: 0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0023 0010 0012 0021 0012 0010 0012 0602 06C3
[15:41:49][I][remote.drayton:230]: Received Drayton: address=0x7FAD (0xff5a), channel=0x01f command=0x037
```
Would be nice if it came nicely formatted like:
```
[15:41:49][I][remote.pronto:234]: Received Pronto: data=
- remote_transmitter.transmit_pronto:
      data: "0000 006D 001F 0000 0068 0021 0012 0010 0012 0010 0012 0021 0012 0021 0034 0021
             0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0023 0021
             0023 0021 0012 0010 0023 0021 0023 0021 0012 0010 0023 0021 0012 0010 0012 0010
             0012 0010 0012 0010 0012 0010 0012 0010 0012 0010 0023 0010 0012 0021 0012 0010
             0012 0602"
[15:41:49][I][remote.drayton:230]: Received Drayton: address=0x7FAD (0xff5a), channel=0x01f command=0x037
```
(or something similar that is C&P-able.)

**Please describe your use case for this integration and alternatives you've tried:**

Collecting all IR sequences for a remote is a time-consuming task anyway, so not having to edit every single block of data would be an improvement.

**Additional context**

juanboro, 2025-05-11:
I'm not sure if it directly helps --- but I created a custom GIRS component for esphome [here](https://github.com/juanboro/esphome/tree/mymaster/juanboro/girs) that makes it pretty easy to collect and manage remote IR sequences using IRscritinizer.  From there vscode/copilot can be helpful at converting the exported data back to yaml.

ssieb, 2025-05-11:
Log lines are limited length and can't be formatted like that anyway.

closed 2025-05-11 by ssieb