esphome/feature-requests #2847: Support for ws2814 32 bit Addressable LED (bk72xx) (Beken SPI LED Strip)

postal302, 2024-08-14:
**Describe the problem you have/What new integration you would like**
Add support for for WS2814 Addressable LEDs to the Beken SPI LED Strip platform [https://esphome.io/components/light/beken_spi_led_strip.html](url) for use with Beken bk72xx chips.

**Please describe your use case for this integration and alternatives you've tried:**
New smart addressable LEDs are shipping with beken chips. I have a set of Ligency outdoor spotlights [https://a.co/d/io0m6lA](url) that use a CBU WiFI module (bk7231n), but there is virtually no support for beken chips and addressable LEDs with 32 bit data packets. I've set the Beken SPI LED Strip platform for ws2812 and it lights the spotlights and changes colors, but the colors are random and not all lights work. None of the other built in LED chipsets work for this either.

Here's the datasheet for the ws2814 [https://www.superlightingled.com/PDF/SPEC/WS2814-RGBW-Datasheet.pdf](url)

**Additional context**
There are other features/issues that use the ws2811 settings to successfully control the ws2814 (see here: [https://github.com/esphome/feature-requests/issues/2049](url)) but that only works for ESP chips and not Beken chips. I've also installed OpenBeken on the CBU and it does not yet support ws2814. [https://www.elektroda.com/rtvforum/topic4028972.html](url)

postal302, 2024-08-15:
It is understood that the WS2814 is a variant of the SK6812 with a W channel: [https://wled.discourse.group/t/are-ws2814-24v-rgbw-led-strips-compatible-with-wled/5601/28](url)

A workaround for using the Beken_SPI_LED_Strip platform is to use the sk6812 chipset and set "is_wrgb" to true. I also had to set "rgb_order" to RGB because the default GRB was incorrect. After this the lights are showing the correct color on all pixels and have white as a color option. Finally, using "- platform: partition" allows individual control of each addressable pixel.

Here's my yaml for my 4-spotlight setup:
```
esphome:
  name: ligency-uplight-group-1
  friendly_name: Ligency_Uplight_Group_1

bk72xx:
  board: cbu

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "SuperSecretKey"

ota:
  - platform: esphome
    password: "SuperSecretPassword"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ligency-Uplight-Group-1"
    password: "SuperSecretPassword"

# Light Configuration
light:
  - platform: beken_spi_led_strip
    rgb_order: RGB
    pin: P16
    num_leds: 4
    chipset: SK6812
    is_wrgb: true
    id: Uplight1

  - platform: partition
    name: "Uplight Pixel 1"
    segments: 
      - id: Uplight1
        from: 0
        to: 0

  - platform: partition
    name: "Uplight Pixel 2"
    segments: 
      - id: Uplight1
        from: 1
        to: 1 

  - platform: partition
    name: "Uplight Pixel 3"
    segments: 
      - id: Uplight1
        from: 2
        to: 2

  - platform: partition
    name: "Uplight Pixel 4"
    segments: 
      - id: Uplight1
        from: 3
        to: 3

captive_portal:

web_server:
  port: 80
```

disruptivepatternmaterial, 2024-11-10:
I am not sure if this is the same thing, but I have this light

https://www.amazon.com/dp/B0DDKV8HZJ?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

I was able to UART flash them, but they appear to be RGB with 2 white channels. My setup has them powering on and white mode works sometimes, but the rest fails.

I cannot tell what chip they are but here are some photos of the LEDs taken out of the bulbs.

[image]
[image]

In a chat on discord with the cloud cutter folks, seems this dual white channel is not supported.  I have been trying to work on the esphome code for this to add a channel, but haven't been able to as of yet.