esphome/feature-requests #2854: JSN-SR04T to add support for AJ-SR04M

tgrosz, 2024-08-17:
**Describe the problem you have/What new integration you would like**

The AJ-SR04M differs only slightly from JSN-SR04T. Mode 1 is the same, it's working, I tested it (it needs 120k resistor on R19).
Mode 2 differs only from AJ-SR04M that it needs a 0x01 byte instead of 0x55 to start measurement (and it needs 47k resistor, so the two are swapped).
I'd like to suggest a new parameter 'start_byte' defaulting to 0x55, but changeable to 0x01 - and so the component will handle the clone as well.

**Please describe your use case for this integration and alternatives you've tried:**

Ultrasonic measurement

**Additional context**

The manual for AJ-SR04M
https://manuals.plus/arduino/aj-sr04m-distance-measuring-transducer-sensor-manual

dumboxp, 2024-09-19:
I have the same problem getting AJ-SR04M to work with ESPHome on my ESP32C6. I would welcome a solution.

becmar39, 2024-11-28:
Try to use Ultrasonic component with open R19 Mode (no resistor)

rsipka, 2024-11-28:
It would be great if we could either override the start_byte or if it was set by the model setting

edouardreg, 2025-01-08:
Same here : start_byte option will be helpfull for me too...

kabongsteve, 2025-10-15:
This support seems to have been added, but it actually fails and makes it worse.

I have raised a separate bug_report for these issues