esphome/feature-requests #2864: Grundfos Alpha 3 on/off switch

larsvinc, 2024-08-27:
**Describe the problem you have/What new integration you would like**
The Grundfos Alpha 3 has a great integration with lots of sensors:
https://esphome.io/components/sensor/alpha3.html

I would like to be able to turn the pump on/off or change the mode. 

**Please describe your use case for this integration and alternatives you've tried:**
The use case is when all the circuits behind the pump are closed. I use the pump for heating in the floors of my house. When all the thermostats sense sufficiently high temp, all the circuits / valves close. Then the pump is working against a blocked system. It is made to handle that, but according to my plummer it is better to turn it off. Ideally I would also like to be able to adjust the throttle (lifting hight) etc., but I can only wish for so much.

I would be more than happy to contribute debugging, ble sniffing to find commands etc. But I need someone knowledgeable to work with on this.

I guess the alternative that I have been considering is to have a zigbee switch etc. and kill the mains to the pump when all circuits are closed. Then I loose the BLE connection, but this could be an option. 

**Additional context**

PvdGulik, 2024-10-28:
Hello @larsvinc, did you manage to find a solution? I also would be intrestesd to change all sort of settings like in Go Remote. Primary in HomeAssistant or EspHome.

Thanks!

larsvinc, 2024-10-29:
@PvdGulik no solution yet. Like you I would want to set settings via Home Assistant.

cryptontgn, 2026-01-30:
It will be a nice feature, i need it also