esphome/feature-requests #2869: Add support for lv_chart

sprior, 2024-08-31:
**Describe the problem you have/What new integration you would like**
I'd like to see support for the LVGL chart widget added to ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
This would seem useful to display data with some history as a chart, like temperature.  With some of the larger rectangular displays this seems like a reasonable use case.

**Additional context**
This is the LVGL widget:
https://docs.lvgl.io/master/widgets/chart.html

Gelisob, 2024-09-28:
yeah, was surprised it's not supported, is it new or something?

pascal-parent, 2024-10-17:
This would be a great addition

sprior, 2024-10-17:
Unless it's limited to the historical values of an entity I don't see how the values would be provided for the chart.

agillis, 2024-10-17:
The chart data could be pulled from home assistant or any other source. It would work exactly like the Graph component does now.

https://esphome.io/components/graph.html

sprior, 2024-10-17:
Ok true, so if it's basically the same idea just with LVGL hopefully it won't be too horrible to implement.

agillis, 2024-11-27:
Your right it would be nice to have it integrated properly into the LVGL part of ESPHome.

numo68, 2025-01-08:
anyone working on this? I might be inclined to try as a way to get familiar with lvgl (no promises nor ETA though). I see some defines (`LV_CHART_TYPES`) but not more.

hwinkel, 2025-01-10:
Yes, LV_CHARTS seems to be one of the last LVGL widgets missing (apart LV_TABLE)

nagyrobi, 2025-01-10:
It's on the todo list.

dimonb, 2025-03-09:
Hi, guys, it's not a solution, but temporarily you can use lv_chart directly
```yaml
        - obj:
            id: weather_chart_obj
            styles: [layout_box]
            grid_cell_x_align: STRETCH
            grid_cell_y_align: STRETCH
            grid_cell_row_pos: 1
            grid_cell_column_pos: 0
            widgets:
              - label:
                  text: !lambda '{draw_weather_chart(weather_chart_obj); return "";}'
```

in chart.h
```cpp
static void draw_weather_chart(lv_obj_t *obj) {
  if(chart == nullptr) {
    chart = lv_chart_create(obj);
    lv_obj_add_style(chart, weather_chart, 0);
    lv_obj_set_style_size(chart, 0, LV_PART_INDICATOR);
    //lv_obj_set_style_bg_color(chart, lv_color_make(50, 50, 50), LV_PART_MAIN); // Dark grey background
    lv_chart_set_point_count(chart, 24);
    //lv_obj_set_size(chart, 230, 150);
    //lv_obj_center(chart);
    lv_chart_set_type(chart, LV_CHART_TYPE_LINE);   /*Show lines and points too*/

    lv_obj_set_style_line_color(chart, lv_color_make(70, 70, 70), LV_PART_TICKS); // Less contrast grid lines
    lv_obj_set_style_line_color(chart, lv_color_make(70, 70, 70), LV_PART_MAIN | LV_STATE_DEFAULT);

    ser1 = lv_chart_add_series(chart, lv_palette_main(LV_PALETTE_RED), LV_CHART_AXIS_SECONDARY_Y);
LV_CHART_AXIS_SECONDARY_Y);
    lv_obj_add_event_cb(chart, draw_event_cb, LV_EVENT_DRAW_PART_BEGIN, NULL);
    lv_obj_update_layout(chart);
    for(int i = 0; i < 13; i++) {
      weather_icons[i] = lv_label_create(obj);
      lv_obj_add_style(weather_icons[i], weather_small_icon, 0);
      lv_obj_update_layout(weather_icons[i]);
      lv_obj_set_x(weather_icons[i], lv_obj_get_x(chart) + lv_obj_get_x(weather_icons[i]) + i * lv_obj_get_width(chart) / 13);
    }
  }
```

```yaml
esphome:
  platformio_options:
    build_flags:
       - -D LV_USE_CHART=1
  includes:
    - helpers.h
```

sprior, 2025-03-09:
@dimonb Thanks for the info, I'm trying to understand it.  Where do I need to put chart.h - in the same directory as my YAML or do I need to add it to the esphome source code?  Also is there a typo on the line "LV_CHART_AXIS_SECONDARY_Y);"?

victorclaessen, 2025-05-27:
@dimonb Thanks for posting your workaround. I'm interested to use it but I have not been able to figure out your code. I'm also confused by the name/location of `chart.h`, because you seem to include a `helpers.h` file instead?

dimonb, 2025-05-27:
> In your example; where does the data come from? I see a reference to an [undeclared in this code?] object `weather_chart`, is that it?

it's id of style, described in 
```yaml
lgvl:
  style_definitions:
    - id: weather_chart
      bg_color: black
      bg_opa: TRANSP
      border_width: 0
      width: 245
      height: 119
      x: 20
      y: 15
    
```

dimonb, 2025-05-27:
> How do you fill it with data? For example to achieve something like what's in [LVGL doc's 1000 points example](https://docs.lvgl.io/8.4/widgets/extra/chart.html#display-1000-data-points-with-zooming-and-scrolling)? Something equivalent to what ESPHome [already supports](https://esphome.io/components/graph) on the low level renderer.

I fill it with weather data which I got from http://api.open-meteo.com/v1/forecast

victorclaessen, 2025-05-27:
Would you be willing to share your complete code? For me it's not clear how you supply the data to the graph (I understand now where the weather comes from, but not how it 'travels' into the graph.)

dimonb, 2025-05-27:
> Would you be willing to share your complete code? For me it's not clear how you supply the data to the graph (I understand now where the weather comes from, but not how it 'travels' into the graph.)

You can find it in my repo: https://github.com/dimonb/esp32-box/tree/lvgl

JimnyVR5, 2025-07-16:
@dimonb 
> Hi, guys, it's not a solution, but temporarily you can use lv_chart directly

I got that to work - so, many thanks!
My non-existent coding skills were only enough for the static line chart from the examples on https://docs.lvgl.io/master/details/widgets/chart.html though :D

Support of charts with the ease of use of ESPHome's graph component would be very appreciated.

pastukhov, 2026-02-14:
I made PR  with chart support https://github.com/esphome/esphome/pull/13987

RubenNL, 2026-02-14:
Did you just write this in 1.5 hours?