esphome/feature-requests #2870: Would be possible to add the Senseair Sunrise CO2 sensor?

Jpm100P2, 2024-09-01:
Dear community, is it possible to add the support for [Senseair Sunrise CO2 sensor](https://senseair.com/product/sunrise/)?

The company changed the protocol so the current existing S8 component can not be used.

Thanks.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

For those that doesn't know it it's a very accurate CO2 sensor that is normally installed in professional CO2 meters.

Sizurka, 2024-10-03:
While it's not that elegant, these sensors speak standard Modbus when they're in UART mode (COMSEL high or floating).  So you can interface with them using ESPHome's generic Modbus support.  Below is part of the configuration I use for the one I have.  If you just want CO2 out of it with all defaults, you can cut off everything below the CO2 sensor, likewise there's lots I didn't implement; check Senseair's Modbus documentation.

```yaml
uart:
  baud_rate: 9600
  tx_pin: GPIO04
  rx_pin: GPIO03

modbus_controller:
  - id: sunrise_sensor
    address: 0x68

sensor:
  - platform: modbus_controller
    name: "CO2 Concentration"
    device_class: carbon_dioxide
    state_class: measurement
    unit_of_measurement: ppm
    accuracy_decimals: 0
    modbus_controller_id: sunrise_sensor
    register_type: read
    address: 0x03
    value_type: S_WORD
#  - platform: modbus_controller
#    id: co2_sensor_temperature
#    device_class: temperature
#    state_class: measurement
#    unit_of_measurement: "°C"
#    accuracy_decimals: 1
#    modbus_controller_id: sunrise_sensor
#    register_type: read
#    address: 0x04
#    value_type: S_WORD
#    filters:
#      - multiply: 0.01

number:
  - platform: modbus_controller
    id: co2_sensor_pressure
    modbus_controller_id: sunrise_sensor
    address: 0x2E
    register_type: holding
    use_write_multiple: true
    value_type: S_WORD
    min_value: 300
    max_value: 1300
    multiply: 10

  - platform: modbus_controller
    name: "Ambient CO2 Baseline"
    device_class: carbon_dioxide
    entity_category: config
    mode: box
    unit_of_measurement: ppm
    modbus_controller_id: sunrise_sensor
    address: 0x0F
    register_type: holding
    use_write_multiple: true
    value_type: S_WORD
    min_value: 400
    max_value: 16000

switch:
  - platform: modbus_controller
    name: "CO2 Pressure Compensation"
    entity_category: config
    disabled_by_default: true
    modbus_controller_id: sunrise_sensor
    register_type: holding
    use_write_multiple: true
[... 171 lines cut ...]
    on_release:
      then:
        lambda: 'id(sunrise_scale_factor_error)->status_clear_warning();'

#text_sensor:
#  - platform: modbus_controller
#    name: "CO2 Sensor Firmware Version"
#    internal: true
#    id: sunrise_firmware_version
#    entity_category: diagnostic
#    modbus_controller_id: sunrise_sensor
#    register_type: read
#    address: 0x1C
#    raw_encode: COMMA
#  - platform: modbus_controller
#    name: "CO2 Sensor Serial Number"
#    internal: true
#    id: sunrise_serial_number
#    entity_category: diagnostic
#    modbus_controller_id: sunrise_sensor
#    register_type: read
#    address: 0x1D
#    register_count: 2
#    raw_encode: HEXBYTES
```

Jpm100P2, 2024-10-25:
Thanks a lot for your suggestion. 
Ok it works but I would like to use single measurement mode including previous measurement data injection before new read. Is it possible? ( I'm expert in C language but newbie on lambda coding).