esphome/feature-requests #2872: Support Hi-Link LD2410S (HLK-LD2410S) 24GHz Ultra Low Power Radar Sensor

besanur, 2024-09-03:
**Describe the problem you have/What new integration you would like**
- Similar to the [LD2410](https://esphome.io/components/sensor/ld2410.html) Module, the [LD2410S](https://www.hlktech.net/index.php?id=1176) is a Ultra low Power 24GHz Radar Sensor. But it works with baudrate 115200 which is the smaller problem. Since it uses a different protocol, it does not work with the current integration.

**Please describe your use case for this integration and alternatives you've tried:**
- Radar sensor with very low power consumption (~0.1 mA)

**Additional context**
- User Manual: (Communication protocol is in chapter 5) https://drive.google.com/drive/folders/1wC8KC-DaNavNbpeVouZ1HdiBzZ9YrAcg
-  Device: https://www.hlktech.net/index.php?id=1176

dafunkydan, 2024-10-03:
While lacking a Sensor (haven't bought it yet due to not beeing supported in ESPhome), this external Component looks very promising!
https://github.com/MrUndead1996/ld2410s-esphome

With my next order i will get some, having a native Support for such a promising Sensor (3,3V, 0.1mA Consumption) would be awesome! 💯

besanur, 2024-10-04:
> While lacking a Sensor (haven't bought it yet due to not beeing supported in ESPhome), this external Component looks very promising! https://github.com/MrUndead1996/ld2410s-esphome
>
> With my next [...]

Thanks for your comment, I'm glad that more and more people are agreeing to this feature request. I've tested this custom component and it's worked well so far. By the way, I still need to understand the settings `Distance reporting frequency` and `Status reporting frequency`. I hope that it will soon be included in ESPHome.

[image]

dafunkydan, 2024-10-05:
Great to hear it is in a somewhat productive state useable!

> By the way, I still need to understand the settings `Distance reporting frequency` and `Status reporting frequency`

In the Docs to the Device the following can be found:

> Status reporting frequency: Frequency at which the sensor reports a manned/unoccupied state (0.5~ 8 Hz)
> Range reporting frequency: The frequency at which the sensor reports the distance to the target when it's manned (0.5~ 8 Hz)

I assume the Trigger Threshold controls the sensitivity - in your Screenshot i can neither see the reported Energy, nor a manual Threshold Setting. Is it only possible to perform an "Auto Threshold"? 

Looking forward to get the thing into my hands finally 😃

sorrypqa, 2024-11-16:
> While lacking a Sensor (haven't bought it yet due to not beeing supported in ESPhome), this external Component looks very promising! https://github.com/MrUndead1996/ld2410s-esphome
>
> With my next [...]

Thank you for sharing! Buy a few now and try them out.

dafunkydan, 2024-11-16:
> Thank you for sharing! Buy a few now and try them out.

You're welcome! Make sure to like the initial Post, so the Interest is visible 😊

vump, 2025-01-03:
> While lacking a Sensor (haven't bought it yet due to not beeing supported in ESPhome), this external Component looks very promising! https://github.com/MrUndead1996/ld2410s-esphome
>
> With my next [...]

I can verify that this works! Just setup this sensor with a Wemos D1 Mini!

maxmaia, 2025-02-20:
Used the gihutb code from https://github.com/MrUndead1996/ld2410s-esphome but it looks that with ESPhome 2025.2.0 it does not work. It somehow blocks even webserver and cannot access it. Also Esphome comsole does not connect to it, despite that the device is somehow pingable.

Would be great if such nice/low consumption sensor will get integrated directly into Esphome.
Thanks.

LittlePapaJohn, 2025-02-21:
I would agree to have the LD2410S integrated in ESPHome

silversegarra, 2025-02-24:
> Thanks for your comment, I'm glad that more and more people are agreeing to this feature request. I've tested this custom component and it's worked well so far. By the way, I still need to [...]

Hello, I have installed by ZHA, but I dont have all these options for configuration... Could yo help me...? thanks

k-korn, 2025-04-13:
Hey @NovakIrs, I'm using your code from [PR #8486](https://github.com/esphome/esphome/pull/8486) and managed to get a graph in HA similar to what the native app produces based on your triggers_hold, triggers_snr, triggers_threshold debug output with ApexChart card ([preview](https://i.imgur.com/Tt7vIaR.png)):

```
type: custom:apexcharts-card
graph_span: 16h
update_delay: 500ms
span:
  start: day
header:
  show: true
  title: LD2140S Energy Values
all_series_config:
  stroke_width: 2
  type: line
yaxis:
  - decimals: 0
    min: 0
    max: "|+10|"
series:
  - entity: sensor.sw_18ad30_triggers_hold
    name: Hold
    data_generator: >
      return entity.state.split(',').map((v, i) => [new Date().setHours(i) + 1,
      +v])
  - entity: sensor.sw_18ad30_triggers_snr
    name: SnR
    data_generator: >
      return entity.state.split(',').map((v, i) => [new Date().setHours(i) + 1,
      +v])
  - entity: sensor.sw_18ad30_triggers_threshold
    name: Threshold
    data_generator: >
      return entity.state.split(',').map((v, i) => [new Date().setHours(i) + 1,
      +v])
```

It uses data_generator feature to map gate indexes from 1 to 16 to hours 1..16, allowing the chart to display the values.
This might be useful to visualize the thresholds.

closed 2026-10-12 by bdraco