esphome/feature-requests #2873: Add LVGL Multi-Display Per-Page Support

ConnorVG, 2024-09-05:
**Describe the problem you have/What new integration you would like**
We currently have an integration for LVGL that allows us to use the vast library to render components to our screens (such as meters to GCA901 circular displays). Many displays utilize CS which allows us to have multiple displays on one device (awesome and already implemented in ESPHome). To my knowledge, which I have tried to research as much as possible, it is not currently possible to set a render target on a `lvgl.page.show` and also no way to have multiple `lvgl` base components in a yaml. With these two being true there is currently no way to have multiple displays on one ESP device with each display rendering a different LVGL page.

**Please describe your use case for this integration and alternatives you've tried:**
Dashboard displays and meter panels for nice measurement visibility of sensors. More than one screen allows for not having to use an ESP for each of the sensors that would want to be displayed in some fixed real life panel.

**Additional context**
I found this a limiting factor whilst designing a 3-display dashboard to fit under my monitor on my desk in the style of a "instrument cluster" such as the ones found in vehicles.

I've attached a reference image as an example of what I mean:
[image]

ConnorVG, 2024-10-09:
@Greg-79 you can add multiple but they will currently show the same thing if using lvgl:
Each display needs defining in the display config section: https://esphome.io/components/display/#display-rendering-engine
LVGL config needs to have displays defined: https://esphome.io/components/lvgl/#main-configuration

Greg-79, 2024-10-10:
Hi! On own esphome engine, I can connect both 4 and 5 displays by writing code under the lambda.  And how to register a second display in LVGL?

clydebarrow, 2024-11-02:
> Where can I point to the second display?

Under the `displays:` key. 🤔 

```
lvgl:
  displays:
    - display_id: my_display
    - display_id: my_other_display
```

clydebarrow, 2024-11-03:
@ConnorVG If you could try this out for your application that would be helpful.

closed 2024-11-07 by jesserockz

philpownall, 2024-12-12:
Oh, I should add to the above that I am able to use LVGL on two screens if I don't use pages.  But, pages are nice...

LVGL Definition:
```
lvgl:
  - id: lvgl_1
    displays: display_1
    buffer_size: 25%
    bg_color: black
    widgets:
      - slider:
          x: 4
          y: 26
          width: 110
          height: 8
          outline_width: 2
          outline_color: white
          outline_pad: 4
          id: slider1_id
          value: 20
          bg_color: black
          min_value: 10
          max_value: 30
  - id: lvgl_2
    displays: display_2
    buffer_size: 25%
    bg_color: black
    widgets:
      - slider:
          x: 6
          y: 28
          width: 110
          height: 8
          outline_width: 2
          outline_color: white
          outline_pad: 4
          id: slider2_id
          value: 0
          bg_color: black
          min_value: -30
          max_value: 30

```

htht2001, 2025-01-03:
It's exactly the same for me.
I can set a different display for each screen, but it only works when I assign pages to just one display.
If I try to assign pages to the second display as well, the displays get mixed up.

pgtest, 2025-04-21:
Annoying issue. Pages only work if setup on one display. If used on two or more... for me all pages are shown on display 1 despite usage of lvgl_id: when calling lvgl.page.next (even if definitions of displayed widgets are on another lvgl_id). Pages currently seems to be limited to first display with pages, rest ignored and drawn on first display.