esphome/feature-requests #2874: Add support for the DFRobot SEN0390 lux sensor

mstuij, 2024-09-08:
**Describe the problem you have/What new integration you would like**
Please add support for the DFRobot SEN0390 lux sensor: 
https://wiki.dfrobot.com/Ambient_Light_Sensor_0_200klx_SKU_SEN0390

**Please describe your use case for this integration and alternatives you've tried:**
To meassure the lux outside.

**Additional context**

csurulyal, 2024-10-19:
Would be nice.

mstuij, 2024-10-27:
I have a working workaround for now:

```
esphome:
  name: achterkant-buiten-controller
  friendly_name: achterkant-buiten-controller
  includes:
    - /config/esphome/lib/DFRobot_B_LUX_V30B/my_custom_sensor.h
  libraries:
    - dfrobot/DFRobot_B_LUX_V30B@^1.0.1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Achterkant-Buiten-Controller"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:
    
i2c:
  sda: GPIO32
  scl: GPIO33
  scan: True
  
sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor();
    App.register_component(my_sensor);
    return {my_sensor};
  sensors:
    name: "Achterkant buiten lux"
    device_class: illuminance
    state_class: measurement
    unit_of_measurement: lx
    accuracy_decimals: 1
- platform: wifi_signal
  name: "Achterkant buiten controller WiFi signal"
  update_interval: 60s
- platform: uptime
  name: "Achterkant buiten controller uptime"          

```

Then you download the DFRobot_B_Lux_V30B and place it in the folder (/config/esphome/lib/DFRobot_B_LUX_V30B/): 

https://github.com/DFRobot/DFRobot_B_LUX_V30B

And then you need one extra file (my_custom_sensor.h):

```
#include "esphome.h"
#include <DFRobot_B_LUX_V30B.h>

class MyCustomSensor : public PollingComponent, public Sensor {
 public:
  // constructor
  MyCustomSensor() : PollingComponent(5000) {} 
  
  DFRobot_B_LUX_V30B    shedLux{13, 33, 32};//The sensor chip is set to 13 pins, SCL and SDA adopt default configuration

  void setup() override {
    shedLux.begin();
  }

  void update() override {
    float outsideLux = shedLux.lightStrengthLux();
    publish_state(outsideLux);
  }
};
```

mstuij, 2025-02-22:
With ESPHome 2025.2.0 the soluction doesn't work anymore. Custom components are now deleted and you have to use external components.

With this new code it works again:

```
esphome:
  name: achterkant-buiten-controller
  friendly_name: achterkant-buiten-controller
  libraries:
    - dfrobot/DFRobot_B_LUX_V30B@^1.0.1

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: components

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "............."

ota:
  - platform: esphome
    password: "..........................."

wifi:
  ssid: !secret IoT1_ssid
  password: !secret IoT1_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Achterkant-Buiten-Controller"
    password: "..................."

captive_portal:
    
i2c:
  sda: GPIO32
  scl: GPIO33
  
sensor:
- platform: dfrobot_sen0390
  name: "Achterkant buiten lux"
- platform: wifi_signal
  name: "Achterkant buiten controller WiFi signal"
  update_interval: 60s
- platform: uptime
  name: "Achterkant buiten controller uptime"          

```

In the ESPHome folder you need to unzip the attached files

[components.zip](https://github.com/user-attachments/files/18922052/components.zip)