esphome/feature-requests #2878: WIFIUdp.h for the esp-idf platform

bkbartk, 2024-09-10:
**Describe the problem you have/What new integration you would like**
I would like to have WIFIUdp.h support for the ESP-IDF framework

**Please describe your use case for this integration and alternatives you've tried:**
I use a component for synchronizing my light switches
https://github.com/Cossid/TasmotaDeviceGroupsForESPHome
this component syncs via UDP and therefor requires WIFIUdp.h 
untill now setteling for arduino worked, but now I want to add a bluetooth proxy and ble tracker to one of the switches.
I tried this, but I immediately found the device to become laggy. 

**Additional context**
almost all sync tools use this component,

randybb, 2024-09-10:
So somebody has to make this external component compatible with esp-idf? Should not be this addressed to its creator? :D

Anyway, maybe this will work for you - https://next.esphome.io/components/udp / https://github.com/esphome/esphome/pull/6865 will be available in 2024.9

nagyrobi, 2024-09-10:
Not only interval based: https://deploy-preview-3918--esphome.netlify.app/components/udp.html
Updates on state change instantly.

bkbartk, 2024-09-10:
I see I missed this line`When any of the configured sensors changes state, the component will broadcast that sensor’s state,`
So I think this could work for me, I will try this one after the next release