esphome/feature-requests #2885: LVGL - implement a Lottie animation player

kyvaith, 2024-09-18:
**Describe the problem you have/What new integration you would like**
Implement a Lottie format animations widget. Reference:
https://docs.lvgl.io/master/libs/rlottie.html

**Please describe your use case for this integration and alternatives you've tried:**
Enabling the use of high-performance animation will unlock the possibility of creating rich user interfaces.
Reference: https://www.youtube.com/watch?v=sYgrNL-UZwA
Reference 2: https://github.com/Samsung/rlottie

nagyrobi, 2024-09-18:
Read the disclaimer related to ESP: https://docs.lvgl.io/master/libs/rlottie.html#esp-idf-example

kyvaith, 2024-09-18:
Well, considering the end result, I think it's worth the effort.

_For all its faults, when running right-sized animations, they provide a wonderful utility to LVGL on embedded LCDs and can look really good when done properly._

In the attached video, we can see that the animations look very cool. I think this is a better way than rendering several images in a row.

There are also ways to optimize Lottie files, for better performance and size. https://lottiefiles.com/blog/working-with-lottie-animations/optimize-lottie-files-for-faster-page-load-speeds

Of course, if the mechanism is used incompetently, there will be performance problems. However, this is like with everything on SoCs such as ESP32.

Think of the animation played when imbedding voice commands in a voice assistant. Think of the animation played while playing music, or even a visual representation of the current weather. The possibilities that far Lottie are very large and will allow for a whole new quality of LVGL-based graphical interfajes.

nagyrobi, 2024-09-18:
Don't get me wrong, I'd love to see them, but I highly doubt the results will be as good as advertised there. I wish it could happen...

kyvaith, 2024-09-18:
Sure. May I ask how much work it would really take to add such functionality for testing? As I understand it, ESPhome is only concerned with packaging existing functionality. Lv_lottie is a ready-made widget, you just need to let it be used in ESPhome. Maybe it is worth a try? https://docs.lvgl.io/master/API/widgets/index.html

nagyrobi, 2024-09-18:
My pessimism is exactly because of the wrapper nesting. The lvgl widget is a wrapper around lottie, the esphome lvgl component is a wrapper around lvgl, arduino on ESP is a wrapper around (an old) IDF and so on. Each wrapping layer adds overhead to the core code, which consumes resources.

But for now, ESPHome supports LVGL 8.4. Afaik lottie is in LVGL just very recently, from v9.2 - plenty of refactoring needs to be done, also trying to do that without breaking changes...