esphome/feature-requests #2886: Davis vantage pro weather station info to esphome-homeassistant

mortenx, 2024-09-18:
**Describe the problem you have/What new integration you would like**

Davis vantage pro weather station info to esphome-homeassistant

**Please describe your use case for this integration and alternatives you've tried:**

could be something like that
https://github.com/dcbo/ISS-MQTT-Gateway

**Additional context**

mortenx, 2024-09-26:
i have davis https://www.raig.co.uk/davis-vantage-pro-2-wireless-professional-weather-station-pack-with-weatherlink-live-b13d5/
lora board with ra-01h   https://tienda.bricogeek.com/accesorios/1415-m5stack-lora-868-mhz.html
and SX1276
i can do some programming, could try something

Is this AcuRite code works with ecowitt sensors also?

swoboda1337, 2024-09-26:
AcuRite code will not work with ecowitt but its very similar. RTL_433 has working reference code. It would only take a few hours to get ecowitt working. 

First step is to setup the radio from my PR or my github: https://github.com/swoboda1337/sx127x-esphome

Set up the radio and dump some raw data, for davis you want to dump fsk. Make sure you use the correct pins, these are just examples:

```yaml
spi:
  clk_pin: GPIO36
  mosi_pin: GPIO37
  miso_pin: GPIO35

sx127x:
  nss_pin: GPIO6
  rst_pin: GPIO5
  frequency: 868000000 
  modulation: FSK
  rx_start: true
  rx_bandwidth: 50_0kHz

remote_receiver:
  pin: GPIO10
  dump: raw
```
  
For ecowitt you would have to use OOK, is it also 868 or is it 433, I think they sell both:

```yaml
spi:
  clk_pin: GPIO36
  mosi_pin: GPIO37
  miso_pin: GPIO35

sx127x:
  nss_pin: GPIO6
  rst_pin: GPIO5
  frequency: 868000000 
  modulation: OOK
  rx_start: true
  rx_bandwidth: 50_0kHz
  rx_floor: -90

remote_receiver:
  pin: GPIO10
  dump: raw
```

Lets see if you can receive anything useful. Can try bandwidth of 25k Hz as well. Hopefully the center frequency is correct. I know for 433MHz I have to use 433920000. I have never tried 828 MHz.

swoboda1337, 2024-09-26:
Updated the filters, I think davis is a pretty high bitrate

mortenx, 2024-09-27:
how can i test that the code works?
I installed but nothing happens, no errors, no logs , no data 
I know that the pins and board are ok because the same board works well with another code
and I have several devices next to it, davis, ecowitt , wmbus ...all 868mhz
is it possible to show all the traffic that happens on 868mhz unfiltered?

swoboda1337, 2024-09-27:
Interesting. You can try with a filter of 0us and rx_floor of like -120. For OOK you should see something:

[07:24:53][I][remote.raw:028]: Received Raw: -600, 615, -600, 627, -601, 614, -601, 196, -418, 196, -417, 184, -418, 195, -418, 444, -170, 196, -405, 444, -170, 444, -170, 196, -418, 182, -418, 445, -169, 444, -170, 431, -183, 196, -405, 196, -418, 444, -170, 196, -405, 443, -170,

What does your yaml look like?

swoboda1337, 2024-09-27:
These are the pins you used?

```yaml
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

sx127x:
  nss_pin: GPIO5
  rst_pin: GPIO26
  frequency: 868000000 
  modulation: FSK
  rx_start: true
  rx_bandwidth: 50_0kHz

remote_receiver:
  pin: GPIO36
  filter: 5us
  idle: 5000us
  dump: raw
  ```

swoboda1337, 2024-09-27:
Ok I have that same exact board and my config is almost the same but I use GPIO32 for remote receiver not GPIO26.

mortenx, 2024-09-27:
.. I changed it to 32, now it works, thanks. Strange why https://www.lilygo.cc/products/lora3?variant=42272562282677 has pin 26

swoboda1337, 2024-09-27:
If you want to try to decode the data can do it in a lambda:

```yaml
remote_receiver:
  pin: GPIO32
  filter: 150us
  idle: 800us
  on_raw:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "Received raw data with length %d", x.size());
          for (auto i : x) {
            // do something
          }
```

mortenx, 2024-09-27:
`[14:59:52][I][remote.raw:028]: Received Raw: -52, 78, -39, 26, -27, 26, -79, 38, -14, 25, -14, 12, -27, 13, -13, 14, -65, 26, -65, 26, -13, 26, -14, 65, -12, 26, -14, 104, -14, 52, -39, 12, -14, 53, -51, 14, -12, 14, -52, 25, -53, 26, -14, 25, -26, 26, -14, 26, -25, 27, -13, 13, -26, 
[14:59:52][I][remote.raw:028]:   53, -38, 40, -13, 26, -91, 27, -78, 13, -14, 12, -14, 51, -40, 26, -13, 12, -14, 13, -26, 14, -12, 26, -14, 13, -78, 39, -39, 14, -38, 53, -65, 26, -26, 13, -52, 53, -12, 40, -13, 66, -13, 65, -13, 65, -27, 103, -40, 53, -51, 26, -27, 65, -26, 40, -12, 
[14:59:52][I][remote.raw:028]:   27, -26, 39, -13, 13, -14, 25, -13, 13, -14, 12, -14, 39, -13, 25, -14, 39, -14, 26, -51, 26, -14, 39, -12, 27, -52, 14, -12, 27, -52, 78, -13, 13, -26, 14, -12, 27, -13, 25, -14, 39, -40, 12, -27, 26, -12, 27, -26, 39, -26, 52, -79, 13, -13, 65, -27, 
[14:59:52][I][remote.raw:028]:   12, -26, 132, -12, 26, -13, 14, -38, 92, -52, 26, -53, 39, -26, 26, -13, 13, -13, 66, -13, 26, -26, 26, -65, 13, -14, 78, -25, 27, -13, 13, -14, 12, -14, 51, -26, 14, -26, 78, -52, 53, -12, 14, -39, 91, -40, 26, -39, 13, -52, 14, -12, 27, -26, 12, -14, 
[14:59:52][I][remote.raw:028]:   53, -91, 39, -78, 78, -14, 38, -14, 26, -13, 26, -13, 26, -14, 12, -26, 65, -14, 65, -13, 13, -65, 27, -38, 40, -39, 26, -13, 117, -14, 79, -12, 65, -27, 26, -26, 26, -92, 78, -27, 12, -14, 77, -27, 13, -13, 39, -52, 26, -14, 51, -67, 25, -26, 14, -26, 
[14:59:52][I][remote.raw:028]:   104, -14, 77, -14, 12, -40, 27, -38, 53, -12, 27, -13, 25, -14, 26, -13, 40, -12, 27, -12, 26, -40, 65, -13, 26, -26, 53, -52, 27, -25, 13, -13, 26, -79, 26, -39, 26, -53, 78, -13, 26, -26, 13, -66, 51, -27, 26, -26, 26, -13, 26, -14, 12, -53, 65, -40, 
[14:59:52][I][remote.raw:041]:   25, -14, 12, -14, 12, -27, 52, -13, 13, -39, 14, -51, 26, -14, 144, -26, 39, -65, 65, -26, 66, -13, 65, -14, 38, -27, 65, -13, 65, -27, 12, -53, 38, -53, 79, -12, 26, -27, 40, -12, 13, -40, 26, -39, 12, -14, 26, -53, 25, -26, 26
[14:59:52][W][component:237]: Component remote_receiver took a long time for an operation (176 ms).
[14:59:52][W][component:238]: Components should block for at most 30 ms.
[14:59:54][I][remote.raw:028]: Received Raw: -40, 12, -26, 27, -12, 40, -26, 13, -12, 79, -39, 26, -14, 39, -26, 26, -13, 13, -27, 12, -26, 14, -25, 14, -12, 27, -13, 27, -12, 14, -12, 27, -12, 26, -14, 13, -13, 52, -91, 26, -14, 39, -13, 27, -25, 52, -14, 39, -38, 40, -27, 65, -12, 
[14:59:54][I][remote.raw:028]:   40, -39, 39, -40, 38, -27, 78, -27, 65, -26, 26, -65, 13, -52, 53, -26, 12, -53, 79, -12, 26, -39, 14, -12, 40, -27, 12, -13, 66, -39, 91, -52, 13, -92, 13, -39, 13, -13, 40, -39, 26, -26, 26, -13, 14, -12, 39, -40, 117, -27, 12, -14, 25, -14, 103, 
[14:59:54][I][remote.raw:028]:   -14, 65, -53, 78, -27, 12, -13, 13, -13, 14, -26, 51, -26, 14, -13, 26, -26, 39, -14, 12, -14, 12, -14, 26, -13, 26, -13, 26, -14, 26, -25, 79, -40, 25, -26, 66, -13, 12, -40, 39, -26, 14, -39, 52, -39, 26, -13, 26, -26, 14, -25, 26, -79, 14, -12, 156, 
[14:59:54][I][remote.raw:028]:   -14, 26, -65, 53, -39, 26, -12, 14, -13, 117, -14, 65, -52, 65, -14, 25, -14, 39, -78, 13, -40, 39, -26, 12, -14, 65, -26, 27, -26, 12, -27, 26, -14, 25, -13, 39, -14, 12, -39, 26, -27, 26, -40, 12, -27, 91, -26, 79, -51, 40, -13, 52, -13, 13, -13, 26, 
[14:59:54][I][remote.raw:028]:   -53, 26, -12, 27, -52, 26, -39, 53, -38, 79, -53, 12, -14, 26, -13, 52, -13, 13, -13, 26, -26, 65, -14, 39, -14, 12, -39, 79, -65, 26, -52, 27, -38, 14, -13, 13, -39, 53, -12, 14, -39, 38, -14, 13, -26, 14, -38, 13, -14, 26, -26, 26, -13, 39, -13, 27, 
[14:59:54][I][remote.raw:028]:   -26, 39, -26, 26, -26, 27, -51, 66, -39, 91, -14, 65, -26, 51, -27, 13, -13, 26, -14, 25, -40, 65, -14, 25, -26, 26, -65, 53, -13, 39, -65, 53, -13, 39, -13, 13, -13, 52, -53, 39, -26, 14, -65, 12, -40, 39, -39, 26, -14, 12, -39, 14, -26, 12, -53, 26, 
[14:59:54][I][remote.raw:041]:   -13, 26, -14, 12, -26, 39, -14, 104, -40, 38, -27, 12, -14, 12, -14, 13, -13, 53, -51, 53, -38, 105, -39, 40, -26, 13, -14, 12, -13, 40, -12, 26, -14, 25, -14, 39, -13, 13, -14, 12, -53, 38, -53, 26, -13, 66, -12, 105, -13, 39, -26, 14
[14:59:54][W][component:237]: Component remote_receiver took a long time for an operation (175 ms).
[14:59:54][W][component:238]: Components should block for at most 30 ms.
[14:59:57][I][remote.raw:028]: Received Raw: -51, 14, -12, 14, -26, 26, -13, 12, -14, 13, -13, 39, -14, 12, -65, 65, -14, 13, -13, 13, -14, 26, -26, 25, -14, 26, -13, 13, -13, 39, -13, 13, -26, 26, -14, 39, -26, 13, -13, 13, -40, 12, -39, 14, -39, 65, -13, 39, -14, 12, -14, 25, -40, 
[14:59:57][I][remote.raw:028]:   26, -13, 14, -38, 65, -27, 13, -25, 67, -12, 26, -13, 26, -40, 65, -13, 13, -39, 53, -38, 27, -13, 27, -38, 13, -14, 12, -26, 65, -14, 39, -26, 26, -13, 40, -39, 26, -12, 40, -13, 40, -12, 27, -38, 27, -26, 26, -13, 40, -12, 26, -14, 51, -40, 13, -14, 
[14:59:57][I][remote.raw:028]:   26, -12, 39, -53, 26, -26, 13, -78, 14, -39, 26, -39, 26, -13, 40, -12, 79, -26, 13, -13, 13, -14, 65, -38, 53, -12, 27, -13, 39, -26, 65, -27, 52, -13, 40, -52, 26, -52, 52, -14, 12, -13, 92, -25, 27, -13, 39, -14, 12, -27, 12, -26, 67, -12, 13, -13, 
[14:59:57][I][remote.raw:028]:   13, -40, 12, -27, 52, -65, 26, -13, 40, -12, 105, -39, 26, -14, 51, -197, 38, -14, 79, -12, 39, -66, 13, -12, 14, -53, 12, -53, 26, -26, 13, -13, 13, -26, 26, -13, 26, -26, 14, -13, 65, -13, 104, -40, 13, -13, 13, -52, 27, -38, 27, -12, 26, -27, 40, 
[... cut ...]
[15:00:20][I][remote.raw:028]:   -12, 79, -26, 13, -12, 66, -13, 39, -26, 26, -14, 12, -53, 26, -26, 39, -39, 26, -40, 53, -25, 26, -14, 12, -53, 26, -12, 27, -40, 12, -65, 40, -13, 52, -79, 25, -40, 12, -27, 26, -26, 26, -13, 40, -51, 14, -12, 53, -52, 13, -14, 38, -53, 12, -53, 14, 
[15:00:20][I][remote.raw:041]:   -77, 66, -13, 52, -13, 14, -26, 12, -39, 26, -67, 51, -14, 38, -14, 26, -13, 26, -13, 26, -14, 26, -51, 184, -65, 12, -40, 52, -26, 26, -14, 130, -26, 14, -12, 27, -26, 12, -27, 13, -13, 14, -12, 26, -27, 12, -53, 78, -13, 14, -12, 13
`

swoboda1337, 2024-09-27:
That looks good. Definitely need a higher bandwidth for better timing resolution. At least 50 maybe 100.

mortenx, 2024-09-27:
with FSK there is nothing in recive...
`
sx127x:
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868000000
  modulation: FSK
  rx_floor: -100
  rx_bandwidth: 50_0kHz
  rx_start: true

remote_receiver:
  pin: GPIO32
  filter: 5us
  idle: 5000us
  dump: raw`

mortenx, 2024-09-27:
https://www.weathershop.co.uk/davis-vantage-pro2-sensors-weatherlink-live-bundle-6112uk

The wireless transmission far out-performs and updates more frequently than any similar wireless weather stations. Using frequency-hopping spread spectrum radio technology at 868mhz; it transmits weather data every 2.5 seconds up to 300m.

swoboda1337, 2024-09-27:
Hmm frequency hoping spread spectrum. OK I will take a look tonight when I get time. Not sure what they are doing then. 

Ecowitt if its close to the board lower the floor to remove all the noise:

sx127x:
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868000000
  modulation: OOK
  rx_floor: -80
  rx_bandwidth: 50_0kHz
  rx_start: true

swoboda1337, 2024-09-27:
Play with rx floor can even go up to like -50 if its really close

swoboda1337, 2024-09-27:
BTW some people are saying Ecowitt is on 867.3M: https://github.com/merbanan/rtl_433/discussions/2202

Would be looking for a signal with something like 500 and 1500.

swoboda1337, 2024-09-27:
These people are claiming 868.3 and 868.4 and its FSK: https://github.com/merbanan/rtl_433/issues/2072

Play around with center frequency and try higher bandwidths too

swoboda1337, 2024-09-27:
Here is a thread on the david one: https://www.wxforum.net/index.php?topic=36267.0

They are saying these frequencies:

868118000
868238000
868360000
868490800
868603000

swoboda1337, 2024-09-27:
Ok found the davis frequencies in the mqtt github:
```
static const uint8_t FRF[DAVIS_FREQ_TABLE_LENGTH][3] =
{
  {0xD9, 0x04, 0x45},
  {0xD9, 0x13, 0x04},
  {0xD9, 0x21, 0xC2},
  {0xD9, 0x0B, 0xA4},
  {0xD9, 0x1A, 0x63}
};
```
Have to convert to freq:

868066711
868297119
868527465
868181884
868412292 
  
 Try those with FSK and a bw of 100

mortenx, 2024-09-27:
`
sx127x:
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868066711
  modulation: FSK
  rx_floor: -80
  rx_bandwidth: 100_0kHz
  rx_start: true

remote_receiver:
  pin: GPIO32
  filter: 5us
  idle: 5000us
  dump: raw
`

`
[17:02:11][I][remote.raw:028]: Received Raw: 27, -13, 27, -13, 40, -14, 40, -26, 14, -13, 93, -14, 13, -27, 13, -27, 40, -26, 14, -26, 40, -14, 26, -14, 13, -107, 13, -27, 26, -14, 13, -13, 14, -26, 27, -40, 40, -40, 27, -66, 14, -26, 67, -13, 14, -13, 27, -66, 14, -53, 27, -26, 27, 
[17:02:11][I][remote.raw:028]:   -80, 27, -93, 27, -26, 14, -66, 27, -13, 27, -13, 27, -13, 14, -40, 13, -40, 53, -54, 13, -53, 14, -26, 14, -40, 40, -13, 13, -14, 26, -27, 13, -27, 27, -53, 27, -53, 13, -80, 27, -27, 13, -27, 26, -67, 13, -14, 26, -40, 40, -40, 14, -26, 14, -40, 26, 
[17:02:11][I][remote.raw:028]:   -27, 13, -14, 13, -53, 14, -13, 13, -14, 13, -13, 27, -40, 40, -13, 27, -13, 27, -27, 26, -27, 13, -27, 40, -53, 27, -107, 26, -14, 13, -67, 13, -27, 40, -13, 13, -40, 14, -26, 40, -27, 13, -40, 27, -27, 53, -13, 94, -13, 13, -14, 53, -27, 26, -14, 13, 
[17:02:11][I][remote.raw:028]:   -80, 13, -107, 27, -40, 13, -53, 27, -53, 14, -13, 13, -54, 13, -107, 26, -14, 40, -26, 40, -40, 14, -26, 14, -13, 53, -14, 26, -14, 26, -40, 14, -26, 27, -13, 14, -13, 13, -54, 26, -27, 13, -40, 14, -26, 14, -13, 40, -80, 27, -93, 53, -40, 14, -13, 
[17:02:11][I][remote.raw:028]:   13, -14, 13, -27, 13, -13, 67, -40, 40, -67, 13, -40, 13, -27, 13, -14, 53, -13, 54, -26, 14, -13, 13, -27, 27, -26, 14, -40, 26, -147, 27, -53, 40, -13, 54, -53, 27, -13, 13, -14, 13, -27, 26, -67, 13, -27, 40, -27, 40, -40, 40, -53, 13, -27, 27, -26, 
[17:02:11][I][remote.raw:028]:   40, -27, 67, -40, 13, -27, 13, -13, 40, -67, 13, -27, 13, -14, 13, -13, 14, -13, 147, -13, 40, -53, 27, -13, 27, -13, 27, -13, 14, -13, 40, -13, 120, -14, 26, -14, 40, -26, 14, -93, 13, -54, 13, -13, 14, -26, 14, -80, 26, -27, 27, -53, 40, -40, 40, 
[17:02:11][I][remote.raw:041]:   -80, 40, -13, 54, -26, 14, -13, 13, -14, 13, -13, 14, -26, 14, -40, 13, -13, 40, -27, 13, -27, 27, -13, 27, -53, 13, -14, 40, -13, 13, -14, 13, -13, 54, -40, 13, -13, 40, -14, 26, -14, 66, -14, 13, -53, 27, -13, 40, -14, 26, -14, 13, -13
`

`
[17:02:51][I][remote.raw:028]: Received Raw: 13, -54, 13, -13, 14, -66, 14, -26, 67, -13, 27, -27, 53, -40, 13, -27, 13, -40, 27, -27, 13, -27, 40, -53, 27, -13, 40, -13, 14, -13, 13, -67, 13, -27, 40, -27, 53, -13, 14, -40, 13, -107, 13, -13, 14, -13, 13, -14, 26, -27, 53, -40, 14, 
[17:02:51][I][remote.raw:028]:   -13, 13, -40, 27, -67, 13, -40, 27, -13, 13, -27, 40, -120, 40, -40, 13, -27, 13, -40, 14, -53, 13, -27, 13, -67, 40, -27, 13, -13, 14, -13, 13, -14, 40, -13, 13, -54, 13, -13, 27, -13, 27, -40, 40, -13, 27, -27, 13, -13, 14, -26, 27, -13, 94, -13, 27, 
[17:02:51][I][remote.raw:028]:   -40, 40, -40, 26, -14, 13, -13, 14, -13, 13, -14, 26, -67, 13, -27, 13, -40, 40, -40, 14, -13, 13, -14, 13, -40, 27, -26, 27, -13, 14, -26, 40, -54, 26, -107, 53, -14, 26, -14, 40, -26, 27, -27, 13, -40, 13, -54, 26, -14, 26, -27, 27, -13, 13, -27, 40, 
[17:02:51][I][remote.raw:028]:   -40, 13, -27, 27, -80, 40, -13, 27, -53, 27, -26, 27, -13, 14, -13, 13, -14, 13, -40, 53, -40, 40, -27, 27, -13, 13, -14, 40, -26, 14, -13, 27, -13, 13, -67, 27, -13, 27, -40, 26, -54, 13, -13, 27, -27, 40, -13, 13, -14, 26, -40, 14, -53, 27, -13, 27, 
[17:02:51][I][remote.raw:028]:   -13, 40, -27, 13, -27, 13, -40, 27, -13, 13, -54, 53, -27, 13, -27, 13, -40, 13, -27, 27, -13, 13, -14, 13, -40, 107, -26, 40, -14, 40, -53, 27, -66, 14, -13, 80, -13, 27, -53, 54, -13, 27, -26, 67, -13, 14, -13, 13, -94, 53, -67, 26, -14, 13, -13, 27, 
[17:02:51][I][remote.raw:028]:   -53, 40, -27, 13, -40, 14, -26, 14, -13, 27, -26, 14, -13, 27, -13, 27, -40, 13, -13, 27, -13, 14, -13, 13, -27, 13, -14, 13, -27, 13, -13, 14, -53, 27, -40, 26, -80, 14, -40, 66, -40, 14, -13, 40, -40, 13, -40, 27, -27, 26, -27, 27, -13, 27, -40, 40, 
[17:02:51][I][remote.raw:041]:   -26, 14, -26, 54, -26, 67, -53, 14, -13, 13, -14, 13, -53, 40, -14, 13, -13, 27, -13, 27, -27, 13, -13, 14, -133, 13, -40, 14, -13, 27, -26, 14, -40, 13, -53, 14, -13, 13, -27, 27, -40, 13, -27, 26, -14, 13, -13, 27, -13, 14, -13

`

swoboda1337, 2024-09-27:
Perfect, now have to try to decide that on a PC. I will take a look later today or tomorrow.

mortenx, 2024-09-27:
what is this  in log?

[17:46:45][C][remote_receiver.esp32:064]: Remote Receiver:
[17:46:45][C][remote_receiver.esp32:065]:   Pin: GPIO32
[17:46:45][W][remote_receiver.esp32:068]: Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' in the pin schema!

mortenx, 2024-09-27:
`
[17:48:58][I][remote.raw:028]: Received Raw: -100, 13, -27, 26, -14, 13, -13, 27, -13, 40, -14, 26, -80, 27, -40, 13, -14, 40, -26, 14, -26, 27, -27, 13, -13, 40, -14, 53, -13, 14, -26, 40, -40, 40, -14, 26, -14, 40, -26, 80, -27, 67, -26, 27, -13, 14, -13, 27, -13, 53, -27, 13, -14, 
[17:48:58][I][remote.raw:028]:   80, -13, 27, -13, 13, -14, 26, -27, 13, -14, 26, -40, 54, -13, 53, -54, 53, -13, 14, -13, 13, -27, 27, -26, 67, -53, 14, -53, 13, -27, 27, -26, 54, -26, 40, -67, 27, -13, 27, -40, 13, -27, 53, -27, 26, -27, 27, -13, 13, -54, 13, -13, 40, -14, 53, -13, 
[17:48:58][I][remote.raw:028]:   14, -26, 40, -40, 80, -14, 40, -13, 27, -13, 13, -107, 53, -27, 27, -13, 13, -27, 13, -27, 53, -67, 13, -54, 26, -54, 13, -13, 14, -40, 26, -14, 66, -27, 40, -80, 13, -54, 13, -67, 26, -27, 67, -53, 27, -13, 27, -26, 27, -27, 40, -13, 13, -14, 26, -14, 
[17:48:58][I][remote.raw:028]:   40, -106, 14, -93, 107, -13, 67, -13, 13, -14, 26, -40, 27, -53, 40, -40, 14, -53, 13, -80, 14, -26, 14, -66, 14, -13, 27, -40, 26, -67, 13, -54, 40, -40, 26, -80, 14, -13, 53, -40, 14, -13, 13, -40, 27, -93, 27, -27, 13, -67, 40, -26, 80, -107, 80, 
[17:48:58][I][remote.raw:028]:   -13, 14, -40, 26, -14, 40, -80, 53, -53, 107, -40, 40, -80, 13, -14, 26, -27, 27, -26, 14, -66, 40, -27, 27, -106, 14, -26, 14, -40, 26, -14, 13, -27, 13, -13, 54, -13, 27, -26, 54, -53, 53, -14, 26, -27, 27, -13, 27, -26, 27, -27, 13, -40, 13, -14, 
[17:48:58][I][remote.raw:028]:   13, -40, 13, -80, 27, -13, 27, -13, 14, -13, 40, -53, 27, -27, 26, -40, 14, -13, 40, -67, 40, -13, 13, -14, 13, -13, 14, -13, 27, -13, 67, -40, 53, -53, 14, -66, 27, -13, 67, -40, 27, -26, 14, -106, 14, -13, 13, -40, 27, -13, 14, -66, 27, -27, 13, -93, 
[17:48:58][I][remote.raw:041]:   54, -13, 40, -27, 26, -40, 14, -13, 40, -40, 13, -40, 14, -13, 13, -40, 14, -26, 14, -80, 26, -27, 40, -13, 14, -40, 13, -67, 26, -14, 13, -53, 54, -13, 13, -14, 13, -13, 14, -53, 53, -14, 80, -26, 27, -67, 13, -27, 13, -13, 40, -14, 13
`

`
[17:48:58][I][remote.raw:028]: Received Raw: 13, -67, 27, -40, 26, -14, 13, -27, 40, -120, 13, -13, 54, -13, 53, -27, 27, -13, 13, -14, 13, -27, 26, -14, 26, -40, 27, -27, 13, -27, 13, -27, 13, -27, 26, -14, 13, -40, 13, -27, 13, -40, 27, -93, 14, -26, 54, -26, 27, -13, 40, -14, 40, 
[17:48:58][I][remote.raw:028]:   -13, 27, -53, 13, -14, 13, -53, 14, -66, 14, -40, 53, -67, 26, -27, 40, -13, 14, -26, 14, -13, 13, -40, 27, -67, 13, -40, 13, -80, 27, -67, 13, -27, 13, -13, 27, -40, 13, -27, 13, -54, 13, -13, 14, -66, 14, -26, 27, -40, 13, -14, 13, -40, 13, -67, 13, 
[17:48:58][I][remote.raw:028]:   -80, 40, -14, 26, -27, 27, -80, 53, -27, 120, -13, 27, -173, 13, -120, 67, -67, 13, -40, 27, -13, 13, -40, 27, -53, 54, -13, 27, -13, 13, -54, 13, -53, 27, -40, 27, -53, 13, -14, 40, -13, 27, -133, 13, -14, 26, -14, 26, -40, 14, -13, 13, -27, 13, -14, 
[17:48:58][I][remote.raw:028]:   26, -27, 13, -40, 27, -27, 13, -40, 13, -14, 40, -26, 14, -80, 13, -13, 40, -14, 13, -40, 13, -14, 40, -26, 27, -13, 27, -40, 13, -14, 13, -13, 27, -40, 27, -13, 53, -14, 13, -13, 14, -80, 26, -40, 67, -13, 40, -54, 26, -40, 40, -40, 14, -53, 13, -27, 
[17:48:58][I][remote.raw:028]:   13, -14, 13, -27, 40, -13, 13, -27, 40, -27, 13, -27, 13, -27, 13, -40, 13, -80, 14, -13, 13, -14, 13, -13, 40, -14, 40, -26, 40, -14, 40, -26, 14, -53, 40, -53, 14, -13, 13, -40, 14, -13, 40, -53, 27, -27, 40, -26, 54, -26, 14, -13, 40, -40, 67, -53, 
[17:48:58][I][remote.raw:028]:   53, -14, 13, -120, 13, -54, 26, -40, 14, -13, 40, -40, 53, -80, 27, -40, 13, -27, 40, -27, 26, -40, 14, -53, 13, -27, 13, -40, 14, -26, 27, -27, 13, -13, 40, -67, 27, -13, 13, -14, 40, -26, 14, -13, 13, -27, 13, -14, 13, -13, 14, -53, 13, -14, 66, -27, 
[17:48:58][I][remote.raw:041]:   120, -53, 27, -27, 26, -67, 53, -40, 67, -93, 14, -53, 13, -54, 13, -13, 14, -26, 14, -26, 27, -27, 13, -53, 14, -1

`

`
[17:48:58][I][remote.raw:028]: Received Raw: 13, -14, 13, -27, 26, -14, 40, -13, 13, -14, 13, -53, 14, -13, 80, -27, 13, -13, 14, -133, 27, -26, 27, -13, 40, -14, 53, -27, 13, -27, 13, -27, 40, -13, 27, -40, 26, -14, 13, -27, 13, -27, 26, -14, 13, -13, 14, -13, 13, -27, 27, -26, 14, 
[17:48:58][I][remote.raw:028]:   -26, 40, -14, 13, -67, 13, -40, 13, -14, 13, -80, 13, -14, 13, -13, 14, -40, 13, -13, 14, -40, 13, -13, 27, -40, 40, -40, 40, -40, 53, -14, 13, -13, 54, -26, 14, -13, 13, -40, 27, -27, 26, -14, 13, -27, 40, -13, 13, -54, 13, -40, 40, -40, 40, -27, 26, 
[17:48:58][I][remote.raw:028]:   -40, 14, -40, 13, -93, 40, -14, 26, -27, 40, -27, 26, -67, 27, -66, 67, -40, 27, -13, 53, -14, 26, -40, 40, -27, 13, -40, 14, -13, 53, -14, 80, -26, 14, -13, 67, -66, 14, -13, 13, -80, 14, -13, 13, -40, 14, -13, 13, -27, 13, -14, 26, -54, 13, -53, 40, 
[17:48:58][I][remote.raw:028]:   -67, 13, -80, 14, -13, 53, -14, 26, -14, 53, -13, 14, -26, 27, -27, 40, -40, 26, -27, 13, -80, 40, -14, 40, -66, 14, -40, 13, -27, 53, -13, 14, -40, 13, -27, 13, -27, 13, -40, 13, -14, 13, -13, 14, -66, 14, -26, 40, -40, 14, -13, 13, -54, 13, -27, 26, 
[17:48:58][I][remote.raw:028]:   -27, 27, -13, 13, -40, 14, -13, 27, -53, 13, -120, 40, -27, 13, -27, 13, -27, 67, -53, 13, -27, 27, -40, 13, -13, 27, -13, 14, -13, 27, -13, 27, -40, 13, -27, 26, -14, 40, -13, 27, -13, 27, -80, 13, -40, 53, -94, 13, -53, 14, -26, 14, -13, 13, -107, 
[17:48:58][I][remote.raw:028]:   13, -27, 27, -13, 40, -53, 14, -26, 40, -54, 40, -26, 14, -13, 40, -67, 26, -27, 13, -27, 13, -120, 27, -27, 26, -14, 26, -27, 27, -26, 14, -26, 14, -40, 93, -53, 14, -53, 27, -4 [... cut ...]
[17:49:37][I][remote.raw:028]:   -26, 14, -13, 120, -13, 14, -13, 27, -26, 14, -13, 40, -13, 14, -26, 40, -14, 26, -67, 13, -14, 13, -40, 40, -13, 40, -134, 40, -40, 26, -40, 14, -13, 13, -27, 27, -26, 14, -26, 14, -13, 13, -27, 13, -14, 53, -13, 54, -26, 14, -26, 14, -40, 26, -27, 
[17:49:37][I][remote.raw:041]:   67, -53, 13, -67, 13, -27, 13, -40, 54, -66, 14, -66, 14, -26, 80, -14, 53, -13, 14, -13, 13, -14, 40, -13, 27, -26, 40, -40, 40, -80, 27, -107, 40, -13, 27, -13, 13, -14, 13, -93, 14, -53, 13, -14, 40, -66, 27, -13, 14, -40, 26, -27

`

mortenx, 2024-09-27:
its working better with 
  frequency: 868297119
  

`
[18:28:55][I][remote.raw:028]: Received Raw: -54, 13, -13, 54, -40, 13, -27, 26, -67, 13, -27, 67, -13, 13, -14, 26, -27, 53, -14, 26, -54, 13, -13, 14, -26, 54, -13, 13, -14, 53, -27, 13, -40, 13, -14, 40, -13, 13, -27, 13, -14, 13, -27, 13, -27, 13, -53, 14, -80, 13, -27, 26, -40, 
[18:28:55][I][remote.raw:028]:   14, -40, 40, -66, 27, -27, 26, -14, 40, -13, 27, -13, 13, -14, 80, -13, 40, -27, 26, -27, 67, -40, 13, -13, 14, -26, 27, -13, 27, -27, 80, -26, 14, -26, 27, -13, 14, -53, 27, -26, 40, -14, 53, -13, 40, -27, 27, -13, 27, -26, 27, -13, 27, -27, 13, -53, 
[18:28:55][I][remote.raw:028]:   40, -67, 27, -13, 27, -26, 27, -40, 53, -14, 13, -27, 53, -13, 14, -26, 14, -13, 13, -14, 26, -14, 13, -67, 26, -40, 14, -13, 13, -40, 14, -13, 13, -14, 26, -14, 13, -13, 14, -13, 27, -40, 40, -13, 27, -26, 14, -53, 13, -94, 13, -27, 13, -13, 14, -26, 
[18:28:55][I][remote.raw:028]:   27, -13, 14, -13, 13, -94, 53, -40, 13, -14, 26, -14, 13, -13, 14, -13, 13, -27, 27, -13, 27, -26, 27, -13, 40, -14, 13, -27, 13, -13, 14, -53, 13, -14, 40, -53, 53, -27, 27, -26, 27, -53, 14, -53, 40, -27, 13, -40, 13, -27, 27, -26, 14, -53, 13, -67, 
[18:28:55][I][remote.raw:028]:   13, -54, 40, -13, 147, -13, 27, -40, 40, -13, 13, -40, 14, -13, 13, -14, 13, -13, 54, -26, 27, -27, 13, -53, 27, -27, 26, -27, 53, -14, 26, -27, 13, -54, 13, -13, 14, -26, 27, -27, 53, -27, 13, -13, 14, -53, 27, -26, 54, -26, 14, -13, 13, -40, 14, -13, 
[18:28:55][I][remote.raw:028]:   40, -27, 66, -27, 53, -14, 13, -53, 14, -13, 27, -13, 13, -40, 14, -13, 53, -14, 26, -14, 13, -13, 27, -13, 54, -13, 13, -14, 53, -27, 146, -54, 13, -67, 13, -13, 14, -53, 13, -54, 53, -13, 14, -13, 13, -14, 40, -13, 13, -40, 27, -53, 54, -13, 13, -14, 
[18:28:55][I][remote.raw:041]:   13, -27, 13, -27, 13, -27, 40, -53, 27, -53, 27, -13, 53, -40, 40, -54, 13, -13, 14, -53, 13, -80, 14, -26, 54, -66, 27, -13, 14, -40, 13, -27, 40, -26, 14, -13, 53, -40, 14, -26, 67, -27, 13, -27, 66, -14, 40, -26, 14, -13, 40

`

swoboda1337, 2024-09-27:
> what is this in log?
>
> [17:46:45][C][remote_receiver.esp32:064]: Remote Receiver: [17:46:45][C][remote_receiver.esp32:065]: Pin: GPIO32 [17:46:45][W][remote_receiver.esp32:068]: Remote Receiver [...]

Yeah that warning should be ignored, mostly for ir receivers

swoboda1337, 2024-09-27:
The only weird thing is the red code says the data rate is 19.2kbps but these captures show 4x that. Let's see if it can be decoded.

swoboda1337, 2024-09-27:
Might be worth trying the max rx bw of 250

mortenx, 2024-09-27:
[20:12:55][C][sx127x:134]:   PA Power: 17 dBm
[20:12:55][C][sx127x:135]:   Frequency: 868.066711 MHz
[20:12:55][C][sx127x:136]:   Modulation: FSK
[20:12:55][C][sx127x:137]:   Rx Bandwidth: 250.0 kHz
[20:12:55][C][sx127x:138]:   Rx Start: true
[20:12:55][C][sx127x:139]:   Rx Floor: -80.0 dBm
[20:12:55][C][sx127x:140]:   FSK Fdev: 5000 Hz
[20:12:55][C][sx127x:141]:   FSK Ramp: 40 us
[20:12:55][C][sx127x:149]:   FSK Shaping: NONE

`
[20:12:57][I][remote.raw:028]: Received Raw: -114, 14, -57, 29, -28, 71, -14, 29, -28, 14, -200, 14, -42, 15, -14, 14, -29, 28, -28, 15, -14, 14, -14, 15, -85, 14, -14, 29, -28, 14, -15, 28, -14, 29, -14, 14, -57, 29, -14, 42, -15, 28, -14, 15, -14, 42, -15, 14, -43, 14, -14, 71, -14, 
[20:12:57][I][remote.raw:028]:   57, -14, 29, -14, 14, -43, 100, -28, 43, -14, 14, -57, 14, -14, 15, -42, 15, -14, 14, -28, 15, -28, 14, -15, 14, -14, 14, -71, 29, -42, 43, -14, 15, -14, 14, -43, 14, -85, 157, -28, 14, -29, 28, -14, 57, -43, 28, -57, 71, -29, 14, -14, 57, -57, 29, 
[20:12:57][I][remote.raw:028]:   -14, 99, -43, 28, -15, 14, -43, 28, -28, 15, -14, 14, -29, 28, -28, 29, -57, 14, -14, 114, -14, 14, -15, 28, -57, 14, -14, 15, -14, 14, -43, 14, -71, 14, -29, 28, -14, 15, -28, 14, -43, 14, -14, 43, -43, 42, -72, 14, -28, 14, -29, 28, -15, 71, -14, 14, 
[20:12:57][I][remote.raw:028]:   -29, 42, -14, 72, -14, 28, -14, 15, -156, 43, -14, 14, -29, 14, -85, 14, -15, 14, -57, 14, -28, 29, -28, 29, -28, 43, -14, 14, -15, 14, -14, 14, -14, 15, -28, 29, -28, 14, -29, 42, -29, 99, -43, 28, -15, 57, -99, 43, -42, 29, -14, 14, -15, 14, -57, 14, 
[20:12:57][I][remote.raw:028]:   -28, 15, -42, 14, -86, 14, -57, 57, -42, 15, -57, 42, -14, 72, -56, 43, -29, 14, -71, 71, -71, 14, -43, 28, -43, 14, -29, 57, -42, 29, -42, 15, -14, 14, -29, 14, -14, 14, -14, 15, -42, 15, -28, 14, -29, 42, -15, 14, -14, 28, -15, 28, -57, 43, -14, 14, 
[20:12:57][I][remote.raw:028]:   -29, 28, -28, 15, -71, 28, -86, 28, -14, 43, -85, 14, -114, 14, -29, 14, -29, 14, -28, 29, -14, 14, -29, 42, -14, 72, -14, 14, -28, 43, -14, 29, -85, 29, -28, 28, -15, 57, -14, 71, -85, 43, -28, 29, -14, 43, -14, 43, -14, 28, -14, 15, -28, 29, -28, 28, 
[20:12:58][I][remote.raw:041]:   -15, 28, -43, 42, -15, 14, -14, 14, -43, 28, -15, 28, -14, 15, -42, 14, -43, 29, -28, 14, -14, 29, -14, 14, -15, 28, -28, 29, -57, 14, -14, 29, -28, 14, -57, 29, -14, 14, -29, 28, -28, 15, -14, 14, -29, 42, -14, 29, -14, 14, -29, 28
`

`
[20:13:03][I][remote.raw:028]: Received Raw: -86, 57, -113, 57, -29, 14, -28, 15, -14, 14, -14, 29, -71, 128, -43, 14, -14, 28, -29, 28, -15, 42, -86, 85, -28, 15, -56, 29, -28, 15, -14, 57, -28, 57, -85, 43, -85, 86, -14, 14, -114, 14, -57, 71, -14, 15, -14, 14, -14, 43, -14, 29, 
[20:13:03][I][remote.raw:028]:   -28, 14, -71, 15, -14, 28, -14, 15, -14, 28, -86, 28, -29, 28, -57, 14, -14, 43, -71, 57, -14, 14, -15, 42, -15, 42, -43, 28, -15, 42, -71, 100, -14, 29, -56, 29, -14, 14, -29, 142, -57, 14, -29, 42, -14, 29, -43, 14, -14, 14, -86, 28, -14, 29, -28, 
[20:13:03][I][remote.raw:028]:   14, -29, 42, -57, 15, -14, 42, -29, 28, -15, 56, -29, 14, -14, 15, -14, 57, -14, 14, -100, 28, -29, 28, -28, 15, -14, 14, -29, 28, -14, 14, -15, 14, -14, 14, -43, 28, -15, 14, -14, 43, -85, 14, -15, 14, -71, 14, -29, 14, -28, 14, -29, 14, -14, 29, -28, 
[20:13:03][I][remote.raw:028]:   29, -71, 42, -43, 43, -85, 57, -28, 15, -71, 28, -57, 14, -15, 28, -28, 15, -57, 14, -28, 14, -15, 14, -28, 43, -71, 85, -29, 43, -14, 28, -29, 71, -57, 28, -29, 28, -71, 43, -28, 29, -28, 43, -14, 57, -57, 28, -43, 28, -15, 14, -14, 28, -29, 14, -43, 
[20:13:03][I][remote.raw:028]:   57, -99, 43, -28, 71, -43, 57, -71, 57, -71, 85, -43, 14, -15, 28, -14, 29, -42, 15, -14, 14, -43, 28, -43, 14, -14, 14, -29, 28, -15, 42, -43, 43, -14, 14, -57, 14, -114, 14, -29, 14, -71, 28, -15, 28, -71, 14, -15, 85, -14, 57, -14, 29, -14, 28, -15, 
[20:13:03][I][remote.raw:028]:   14, -28, 14, -29, 14, -71, 14, -57, 43, -14, 29, -71, 28, -57, 43, -57, 14, -43, 14, -42, 128, -72, 14, -42, 29, -43, 28, -14, 29, -28, 14, -57, 43, -14, 14, -15, 56, -29, 71, -43, 28, -100, 28, -29, 28, -71, 71, -57, 71, -14, 29, -57, 28, -43, 28, 
[20:13:03][I][remote.raw:041]:   -29, 14, -14, 14, -29, 43, -14, 14, -14, 14, -57, 29, -28, 29, -42, 29, -14, 14, -14, 15, -28, 57, -14, 14, -15, 28, -29, 28, -28, 29, -14, 14, -15, 14, -14, 14, -57, 14, -15, 56, -57, 29, -14, 28, -15, 28, -29, 14, -14, 14, -14, 15, -28, 29

`

`

[20:13:10][I][remote.raw:028]: Received Raw: -85, 43, -14, 14, -43, 28, -15, 14, -43, 28, -43, 28, -14, 29, -28, 14, -29, 14, -14, 15, -71, 14, -71, 43, -42, 71, -29, 14, -57, 28, -100, 57, -114, 14, -28, 43, -43, 57, -71, 42, -15, 42, -43, 43, -28, 57, -28, 29, -14, 14, -29, 28, -29, 
[20:13:10][I][remote.raw:028]:   14, -142, 28, -15, 42, -15, 28, -14, 43, -14, 14, -57, 71, -15, 14, -14, 14, -14, 29, -28, 43, -14, 29, -42, 15, -71, 42, -15, 14, -14, 43, -71, 28, -14, 15, -128, 14, -128, 28, -15, 14, -28, 14, -43, 14, -29, 28, -14, 72, -14, 28, -57, 85, -29, 57, 
[20:13:10][I][remote.raw:028]:   -14, 43, -42, 86, -14, 14, -29, 14, -42, 86, -71, 14, -29, 28, -14, 29, -28, 14, -43, 28, -15, 14, -28, 43, -28, 72, -14, 71, -28, 29, -42, 15, -14, 43, -28, 43, -85, 43, -28, 29, -14, 28, -14, 29, -57, 14, -28, 29, -28, 29, -57, 14, -14, 14, -29, 14, 
[20:13:10][I][remote.raw:028]:   -14, 14, -29, 14, -57, 14, -57, 100, -14, 14, -57, 43, -14, 14, -29, 42, -43, 14, -14, 29, -42, 29, -43, 14, -28, 14, -15, 42, -15, 28, -14, 43, -14, 29, -28, 71, -43, 28, -14, 43, -29, 56, -15, 7 [... cut ...]
[20:13:15][I][remote.raw:028]:   -14, 28, -29, 57, -14, 14, -14, 15, -14, 28, -29, 14, -28, 29, -28, 15, -42, 14, -15, 14, -14, 14, -15, 56, -57, 15, -14, 57, -42, 43, -14, 43, -57, 57, -71, 14, -71, 71, -71, 57, -43, 28, -15, 57, -14, 14, -71, 29, -85, 14, -29, 14, -14, 14, -43, 14, 
[20:13:15][I][remote.raw:041]:   -14, 15, -14, 28, -114, 28, -86, 57, -28, 14, -43, 14, -29, 14, -57, 57, -14, 28, -71, 15, -14, 14, -29, 14, -14, 128, -28, 15, -28, 14, -57, 57, -29, 28, -14, 29, -14, 28, -15, 14, -42, 15, -14, 14, -29, 42, -29, 14, -14, 14, -86, 42

`

mortenx, 2024-09-28:
wn30al ecowitt

id":248,"channel":1,"battery_ok":1,"temperature_C":24.1,"humidity":234,"data":"152c000200",
`
19:59:50	[I]	[remote.raw:041]	
Received Raw: -135, 822, -862, 189, -148, 215, -337, 216, -54, 471, -202, 162, -135, 1482, -175, 148, -54, 431, -229, 81, -849, 243, -54, 161, -54, 768, -202, 68, -242, 243, -269, 148, -256, 431, -189, 404, -149, 202
19:59:50	[I]	[remote.raw:041]	
Received Raw: -471, 351, -363, 162, -229, 189, -1091, 270, -565, 189, -229, 283, -108, 458, -728, 970, -54, 188, -1132, 256, -54, 67, -256, 54, -418, 108, -269, 162, -579, 445, -161, 95, -229, 283

`

"id":248,"channel":1,"battery_ok":1,"temperature_C":24.2,"humidity":234,"data":"432c000200",

`

20:01:52	[I]	[remote.raw:041]	
Received Raw: -269, 512, -391, 391, -175, 296, -95, 242, -404, 580, -134, 162, -404, 108, -94, 203, -80, 162, -162, 323, -418, 54, -350, 121, -108, 202, -324, 188, -297, 175, -552, 54, -54, 741, -121, 229, -162, 270, -309, 297, -418, 148, -121, 54, -633
20:01:52	[I]	[remote.raw:041]	
Received Raw: -256, 108, -215, 688, -175, 94, -1024, 81, -229, 1172, -54, 108, -512, 1024, -269, 337, -68, 161, -297, 67, -472, 202, -431, 148, -445, 67, -391, 296, -431, 459, -498, 202, -701

`

mortenx, 2024-09-28:
wn31 ecowitt

"id":171,"channel":2,"battery_ok":1,"temperature_C":22.8,"humidity":55,"data":"aa6c000200"

20:21:45	[I]	[remote.raw:041]	
Received Raw: -229, 229, -135, 148, -485, 81, -458, 512, -1199, 108, -337, 215, -135, 175, -68, 175, -54, 161, -297, 902, -553, 903, -822, 53, -68, 364, -67, 54, -903, 53, -54, 512
20:21:45	[I]	[remote.raw:041]	
Received Raw: -378, 943, -431, 283, -364, 512, -283, 54, -660, 296, -121, 216, -445, 229, -108, 53, -54, 391, -108, 310, -606, 256, -256, 606, -162, 202, -526, 458, -108, 444, -997, 404

"id":171,"channel":2,"battery_ok":1,"temperature_C":22.8,"humidity":55,"data":"aa6c000200","rssi":-44,"duration":9000}

`
20:22:47	[I]	[remote.raw:041]	
Received Raw: -539, 619, -256, 593, -68, 485, -579, 202, -54, 957, -377, 431, -337, 808, -741, 809, -108, 256, -1805, 458, -728
20:22:47	[I]	[remote.raw:041]	
Received Raw: -687, 134, -149, 444, -418, 539, -620, 808, -162, 310, -94, 364, -135, 552, -242, 499, -445, 54, -148, 417, -81, 135, -202, 108, -121, 162, -323, 135, -283, 296, -553, 94, -135, 242, -1240

`

id":171,"channel":2,"battery_ok":1,"temperature_C":22.8,"humidity":55,"data":"aa6c000200","rssi":-62,"duration":9000}

`
20:24:51	[I]	[remote.raw:041]	
Received Raw: -67, 108, -418, 161, -202, 1051, -526, 148, -54, 2412, -67, 1550, -1886, 485, -162, 81, -282, 459, -458
20:24:51	[I]	[remote.raw:041]	
Received Raw: -81, 673, -459, 350, -54, 134, -216, 54, -202, 67, -1402, 81, -471, 472, -1037, 768, -445, 202, -108, 67, -418, 660, -216, 323, -283, 754, -162, 728, -161, 175

`

swoboda1337, 2024-09-28:
These ecowitts are OOK or FSK?

mortenx, 2024-09-28:
sx127x:
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868294500
  modulation: FSK
  rx_floor: -80
  rx_bandwidth: 125_0kHz
  rx_start: true

remote_receiver:
  pin: GPIO34
  filter: 50us
  idle: 5000us
  dump: raw
captive_portal:

swoboda1337, 2024-09-28:
Ok, since you are seeing values of  -54 etc you need to reduce the filter to something like 5us and increase the bw to 250 again. Hopefully I have sometime tonight to look at the data.

mortenx, 2024-09-28:
sx127x:
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868294500
  modulation: FSK
  rx_floor: -80
  rx_bandwidth: 250_0kHz
  rx_start: true

remote_receiver:
  pin: GPIO34
  filter: 5us
  idle: 5000us
  dump: raw
captive_portal:

`

22:40:11 | [I] | [remote.raw:028] | Received Raw: -71, 57, -14, 14, -14, 57, -14, 15, -42, 29, -28, 14, -43, 43, -42, 114, -57, 43, -28, 29, -14, 14, -14, 71, -71, 29, -14, 128, -14, 15, -42, 29, -57, 14, -14, 28, -15, 14, -28, 71, -15, 14, -14, 57, -14, 43, -14, 14, -29, 14, -43, 28, -14,
-- | -- | -- | --
22:40:11 | [I] | [remote.raw:028] | 15, -14, 14, -14, 43, -28, 100, -71, 28, -29, 43, -28, 14, -29, 28, -43, 14, -14, 57, -14, 29, -14, 43, -14, 71, -43, 42, -29, 14, -14, 29, -42, 43, -14, 43, -28, 43, -14, 14, -15, 57, -71, 28, -14, 43, -14, 14, -43, 43, -28, 43, -14, 14, -15, 99, -85,
22:40:11 | [I] | [remote.raw:028] | 15, -14, 14, -14, 15, -14, 14, -14, 14, -57, 15, -14, 14, -43, 14, -14, 14, -29, 14, -28, 15, -28, 14, -15, 14, -14, 14, -29, 57, -28, 14, -29, 42, -15, 42, -14, 43, -14, 43, -14, 29, -14, 14, -14, 43, -43, 71, -42, 29, -28, 15, -14, 42, -15, 42, -29,
22:40:11 | [I] | [remote.raw:028] | 14, -14, 29, -14, 28, -15, 56, -29, 14, -43, 14, -28, 15, -14, 14, -29, 28, -14, 29, -113, 86, -71, 28, -15, 14, -28, 43, -43, 28, -14, 14, -29, 14, -29, 14, -28, 29, -14, 14, -14, 15, -28, 14, -29, 28, -14, 15, -28, 28, -57, 29, -14, 28, -29, 57, -57,
22:40:11 | [I] | [remote.raw:028] | 28, -114, 14, -14, 43, -14, 14, -15, 71, -28, 43, -14, 29, -14, 71, -14, 29, -56, 29, -14, 29, -28, 14, -14, 43, -14, 15, -14, 114, -28, 43, -14, 14, -85, 15, -14, 57, -28, 14, -72, 14, -42, 29, -43, 14, -14, 43, -28, 14, -29, 14, -100, 56, -29, 14,
22:40:11 | [I] | [remote.raw:028] | -14, 43, -28, 29, -43, 14, -28, 14, -29, 28, -15, 14, -14, 71, -14, 15, -14, 14, -57, 28, -15, 42, -14, 15, -14, 43, -42, 43, -14, 14, -15, 14, -42, 128, -29, 14, -14, 15, -56, 29, -28, 29, -14, 43, -28, 29, -28, 14, -14, 29, -28, 15, -14, 28, -14, 29,
22:40:11 | [I] | [remote.raw:041] | -28, 29, -42, 15, -28, 14, -29, 14, -14, 29, -28, 14, -15, 28, -14, 29, -28, 43, -28, 29, -14, 14, -71, 43, -14, 57, -28, 15, -28, 43, -14, 43, -28, 14, -15, 14, -14, 28, -15, 14, -57, 14, -14, 29, -57, 14, -14, 28, -15, 14, -14, 14
22:40:11 | [W] | [component:237] | Component remote_receiver took a long time for an operation (178 ms).
22:40:11 | [W] | [component:238] | Components should block for at most 30 ms.
22:40:11 | [I] | [remote.raw:028] | Received Raw: -43, 43, -14, 28, -15, 42, -15, 14, -14, 28, -15, 28, -14, 43, -57, 14, -14, 43, -85, 185, -29, 42, -29, 14, -43, 42, -15, 113, -15, 28, -43, 28, -14, 15, -28, 14, -14, 29, -14, 128, -29, 28, -14, 57, -14, 86, -14, 28, -15, 28, -14, 29,
22:40:12 | [I] | [remote.raw:028] | -14, 28, -15, 56, -15, 71, -14, 57, -28, 15, -14, 28, -29, 14, -43, 14, -28, 14, -15, 28, -14, 15, -14, 14, -14, 29, -14, 14, -14, 15, -28, 14, -14, 72, -14, 28, -14, 57, -15, 99, -14, 29, -14, 43, -28, 14, -29, 42, -43, 85, -29, 28, -15, 14, -14, 71,
22:40:12 | [I] | [remote.raw:028] | -29, 42, -43, 14, -14, 15, -42, 71, -29, 85, -29, 56, -15, 28, -85, 15, -14, 14, -14, 43, -14, 57, -14, 15, -14, 28, -14, 29, -14, 14, -15, 28, -28, 86, -14, 85, -29, 14, -57, 57, -42, 29, -71, 114, -28, 29, -14, 185, -28, 86, -14, 14, -71, 29, -14,
22:40:12 | [I] | [remote.raw:028] | 28, -43, 14, -29, 184, -57, 15, -42, 14, -43, 14, -15, 28, -14, 29, -42, 43, -28, 15, -14, 71, -14, 29, -57, 28, -14, 14, -43, 14, -15, 56, -43, 43, -14, 57, -14, 14, -15, 14, -14, 114, -14, 14, -14, 29, -14, 14, -29, 14, -14, 29, -28, 43, -14, 43,
22:40:12 | [I] | [remote.raw:028] | -14, 14, -57, 43, -14, 14, -29, 14, -14, 14, -14, 15, -42, 15, -28, 28, -57, 43, -28, 57, -15, 14, -14, 28, -15, 14, -28, 15, -28, 14, -71, 43, -28, 15, -14, 14, -14, 43, -14, 29, -14, 14, -57, 57, -14, 57, -14, 14, -15, 42, -43, 14, -14, 57, -15, 14,
22:40:12 | [I] | [remote.raw:028] | -28, 14, -15, 14, -28, 43, -43, 28, -29, 14, -28, 14, -15, 28, -29, 14, -14, 28, -15, 14, -14, 29, -14, 14, -43, 42, -15, 42, -14, 15, -28, 14, -43, 28, -15, 14, -71, 43, -28, 14, -15, 14, -14, 14, -14, 15, -28, 43, -14, 28, -15, 14, -14, 14, -15, 42,
22:40:12 | [I] | [remote.raw:041] | -57, 14, -15, 28, -14, 114, -28, 15, -28, 14, -15, 14, -14, 28, -29, 71, -14, 29, -14, 28, -29, 99, -15, 56, -15, 14, -14, 57, -14, 100, -42, 15, -57, 42, -29, 14, -28, 29, -14, 28, -15, 28, -14, 15, -14, 28, -29, 42, -15, 28, -57, 28, -43, 14
22:40:12 | [W] | [component:237] | Component remote_receiver took a long time for an operation (179 ms).
22:40:12 | [W] | [component:238] | Components should block for at most 30 ms.
22:41:06 | [W] | [api.connection:104] | ESPHome Logs 2024.9.1 (192.168.1.223): Connection closed
22:41:13 | [I] | [remote.raw:028] | Received Raw: -71, 15, -14, 14, -14, 14, -15, 71, -14, 43, -14, 99, -29, 14, -14, 15, -14, 14, -14, 43, -28, 57, -29, 85, -28, 72, -28, 14, -29, 14, -14, 14, -15, 42, -57, 14, -15, 56, -15, 42, -29, 71, -14, 57, -14, 29, -14, 14, -14, 29, -14, 28, -15,
22:41:13 | [I] | [remote.raw:028] | 85, -14, 29, -42, 57, -86, 14, -14, 14, -14, 15, -42, 15, -14, 14, -28, 43, -14, 15, -14, 57, -14, 142, -14, 29, -14, 28, -29, 14, -14, 43, -43, 14, -14, 14, -15, 14, -42, 15, -14, 14, -57, 57, -14, 128, -29, 128, -71, 42, -71, 29, -14, 43, -28, 14,
22:41:13 | [I] | [remote.raw:028] | -15, 28, -43, 170, -15, 14, -43, 85, -28, 43, -43, 99, -43, 14, -14, 29, -42, 29, -14, 14, -29, 14, -14, 114, -57, 14, -85, 15, -14, 14, -57, 85, -15, 42, -43, 28, -15,  [... cut ...]
22:41:13 | [I] | [remote.raw:028] | 43, -14, 14, -43, 14, -14, 43, -14, 28, -15, 42, -15, 14, -14, 28, -15, 85, -43, 14, -14, 71, -14, 43, -14, 29, -14, 28, -15, 42, -15, 56, -15, 14, -14, 57, -43, 28, -29, 14, -14, 57, -14, 29, -28, 28, -15, 42, -29, 42, -29, 57, -14, 14, -14, 15, -14,
22:41:13 | [I] | [remote.raw:041] | 14, -100, 14, -28, 43, -14, 43, -14, 43, -14, 43, -14, 43, -14, 28, -14, 15, -28, 29, -14, 28, -14, 86, -14, 57, -99, 15, -14, 71, -28, 15, -14, 57, -14, 14, -14, 43, -14, 71, -15, 14, -14, 29, -42, 14, -15, 28, -43, 99, -29, 14, -28, 100

`

swoboda1337, 2024-09-28:
So a few things:
1: why are you using GPIO34? Its GPIO32
2: the rmt is running out of memory, each capture is 384 samples long which is example 3 memory blocks worth (the default size), lets increase this to the max, put this in remote_receiver "memory_blocks: 8"
3: ecowitt data looks exactly like the davis I think you are capturing the same thing, might have to pull the batteries on everything but one device to be sure.

mortenx, 2024-09-29:
i have 2 boards i think i made different setups for testing for davis is LILYGO® TTGO LoRa32 board with pin32 and other for ecowitt Heltec WIFI lora 32 V3  is pin 34

with  "memory_blocks: 8"...

`
[12:27:44][I][remote.raw:028]: Received Raw: -28, 14, -57, 14, -57, 57, -14, 15, -14, 14, -14, 14, -15, 57, -14, 99, -57, 43, -14, 14, -15, 14, -14, 14, -14, 29, -14, 14, -29, 28, -43, 71, -28, 15, -42, 29, -14, 43, -28, 29, -14, 42, -43, 71, -14, 15, -28, 14, -15, 42, -14, 29, -14, 
[12:27:44][I][remote.raw:028]:   114, -14, 14, -71, 72, -42, 14, -29, 14, -29, 14, -57, 71, -14, 28, -15, 14, -14, 14, -15, 42, -14, 29, -14, 43, -28, 71, -15, 14, -14, 71, -14, 29, -14, 43, -14, 43, -14, 71, -14, 57, -57, 14, -14, 15, -42, 29, -14, 43, -56, 15, -14, 85, -57, 14, -15, 
[12:27:44][I][remote.raw:028]:   85, -28, 29, -28, 114, -14, 43, -14, 14, -15, 14, -14, 14, -14, 143, -14, 28, -29, 42, -15, 14, -28, 29, -28, 29, -28, 57, -14, 71, -15, 42, -86, 14, -14, 14, -29, 14, -14, 43, -42, 29, -14, 14, -15, 42, -43, 14, -29, 14, -14, 14, -14, 29, -43, 14, 
[12:27:44][I][remote.raw:028]:   -14, 14, -14, 29, -14, 14, -15, 14, -42, 29, -57, 28, -29, 99, -43, 14, -14, 86, -14, 28, -15, 14, -14, 14, -29, 142, -57, 14, -14, 14, -15, 28, -43, 42, -15, 14, -14, 43, -14, 14, -14, 29, -28, 15, -14, 42, -15, 71, -14, 43, -85, 100, -28, 57, -28, 
[12:27:44][I][remote.raw:028]:   43, -14, 14, -43, 14, -100, 71, -57, 57, -14, 71, -14, 15, -14, 14, -43, 42, -15, 28, -14, 14, -86, 14, -14, 43, -14, 14, -29, 28, -14, 72, -14, 14, -57, 14, -14, 43, -57, 28, -15, 28, -14, 43, -14, 114, -14, 71, -43, 14, -14, 15, -28, 14, -14, 15, 
[12:27:44][I][remote.raw:028]:   -14, 43, -14, 85, -14, 86, -28, 14, -43, 14, -57, 14, -29, 57, -14, 14, -43, 14, -28, 29, -28, 57, -14, 15, -14, 28, -15, 113, -15, 42, -71, 43, -14, 14, -29, 14, -14, 71, -43, 14, -43, 85, -15, 14, -57, 14, -14, 14, -15, 28, -28, 43, -29, 14, -14, 28, 
[12:27:44][I][remote.raw:028]:   -29, 28, -43, 28, -43, 100, -28, 29, -14, 71, -14, 14, -15, 42, -14, 72, -14, 14, -14, 43, -28, 57, -14, 15, -28, 29, -42, 14, -15, 42, -57, 29, -28, 14, -15, 14, -28, 14, -15, 156, -29, 28, -43, 42, -15, 14, -28, 43, -14, 14, -15, 14, -14, 14, -29, 
[12:27:44][I][remote.raw:028]:   42, -29, 28, -14, 15, -14, 28, -29, 14, -71, 57, -28, 72, -14, 14, -14, 14, -15, 14, -28, 15, -14, 42, -29, 14, -43, 42, -15, 14, -28, 15, -14, 42, -15, 57, -14, 57, -14, 71, -43, 14, -85, 57, -14, 15, -28, 43, -42, 29, -14, 43, -14, 28, -29, 14, -14, 
[12:27:44][I][remote.raw:028]:   57, -14, 43, -14, 43, -57, 14, -43, 14, -14, 114, -14, 71, -29, 14, -14, 29, -14, 14, -14, 14, -29, 71, -43, 14, -28, 43, -14, 43, -14, 57, -43, 14, -71, 14, -14, 29, -14, 14, -15, 56, -15, 42, -15, 28, -28, 29, -43, 28, -43, 28, -29, 42, -14, 72, -28, 
[12:27:44][I][remote.raw:028]:   57, -14, 43, -14, 71, -57, 14, -14, 29, -71, 114, -57, 28, -29, 71, -71, 42, -57, 43, -14, 14, -43, 14, -15, 14, -14, 28, -15, 42, -29, 14, -14, 57, -14, 15, -14, 14, -14, 29, -28, 14, -15, 28, -14, 71, -43, 14, -14, 72, -14, 28, -29, 42, -15, 28, -14, 
[12:27:44][I][remote.raw:028]:   71, -43, 14, -29, 199, -14, 28, -57, 100, -43, 28, -28, 29, -57, 14, -57, 43, -14, 14, -14, 14, -100, 28, -15, 28, -14, 15, -14, 42, -57, 100, -14, 14, -29, 42, -15, 14, -28, 15, -28, 14, -29, 14, -14, 43, -14, 28, -29, 14, -14, 15, -14, 14, -14, 57, 
[12:27:44][I][remote.raw:028]:   -43, 14, -71, 29, -14, 114, -71, 99, -14, 15, -42, 15, -28, 14, -14, 29, -28, 15, -14, 28, -29, 14, -14, 29, -14, 42, -15, 99, -29, 42, -15, 85, -14, 14, -57, 14, -57, 29, -14, 28, -43, 71, -29, 14, -43, 42, -29, 99, -57, 43, -28, 29, -28, 29, -56, 
[12:27:45][I][remote.raw:028]:   114, -14, 15, -14, 28, -15, 42, -14, 57, -15, 56, -15, 28, -57, 100, -14, 42, -57, 43, -114, 28, -29, 14, -43, 14, -14, 14, -29, 42, -15, 28, -14, 14, -15, 14, -43, 42, -43, 199, -14, 114, -14, 43, -14, 29, -14, 28, -29, 14, -43, 71, -57, 14, -14, 57, 
[12:27:45][I][remote.raw:028]:   -14, 29, -28, 14, -43, 57, -14, 28, -15, 71, -57, 57, -14, 14, -14, 43, -43, 71, -42, 29, -14, 57, -14, 71, -14, 15, -42, 157, -28, 29, -42, 15, -14, 14, -14, 43, -14, 29, -28, 57, -57, 28, -29, 14, -14, 29, -14, 28, -29, 57, -28, 43, -28, 14, -57, 29, 
[12:27:45][I][remote.raw:028]:   -28, 14, -15, 14, -14, 14, -29, 284, -29, 28, -14, 43, -14, 85, -29, 114, -14, 71, -57, 43, -14, 14, -14, 14, -15, 99, -29, 57, -56, 29, -14, 14, -15, 28, -14, 14, -15, 14, -43, 14, -14, 28, -15, 14, -28, 15, -42, 29, -14, 114, -14, 57, -14, 28, -29, 
[12:27:45][I][remote.raw:028]:   43, -28, 71, -29, 14, -14, 43, -42, 29, -42, 57, -43, 57, -14, 14, -15, 14, -14, 14, -14, 57, -15, 71, -14, 85, -86, 28, -14, 57, -29, 14, -14, 57, -28, 29, -14, 28, -15, 71, -42, 72, -28, 28, -43, 43, -14, 28, -15, 14, -14, 14, -29, 28, -14, 57, -15, 
[12:27:45][I][remote.raw:028]:   85, -14, 43, -28, 128, -14, 29, -14, 14, -43, 14, -43, 128, -28, 15, -28, 71, -14, 29, -14, 71, -14, 15, -28, 43, -28, 14, -29, 99, -15, 28, -28, 57, -15, 28, -28, 15, -14, 14, -43, 28, -57, 29, -14, 42, -43, 14, -15, 85, -14, 85, -15, 42, -15, 42, 
[12:27:45][I][remote.raw:028]:   -14, 43, -29, 14, -14, 14, -29, 14, -14, 43, -28, 29, -14, 71, -14, 57, -14, 57, -14, 29, -99, 15, -28, 29, -14, 71, -28, 15, -14, 14, -14, 100, -14, 28, -29, 14, -29, 42, -14, 15, -42, 15, -14, 14, -14, 14, -29, 57, -42, 29, -43, 99, -43, 57, -71, 14, 
[12:27:45][I][remote.raw:041]:   -14, 43, -28, 15, -14, 42, -15, 14, -28, 15, -28, 14, -57, 29, -28, 28, -29, 14, -43, 14, -14, 14, -29, 43, -14, 28, -14, 86
[... cut ...] , -43, 28, -14, 29, -42, 15, -14, 14, 
[12:27:45][I][remote.raw:028]:   -14, 29, -14, 43, -14, 14, -14, 29, -14, 28, -43, 14, -15, 28, -28, 29, -14, 14, -15, 28, -14, 57, -14, 71, -29, 14, -14, 43, -14, 43, -14, 57, -14, 29, -42, 29, -14, 85, -57, 14, -15, 57, -28, 14, -57, 29, -14, 14, -14, 29, -14, 128, -14, 14, -15, 28, 
[12:27:45][I][remote.raw:041]:   -14, 57, -14, 57, -14, 15, -14, 99, -29, 71, -14, 43, -14, 43, -28, 14, -86, 28, -14, 43, -14, 15, -14, 42, -15, 14, -71, 14, -29, 28, -14

`

swoboda1337, 2024-09-30:
I spent some time on this today. Looks like something might be wrong on my side. I have mainly tested these radios in OOK mode. I did a quick FSK sanity test and it looked but now I am not sure. I tried to transmit something similar to davis and it doesn't receive properly on the other board, looks similar to what you are seeing. Let me play around with this some more.

swoboda1337, 2024-10-03:
Ok I found some config issues in FSK mode. Hopefully will have something to try by the weekend.

swoboda1337, 2024-10-05:
I have it working 100% reliably now when I send FSK from one board to another. It was sort of working before just too much noise. I will clean this up and update my github, hopefully tomorrow.

swoboda1337, 2024-10-06:
Can try with this config (change the frequency), need to set rx_duration, dio0 and dio2. I am also limiting the length of the dumps as those can slow things down. Hopefully see something similar to: -50 +50 -100 +50 -100 etc without any -10s or -1000s in there. 

```
sx127x:
  dio0_pin: GPIO26
  dio2_pin: 
    number: GPIO32
    allow_other_uses: true
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 433920000
  rx_bandwidth: 100_0kHz
  rx_duration: 100ms
  rx_floor: -90
  rx_start: true
  modulation: FSK

remote_receiver:
  pin:
    number: GPIO32
    allow_other_uses: true
  filter: 5us
  idle: 5000us
  buffer_size: 100000b
  memory_blocks: 8
  on_raw:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "Received raw data with length %d", x.size());
          remote_base::RawDumper dumper;
          x.resize(128);
          dumper.dump(remote_base::RemoteReceiveData(x, 0, remote_base::TOLERANCE_MODE_TIME));
```

swoboda1337, 2024-10-07:
I also added bitrate, which we know is 19200 for davis. Can reduce the bandwidth and add the bitrate:

```
sx127x:
  dio0_pin: GPIO26
  dio2_pin: 
    number: GPIO32
    allow_other_uses: true
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 433920000
  bitrate: 19200
  rx_bandwidth: 25_0kHz
  rx_duration: 100ms
  rx_floor: -80
  rx_start: true
  modulation: FSK

remote_receiver:
  pin:
    number: GPIO32
    allow_other_uses: true
  filter: 5us
  idle: 5000us
  buffer_size: 100000b
  memory_blocks: 8
  on_raw:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "Received raw data with length %d", x.size());
          remote_base::RawDumper dumper;
          x.resize(128);
          dumper.dump(remote_base::RemoteReceiveData(x, 0, remote_base::TOLERANCE_MODE_TIME));
```

mortenx, 2024-10-07:
With these last settings i get this result from davis. I also checked that I took out the Davis battery to make sure there was a Davis signal

`
[13:44:48][D][lambda:071]: Received raw data with length 934
[13:44:48][I][remote.raw:028]: Received Raw: 618, -1892, 55, -273, 112, -1608, 122, -1340, 48, -100, 207, -255, 255, -840, 155, -53, 52, -48, 50, -155, 155, -47, 150, -53, 50, -107, 58, -202, 48, -102, 103, -52, 50, -310, 158, -52, 160, -363, 207, -105, 53, -317, 213, -47, 265, -108, 
[13:44:48][I][remote.raw:028]:   57, -48, 57, -50, 53, -100, 52, -95, 103, -50, 102, -160, 50, -208, 52, -50, 58, -50, 150, -57, 50, -50, 53, -50, 47, -320, 58, -100, 57, -55, 50, -215, 108, -165, 107, -108, 67, -160, 118, -55, 162, -225, 48, -207, 53, -102, 168, -45, 160, -52, 323, 
[13:44:48][I][remote.raw:041]:   -207, 108, -50, 52, -60, 55, -48, 210, -60, 152, -63, 55, -165, 115, -55, 55, -55, 112, -48, 162, -50, 218, -60, 165, -50, 112, -385, 45, -113, 55
[13:44:48][W][component:237]: Component remote_receiver took a long time for an operation (67 ms).
[13:44:48][W][component:238]: Components should block for at most 30 ms.
[13:45:01][D][lambda:071]: Received raw data with length 964
[13:45:01][I][remote.raw:028]: Received Raw: 654, -1667, 50, -245, 100, -695, 98, -1230, 105, -925, 205, -90, 97, -248, 940, -42, 55, -143, 97, -53, 52, -48, 50, -47, 53, -150, 52, -58, 265, -152, 108, -62, 268, -425, 107, -208, 115, -57, 100, -163, 100, -265, 165, -102, 160, -210, 
[13:45:01][I][remote.raw:028]:   53, -52, 53, -100, 97, -60, 105, -48, 207, -60, 48, -52, 50, -110, 60, -158, 327, -268, 217, -48, 62, -55, 215, -320, 58, -47, 113, -157, 60, -115, 45, -60, 48, -115, 47, -115, 110, -50, 58, -50, 117, -48, 105, -57, 168, -105, 157, -53, 52, -53, 105, 
[13:45:01][I][remote.raw:041]:   -47, 210, -50, 60, -45, 115, -100, 53, -52, 53, -50, 52, -60, 48, -62, 65, -115, 208, -62, 153, -60, 272, -160, 55, -223, 55, -107, 48, -52, 53
[13:45:01][W][component:237]: Component remote_receiver took a long time for an operation (68 ms).
[13:45:01][W][component:238]: Components should block for at most 30 ms.
[13:45:13][D][lambda:071]: Received raw data with length 864
[13:45:13][I][remote.raw:028]: Received Raw: 45, -63, 441, -1945, 63, -270, 117, -1623, 280, -387, 110, -933, 47, -108, 60, -207, 158, -867, 215, -55, 50, -58, 272, -165, 158, -175, 55, -275, 47, -60, 58, -380, 65, -105, 157, -58, 110, -47, 168, -50, 322, -120, 50, -225, 163, -107, 
[13:45:13][I][remote.raw:028]:   55, -110, 45, -210, 58, -60, 152, -58, 102, -58, 157, -38, 55, -107, 113, -57, 108, -102, 50, -63, 105, -107, 165, -50, 163, -52, 58, -205, 212, -110, 50, -118, 270, -115, 47, -60, 55, -38, 115, -57, 103, -207, 215, -588, 155, -52, 63, -100, 117, -158, 
[13:45:13][I][remote.raw:041]:   102, -105, 270, -105, 108, -165, 62, -108, 45, -167, 48, -52, 170, -310, 165, -55, 223, -110, 277, -58, 55, -112, 113, -107, 55, -225, 118, -115, 55, -47, 225
[13:45:13][W][component:237]: Component remote_receiver took a long time for an operation (66 ms).
[13:45:13][W][component:238]: Components should block for at most 30 ms.
[13:45:26][D][lambda:071]: Received raw data with length 958
[13:45:26][I][remote.raw:028]: Received Raw: 687, -1805, 57, -275, 48, -2212, 58, -720, 110, -500, 47, -158, 155, -917, 213, -115, 55, -55, 57, -110, 58, -60, 55, -110, 60, -110, 162, -105, 313, -62, 163, -110, 100, -52, 215, -63, 217, -53, 52, -53, 217, -50, 58, -50, 50, -160, 52, 
[13:45:26][I][remote.raw:028]:   -53, 105, -47, 50, -48, 50, -152, 53, -207, 63, -55, 732, -213, 57, -50, 108, -50, 50, -112, 100, -118, 50, -110, 57, -55, 170, -115, 435, -60, 110, -168, 110, -55, 57, -115, 165, -113, 57, -110, 278, -62, 45, -60, 65, -53, 55, -57, 115, -58, 55, -57, 
[13:45:26][I][remote.raw:041]:   110, -55, 65, -63, 52, -58, 110, -160, 55, -117, 105, -385, 55, -113, 50, -172, 115, -110, 108, -47, 113, -57, 55, -60, 55, -55, 223, -167, 168
[13:45:26][W][component:237]: Component remote_receiver took a long time for an operation (67 ms).
[13:45:26][W][component:238]: Components should block for at most 30 ms.
[13:45:39][D][lambda:071]: Received raw data with length 956
[13:45:39][I][remote.raw:028]: Received Raw: 573, -1918, 67, -275, 48, -1720, 47, -433, 57, -835, 58, -115, 110, -340, 225, -875, 107, -225, 218, -45, 60, -215, 105, -57, 108, -162, 160, -163, 115, -162, 165, -103, 57, -50, 50, -50, 53, -110, 160, -205, 107, -50, 58, -160, 47, -115, 
[13:45:39][I][remote.raw:028]:   45, -113, 115, -115, 222, -168, 50, -57, 108, -107, 50, -48, 52, -60, 58, -55, 377, -58, 107, -58, 60, -107, 55, -55, 225, -113, 162, -168, 220, -57, 163, -45, 57, -58, 110, -55, 107, -45, 163, -57, 55, -163, 112, -50, 218, -102, 215, -60, 113, -160, 
[13:45:39][I][remote.raw:041]:   165, -55, 55, -55, 60, -220, 165, -57, 280, -50, 58, -50, 57, -55, 115, -55, 55, -50, 58, -382, 113, -55, 60, -160, 50, -52, 60, -163, 47, -60, 55
[13:45:39][W][component:237]: Component remote_receiver took a long time for an operation (67 ms).
[13:45:39][W][component:238]: Components should block for at most 30 ms.
[13:45:52][D][lambda:071]: Received raw data with length 950
[13:45:52][I][remote.raw:028]: Received Raw: 623, -1928, 57, -265, 108, -632, 108, -1255, 60, -992, 168, -102, 113, -272, 318, -797, 108, -57, 50, -100, 63, -112, 220, -58, 50, -167, 50, -175, 223, -330, 57, -165, 118, -55, 382, -48, 50, -105, 210, -52, 60, -48, 107, -105, 105, -48, 
[... cut ...]
[13:46:17][I][remote.raw:028]:   -102, 218, -215, 165, -102, 68, -260, 112, -155, 373, -52, 103, -265, 57, -60, 165, -60, 108, -112, 50, -173, 55, -57, 103, -60, 115, -270, 170, -55, 55, -55, 165, -57, 165, -58, 107, -158, 62, -55, 55, -48, 67, -55, 225, -60, 168, -280, 60, -170, 57, 
[13:46:17][I][remote.raw:041]:   -108, 55, -105, 210, -107, 60, -45, 218, -70, 55, -55, 107, -168, 55, -215, 162, -100, 60, -263, 162, -160, 160, -58, 105, -117, 265, -373, 52, -50, 63
[13:46:18][W][component:237]: Component remote_receiver took a long time for an operation (66 ms).
[13:46:18][W][component:238]: Components should block for at most 30 ms.

`

swoboda1337, 2024-10-07:
Since Davis hops it might be harder to catch it. Should see ~32 values (alternating 52 and -52) at the start of the capture. Can change rx floor to like -50 if its really close to get less noise. I will add packet mode this week. Once that is there it will be much easier as it can look for the sync bytes automatically.

swoboda1337, 2024-10-14:
I added packet mode support. These settings are from the mqtt project. I think it should work:

sx127x:
  dio0_pin: GPIO26
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868066711
  bitrate: 19200
  sync_value: [0xcb, 0x89]
  payload_length: 8
  preamble_size: 4
  preamble_errors: 8
  preamble_polarity: 0xAA
  rx_bandwidth: 50_0kHz
  rx_floor: -90
  rx_start: true
  modulation: FSK
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());

Let it sit on one frequency for a few minutes. Can try these frequencies hopefully get something:

868066711
868297119
868527465
868181884
868412292

Don't need remote receiver for this, should be easier. If it works can check the crc offline.

mortenx, 2024-10-14:
i get an error message

In file included from src/esphome/components/sx127x/sx127x.cpp:1:
src/esphome/components/sx127x/sx127x.h:228:3: error: 'Trigger' does not name a type
   Trigger<std::vector<uint8_t>> *get_packet_trigger() const { return this->packet_trigger_; };
   ^~~~~~~
src/esphome/components/sx127x/sx127x.h:234:3: error: 'Trigger' does not name a type
   Trigger<std::vector<uint8_t>> *packet_trigger_{new Trigger<std::vector<uint8_t>>()};
   ^~~~~~~
src/esphome/components/sx127x/sx127x.cpp: In member function 'virtual void esphome::sx127x::SX127x::loop()':
src/esphome/components/sx127x/sx127x.cpp:203:13: error: 'class esphome::sx127x::SX127x' has no member named 'packet_trigger_'
       this->packet_trigger_->trigger(packet);
             ^~~~~~~~~~~~~~~
Compiling .pioenvs/davis1/src/esphome/components/wifi/wifi_component.cpp.o
*** [.pioenvs/davis1/src/esphome/components/sx127x/sx127x.cpp.o] Error 1
========================= [FAILED] Took 50.68 seconds =========================

swoboda1337, 2024-10-14:
Sorry, it built for me because I had another component bringing in that header. Fixed now.

mortenx, 2024-10-14:
i get this

In file included from src/esphome/components/sx127x/sx127x.cpp:1:
src/esphome/components/sx127x/sx127x.h:228:3: error: 'Trigger' does not name a type
   Trigger<std::vector<uint8_t>> *get_packet_trigger() const { return this->packet_trigger_; };
   ^~~~~~~
src/esphome/components/sx127x/sx127x.h:234:3: error: 'Trigger' does not name a type
   Trigger<std::vector<uint8_t>> *packet_trigger_{new Trigger<std::vector<uint8_t>>()};
   ^~~~~~~
src/esphome/components/sx127x/sx127x.cpp: In member function 'virtual void esphome::sx127x::SX127x::loop()':
src/esphome/components/sx127x/sx127x.cpp:203:13: error: 'class esphome::sx127x::SX127x' has no member named 'packet_trigger_'
       this->packet_trigger_->trigger(packet);
             ^~~~~~~~~~~~~~~
*** [.pioenvs/davis1/src/esphome/components/sx127x/sx127x.cpp.o] Error 1
========================= [FAILED] Took 16.23 seconds =========================

mortenx, 2024-10-14:
`
external_components:
  - source:
      type: git
      url: https://github.com/swoboda1337/sx127x-esphome
      ref: main
      
   
sx127x:
  dio0_pin: GPIO26
  #dio2_pin: 
   # number: GPIO32
   # allow_other_uses: true
  nss_pin: GPIO18
  rst_pin: GPIO23
  frequency: 868066711
  bitrate: 19200
  sync_value: [0xcb, 0x89]
  payload_length: 8
  preamble_size: 4
  preamble_errors: 8
  preamble_polarity: 0xAA
  
  modulation: FSK
  rx_floor: -60
  rx_bandwidth: 50_0kHz
 # rx_duration: 100ms
  rx_start: true
  on_packet:
    then:
      - lambda: |-
          ESP_LOGD("lambda", "packet %s", format_hex(x).c_str());

`

mortenx, 2024-10-14:
[17:42:44][D][lambda:069]: packet 4a7b5806effbf61f
[17:43:07][D][lambda:069]: packet b0644b793484c4aa
[17:43:16][D][lambda:069]: packet 61c7e89fe42f2faa
[17:43:23][D][lambda:069]: packet 7290d4620ca92da3
[17:43:36][D][lambda:069]: packet 10890081706db59a
[17:43:49][D][lambda:069]: packet d77be2e1f5352f5d
[17:44:03][D][lambda:069]: packet 0959903a42972849
[17:44:14][D][lambda:069]: packet 673fd6f5d3668b08

swoboda1337, 2024-10-14:
I think its a good sign its decoding packets though. The davis mqtt project seems to be for Davis Vantage Vue ISS and you are running Davis Vantage Pro 2? I wonder if the packet format is different. Maybe longer? I will google a bit more tonight.

swoboda1337, 2024-10-14:
Possibly fsk + hopping?? They don't go into detail. I will do some googling and see if anyone else has decoded this exact davis system before. 

I just pushed a fix, there was a small bug in the sync byte code.

mortenx, 2024-10-14:
now there is passed crc
i chek with sdr also , biger lines are davis packets
[image]

[19:52:21][D][lambda:069]: packet 4fc3027cd6638f75
[19:52:21][D][lambda:082]: crc failed: 19f4
[19:52:34][D][lambda:069]: packet 0a0000ffae0012da
[19:52:34][D][lambda:080]: crc passed: 485b
[19:52:46][D][lambda:069]: packet 0500007cb400f26c
[19:52:46][D][lambda:080]: crc passed: 4f36
[19:53:12][D][lambda:069]: packet 07000001a000f2e9
[19:53:12][D][lambda:080]: crc passed: 4f97
[19:53:38][D][lambda:069]: packet 050000fcb4001e60
[19:53:38][D][lambda:080]: crc passed: 7806

swoboda1337, 2024-10-14:
Ok those look legit now. I think we finally decoded some packets.

swoboda1337, 2024-10-14:
Just have to reimplement parseIssData, connect some sensors and possibly reimplement frequency hopping.

mortenx, 2024-10-14:
I have total rain, current rain , temp, humidity ,

mortenx, 2024-10-14:
[20:15:12][D][lambda:069]: packet 010000d4b700239f
[20:15:12][D][lambda:080]: crc passed: c4f9
[20:15:25][D][lambda:069]: packet 07000001a000f2e9
[20:15:25][D][lambda:080]: crc passed: 4f97
[20:15:50][D][lambda:069]: packet 05000002b40031fc
[20:15:50][D][lambda:080]: crc passed: 8c3f
[20:16:03][D][lambda:069]: packet 010000d4b700239f
[20:16:03][D][lambda:080]: crc passed: c4f9
[20:17:07][D][lambda:069]: packet 07000001a000f2e9
[20:17:07][D][lambda:080]: crc passed: 4f97
[20:17:46][D][lambda:069]: packet 010000d4bb008336
[20:17:46][D][lambda:080]: crc passed: c16c
[20:17:58][D][lambda:069]: packet 070000c1a00068e3
[20:17:58][D][lambda:080]: crc passed: 16c7
[20:18:37][D][lambda:069]: packet 010000d4b700239f
[20:18:37][D][lambda:080]: crc passed: c4f9
[20:18:50][D][lambda:069]: packet 070000c1a00068e3
[20:18:50][D][lambda:080]: crc passed: 16c7
[20:19:02][D][lambda:069]: packet 0a0000faa2000f4a
[20:19:02][D][lambda:080]: crc passed: f052

mortenx, 2024-10-14:
yes it is working 

[21:36:59][D][lambda:069]: packet 05000092b4004875
[21:36:59][D][lambda:094]: Davis: id a, battery 0, humidity 58.5
[21:36:59][D][sensor:094]: 'Davis Humidity': Sending state 58.50000  with 1 decimals of accuracy
[21:37:12][D][lambda:069]: packet 010000d4b50093ac
[21:37:12][D][lambda:086]: Davis: id 8, battery 0, temp 20.6
[21:37:12][D][sensor:094]: 'Davis Temperature': Sending state 20.55556  with 1 decimals of accuracy
[21:37:25][D][lambda:069]: packet 070000efa00048f0
[21:37:25][D][lambda:098]: Davis: id e, battery 0, data e00000f70500120f
[21:37:37][D][lambda:069]: packet 0a0000fdaa00ca08
[21:37:37][D][lambda:098]: Davis: id 5, battery 0, data 500000bf55005310
[21:37:43][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[21:37:43][D][esp32.preferences:114]: Saving 1 preferences to flash...
[21:37:43][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[21:37:50][D][lambda:069]: packet 05000092b4004875
[21:37:50][D][lambda:094]: Davis: id a, battery 0, humidity 58.5
[21:37:50][D][sensor:094]: 'Davis Humidity': Sending state 58.50000  with 1 decimals of accuracy
[21:38:16][D][lambda:069]: packet 07000028a000d776
[21:38:16][D][lambda:098]: Davis: id e, battery 0, data e00000140500eb6e
[21:38:41][D][lambda:069]: packet 05000012b400a479
[21:38:41][D][lambda:094]: Davis: id a, battery 0, humidity 58.4
[21:38:41][D][sensor:094]: 'Davis Humidity': Sending state 58.40000  with 1 decimals of accuracy
[21:38:54][D][lambda:069]: packet 010000d4bd005362
[21:38:54][D][lambda:086]: Davis: id 8, battery 0, temp 20.6
[21:38:54][D][sensor:094]: 'Davis Temperature': Sending state 20.55556  with 1 decimals of accuracy
[21:39:07][D][lambda:069]: packet 0700004ca000fb10
[21:39:07][D][lambda:098]: Davis: id e, battery 0, data e00000320500df08
[21:39:20][D][lambda:069]: packet 0a0000d0a000fe09
[21:39:20][D][lambda:098]: Davis: id 5, battery 0, data 5000000b05007f90
[21:39:33][D][lambda:069]: packet 05000012b400a479
[21:39:33][D][lambda:094]: Davis: id a, battery 0, humidity 58.4
[21:39:33][D][sensor:094]: 'Davis Humidity': Sending state 58.40000  with 1 decimals of accuracy

mortenx, 2024-10-14:
what is id 5  and id e

swoboda1337, 2024-10-14:
Look here on line 650 and 699: https://github.com/dcbo/ISS-MQTT-Gateway/blob/master/src/main.cpp

Both are rain

mortenx, 2024-10-14:
"battery 0" is when battery is ok  and "battery 1" is when battery is bad

rain rate is ok 
(difference between last two clicks in seconds  )
one clik is 0.2mm so rate in hour is 3600  / data * 0.2

`            } else if (msg_id == 0x5) {
              // outside rain_rate (MSG-ID 5)   
              float rain_rate = (float) 720 / (data[3] + ((data[4] & 0x30) * 16));   
              ESP_LOGD("lambda", "Davis: id %x, battery %x, rain_rate %.1f", 
                       msg_id, battery, rain_rate);
              id(davis_rain_rate_id)->publish_state(rain_rate);`

but cant figure out what to do this case 0xe:  

rain counter it is counting tipping bucket clicks 0-128 and starts over

`            } else if (msg_id == 0xe) {
              // outside rain_total (MSG-ID e)      
              float rain = (float) ((data[3] & 0x7F )  );
              ESP_LOGD("lambda", "Davis: id %x, battery %x, rain %.1f", 
                       msg_id, battery, rain);
              id(davis_rain_id)->publish_state(rain);`

[39 more replies]