esphome/feature-requests #2889: Support for custom CA certificates

danierukun, 2024-09-18:
**Describe the problem you have/What new integration you would like**

It would be nice to have a configuration option to include custom Certificate Authority certificates in the ESPHome container.

**Please describe your use case for this integration and alternatives you've tried:**

I need to be able to add a custom CA certificate to the ESPHome container so that it may connect to my local Git server. I currently do this by `docker cp` the cert into the container and then `update-ca-certificates` in the container shell.

**Additional context**

I have a homelab that does not have a public domain name. I have ESPHome as an addon in my Home Assistant instance. I also have an internal Git server secured by a TLS cert of my own CA. If I want to use ESPHome's package feature I need to be able to trust my own CA so that the container may checkout my code from my Git server and complete firmware builds.

noseshimself, 2024-10-13:
If this is added there should be an API for automatic installation of new keys and certificates without user intervention. Certbot-like tools are usually running on systems that insist on exporting certificates but should not permit incoming access.

MadnessASAP, 2024-12-02:
I would like to +1 this, I have an external component living on a private git server. The esphome requires manual intervention as above to play nice.

oldboys92, 2025-01-28:
for the sake of keep it simple, this could be easily solved by mounting ca-certificates from the docker host. at least as an option to enable this via add-on configuration would be awesome.

people who decided to be cloud free, would love to use their own CA instead of depending on letsencrypt online service. That way one can fully build everything in their homelab / homelan. I use for example `step-CA` as ACME service which runs on a rpi2. I take the extra effort to distribute my CA certs on local devices, but is a hassle in HA and almost each HA addon to add own CA certs.

Cronvs, 2025-08-03:
Why not take it all the way and support integrated acme clients for esp devices that want to host something, I know of https://sourceforge.net/projects/esp32-acme-client/ and https://www.oryx-embedded.com/products/CycloneACME.html that already implement ACME targeting ESP32 and microcontrollers.