esphome/feature-requests #2890: Add support for TMOS (Thermal Metal Oxide Semiconductor)  PIR sensors.

GinAndBacon, 2024-09-18:
**Describe the problem you have/What new integration you would like**

TMOS PIR sensors are relatively new, currently M5stack is the only one that sells one but I expect more manufacturers to follow. Below is a brief summary of how the technology works but essentially it uses IR to detect temperature allowing it to sense humans and their movement. It's essentially like an mmwave sensor but with a smaller detection area and won't be triggered by anything that is moving, just heat.

I had created a thread in the HA forums for feedback from anyone that has used one but then I remembered it would need to be added on the ESPHome backend as a supported device as this isn't a normal PIR sensor. Below is a link to that post and a link to the m5stack TMOS PIR unit (which is also in the HA link) Currently M5stack also has a micropython example which I have also linked.

HA forums post
https://community.home-assistant.io/t/has-anyone-used-tmos-pir-and-does-it-work-as-advertised/771825

M5stack TMOS PIR sensor
https://shop.m5stack.com/products/tmos-pir-unit-sths34pf80

Micropython example 
https://uiflow-micropython.readthedocs.io/en/latest/units/tmos.html

`TMOS PIR Unit is a high-sensitivity infrared sensor unit for presence and motion detection. It uses the STHS34PF80 chip solution. It communicates with M5 devices through I2C (default address: 0x5A). Its working principle is based on the black-body radiation principle described by Planck’s law, which allows it to monitor ambient temperature and detect human presence and motion. The sensor can distinguish between stationary and moving objects and has an 80-degree field of view, providing a wide detection range. Additionally, it supports adjustable sampling frequency and gain modes to meet different application needs. It is suitable for applications such as alarm systems, smart lighting, and occupancy detection.`

**Please describe your use case for this integration and alternatives you've tried:**

Without zones mmwave is to finicky and these seems like a good alternative for things when your in a certain area like in bed or the bathroom. If I’m in bed and get up I can do whatever with HA. With mmwave if my ceiling fan on that’s just not possible without zones when using mmwave. That and it’s much, much cheaper outside DIY.

**Additional context**

[More info](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.st.com/content/dam/AME/2023/sensors-converge-2023/pdf/sensors-converge-23-presence-detection-tmos-tof-gogonea-uphadhyay.pdf&ved=2ahUKEwiawbKopsqIAxVIG9AFHQ0AIxAQFnoECCAQAQ&usg=AOvVaw0y6gCIPygUhCl0XUp2QyjB)

[Mouser article about TMOS PIR 
](https://www.mouser.com/blog/overcoming-limits-pir-sensors-tmos-technology)

GinAndBacon, 2024-09-18:
Just noticed that m5stack has an Arduino examples on git. Does this mean it may already be possible? I'm just unsure what has to be on the ESPHome GitHub end as usually it lists specific sensor modules the user defines in their config which leads me to believe the STHS34PF80 sensor would have to be added to the ESPHome library of devices.

https://github.com/m5stack/M5-STHS34PF80

lblokland, 2025-01-02:
Anyone who can help me get this sensor up and running in ESPHome, maybe we can team up!

GinAndBacon, 2025-01-12:
Finally got around to burning the M5stack uiflow 2.0 project as I had a CoreS3 which didn't make a great voice assistant. Works best when it's focused on the chest area. One pick in the clamp is roughly 2 feet away. When holding it above me it gets 3 to 3 1/2 feet, however long my arm and shoulder on. I am going to try mounting it to the ceiling, just haven't gotten the chance. Considering the price, I certainly can't complain and it's working as advertised. I put a blowtorch on about a foot in front of it and presence got over 3K. Typically 200 to 500 is the average when focused on the chest area. It works on other areas also, just not as reliant. I can post the python code from uiflow but anyone willing to make an account (100% free) you can just open the project and grab the code. Just search for TMOS under project discovery. 

[image]
[image]
[image]
[image]
[image]
[image]

GinAndBacon, 2025-03-05:
Stale

closed 2025-03-05 by GinAndBacon

daiyyr, 2025-04-08:
Sorry why do we close this issue? The I2C motion sensor TMOS PIR is still not supported in ESPHome. So currently ESPHome only support GPIO PIRs.

ianmills122, 2025-04-08:
@daiyyr agreed this has not been solved yet so it should remain open.