esphome/feature-requests #2892: Add support for Yaml packages / Templates for dashboard

BeatLink, 2024-09-22:
**Describe the problem you have/What new integration you would like**

It would be great if the ESPHome dashboard could support non device yaml files for templates and packages. These would not be treated as devices (with the associated UIs and controls) but would only be available separately to be imported into actual device configs. The only actions for these yaml files would be to validate them. 

**Please describe your use case for this integration and alternatives you've tried:**

I currently use the dashboard to manage and update my esphome components. To make things simpler I have utilized templates and packages to modularize the configurations, minimize duplications and consolidate common settings for easier management. However, to edit them from the dashboard I have to add them as fake devices, which the dashboard then (expectedly) throws errors when they dont have all of the required device configurations and settings. 

**Additional context**

BugsBunny1403, 2024-11-14:
Manage package and subdirectory can be very useful. It can be manage by an additionnal button like secrets or update all.

BeatLink, 2025-11-28:
Any updates on if and when this can be implemented?