esphome/feature-requests #2893: Read and control Gecko spa controller (IN.YE)

dala318, 2024-09-23:
**Describe the problem you have/What new integration you would like**

The [Gecko IN.YE](https://www.geckodocs.com/yseries) Spa main controller has the option to be controlled via an external [IN.Touch2](https://geckoalliance.com/products/in-touch-2/) (official link broken, a random [example](https://www.amazon.com/GECKO-Touch-Internet-Module-Version/dp/B08L3TMHS9) ) WiFi connected device. Is an extensive [discussion ongoing in the Home Assistant feature-request forum](https://community.home-assistant.io/t/gecko-in-touch2-integration-spa-wireless-remote-control-module/150764) but it's a mix of hijacking and reverse-engineering the C0 interface and integrating the OEM IN.Link2 device. So let's try to keep one discussion open for the hijacking part.

**Please describe your use case for this integration and alternatives you've tried:**

Find a way to interact with the IN.YE spa controllers without having to buy an IN.Touch. RS485 on the C0 port seem like the best bet at the moment but looking around on the main-board there are also I2C mentioned on some connectors.

**Additional context**

Useful links from the other forum.
<https://community.home-assistant.io/t/gecko-in-touch2-integration-spa-wireless-remote-control-module/150764/102>
<https://community.home-assistant.io/t/gecko-in-touch2-integration-spa-wireless-remote-control-module/150764/143>

alexpalmer, 2024-09-25:
Following this... If anyone figures out the RS485 pins (GND, A, B), I'm willing to try to sniff some data.

rct, 2024-09-26:
There's a lot of info scattered around.  Here's some useful info on pin probing work that agittins had done over in the geckolib repo: https://github.com/gazoodle/geckolib/issues/8#issuecomment-876631237

dala318, 2024-09-27:
Thanks, had missed that thread, more or less a kick-start to my planned investigations.
Right now I'm stuck on getting a proper cabling to extend the ports to a decent working height (recessed spa in deck and can't do the probing laying up-side-down), but now realize I might actually want a break-out cabling to connect between the main-board and keypad.

rct, 2024-09-27:
It might be worth trying to get a hold of a broken spa side gecko unit. If nothing else for the cable. You linked to t[he post on the Hass forums with the google photos of the internals of the gecko unit](https://community.home-assistant.io/t/gecko-in-touch2-integration-spa-wireless-remote-control-module/150764/143).  Maybe the OP would part with that, or could be convinced to map out the cable and get some closer pictures around the 8 pin connector.

agittins, 2024-09-28:
Howdy - wow hadn't realised it had been 3 years since I did that initial probing about! I haven't got further since then, however I have been donated a faulty in.touch unit which means I should be able to gather more authorative info on the pinouts.

Re making connections - the header pins are standard IDC / DuPont / PCB Header pins in a 4x2 layout. My spa pack has a "port cover" thingy to fill the empty comms ports - it's just a plastic shell without keying on it so it will fit into any empty port to protect it from water ingress.

The shell is hollow, and you can actually jam a 2x4 header block right into it. I drilled a 6mm hole in the outer plate, threaded a cat5 cable through it, and terminated a 2x4 block on the end, which then pushes into the front end of the shell. This gives a solid, easy connector to the spa pack that can be used on any of the control ports. One could put another 2x4 at the other end and you'd have an extender / breakout / interception cable pretty easily.

I'll take photos of the plug I made to make it easier to understand - if you have one of the spare port covers it's very simple, but explaining it makes it sound complicated! 😅 Right now though it's cold and raining, and I'm pretty sure the cable is sitting under the hot tub outside 😆

agittins, 2024-09-28:
Whoo! Good news - I dug out the dead in.touch-2 unit that was donated to me and I think I have a pretty confident answer on pinout and protocol!

[image]

I traced out the pins and most appear not-connected in the in.touch-2, (which is great, actually) and the only ones that seem connected are GND, 12v supply, SDA/SCL for i2c and a 5v line which might be able to provide supply, but might just be an `ENABLE` line, as the in.touch-2 only uses it for ESD clamping the i2c and for turning on it's 12v VREGs, it seems.

This is the relevant part of the blended photos of the front and back of the in.touch-2 pcb I used to help trace, and identify the mcu etc:
[image]

- The i2c pins are connected to *two* sets of i2c pins on the mcu, just paralelled. I don't know why, maybe their code switches to the second bus if the first one dies of ESD or something, or maybe they're using the other pair of pins for generating interrupts to help timing, no idea.
- I'm a bit surprised at the complexity of the supply section (everything right of the P1 header). I haven't traced it, but the SHAB looks like a 750mA step-down regulator, the PCN chip *might* be a bigger reg, not sure. There are parts with that code that are wimpy little vregs (like 100mA or so) and others that are 5A!
- The voltage rails I actually went out and measured on my hot-tub since there was no way to infer them from the board. Note how the 5v3 rail on pin 4 runs to the TVS ESD protection CR1 and a bypass cap, then runs up to the middle pin on Q1 - which might be a fet or a "voltage detection chip" going by what looks like P0n or P0 then a sideways-C.

So, I think this means we have enough info that for the CO port at least, people could hook up something to sniff i2c and see if they get anything useful coming through.

I suspect that there is probably handshaking that needs to go on first, so just passively sniffing may or may not get anywhere, but it's worth a shot. Anyone with a working in.touch-2 has a decent shot at capturing the entire protocol though, so I think this is definitely a viable project when someone with the time also has the resources :-)

My next steps are...

- See if I can get the in.touch-2 I have working. It was known faulty originally (hence why I now have it), but my initial poking around on it found that the "inside" unit was trying to contact a nonexistant dns name, so it was stuck in a boot-loop trying to initialise. I haven't hooked it up to my hot tub yet, but when I was first messing with it creating a faked dns definitely made it... "happier" in that it completed some sort of handshake (ping2, for anyone playing along) with the gecko server, so maybe the unit will work. If so...
- I plan to set up a passive i2c listener to sniff the traffic between the spa pack and the spa-side of the in.touch. I'm hoping I can use an esp32 and do it over wifi, because it's a pain hacking outside at midnight :-) For low-level stuff there's a sigrok sink for esp32: https://github.com/Ebiroll/esp32_sigrok and for proper i2c decoding and logging there seems to be https://github.com/WhitehawkTailor/I2C-sniffer - but both projects have been pretty quiet of late, and the sniffer one doesn't have built-in wifi sending but the `main-send.cpp` version has part of that plumbing present.

Anyone who knows of an easy, via-wifi way to sniff i2c please let me know! I've got esp32's, and a raspberri pi Zero W (the old one) which might be a better option, as it can probably bit-bang a lot harder than the esp32.

Of course if I *can* fix my dead in.touch-2 I won't necessarily *need* a native esphome version myself, but I'd love to see it happen because $300 is robbery, hacking is fun and the dead unit was sent to me by someone for free in the hopes of making this happen for everyone 🚀 

This is the full board of the spa-side pcb of the in.touch-2. I will probably set up a repo to upload the full-res versions of this and my notes, but wanted to get this out here in case people are ready to jump on the next step!
[image]

agittins, 2024-09-28:
Quick and dirty pics of how I made a connector using the empty port covers and a 2x4F pcb header connector. 
[image]
[image]
[image]

rct, 2024-09-28:
@agittins - This is all great info!  Thanks for sharing your work.  Nice job on building that cable!

I have a working in.touch2 unit. I had posted on 2023-11-18 [about the DNS problem in the Hass forum](https://community.home-assistant.io/t/gecko-in-touch2-integration-spa-wireless-remote-control-module/150764/181).  

My unit didn't stop working with Home Assistant, I only noticed that the clock on the spa was wrong. However my unit was already fully "provisioned". I don't know whether it was a missed update/corrupted firmware or something.  I suspect/hope you should be able to resurrect that in.touch2 pair.  Very curious to hear whether that fixes your unit.

rct, 2024-09-28:
@agittins - A possibly tangential thought if you want to do some hacking from the comfort of your desk.  The [microchip RF board mrf89xam9a is SPI](https://www.microchip.com/en-us/product/mrf89xam9a). Might be interesting to get an SPI sniffer going attached to the network side in.touch2 unit's RF board, particularly If your unit ever seems to sync with your spa.  Some clues about what's going over the RF connection might yield clues about how much smarts are in the in.touch.

nickrout, 2024-09-28:
A spa manufacturer went broke here recently, and I got 3 in.touch units cheap from the liquidation sale. Happy to sacrifice one to the cause. Either by taking it apart or sending to an expert to investigate. The latter would probably be more productive.

agittins, 2024-09-29:
@rct 
> @agittins - A possibly tangential thought if you want to do some hacking from the comfort of your desk. The [microchip RF board mrf89xam9a is [...]

Yes that could be helpful at some point. My main focus is on being able to interface directly with the controller pack though, so that folks don't need an in.touch at all. But yes, if we can't sniff the i2c it's a next-best-step.

@nickrout 
> I got 3 in.touch2 units cheap from the liquidation sale.

Nice score! I wouldn't have use of one yet myself (on the assumption that my current issue is the controller's firmware) but if my in.touch turns out to be truly faulty maybe we could work something out (being we share a hemisphere!)

@rct 
> my work around of having a host table entry on my router for intouch that returns the IP address of intouch2 is still needed.

Thanks for the tip on the dns - I had forgotten who it was I should credit for that discovery! That got me from the home unit flashing RED through flashing GREEN then finally flashing BLUE, indicating that everything was good except for its connection to the Spa transmitter.

This LED status chart from the touch2 techbook is the only comprehensive status table I've seen, so worth popping in here I think:

[image]

Worth mentioning that when the home unit is in pairing mode (fast-flash YELLOW) it still lights the LED based on network status as well, so you get some awkward combinations of colour!

For the home unit, I think a more accurate diag is:
- Any yellowish (red+green) fast flashing: pairing mode in progress
- Any slow flashing: no comms between spa sender and home receiver
- Red: no DHCP reply received, check basic local networking
- Green: LAN ok, searching for Gecko server. If stuck here, rct DNS fix might be required (alias intouch.geckoal.com to intouch2.geckoal.com)
- Blue: comms established to Gecko server OK.

For the Spa sender unit:
- Any flashing: No connection to home receiver unit
- Yellow: pairing mode
- Red: No comms to pack controller. Software version of controller too old, power cycle required, or possible hardware fault.
- Blue: Normal ops.

My spa sender doesn't flash yellow at all, only red. So my guess (hope) is that it will only pair *after* it has good comms to the controller, and I just need a firmware update for my in.XM2.

Given the RED sender led, my next step is to try and get a software update from Gecko. Fingers are crossed, hopes are not all that high. The blurb they have on in.stik (their update device, a thumbdrive-like dongle for the CO port) is that they are keyed to a specific spa pack's serial. However, given that they also say that dealers can update spas during provisioning etc (and resellers indicate the same) my guess is that they are simply keyed to a model's revision level, and that there must be an in.stik programmer that dealers use, or they simply get sent a batch of stiks for each model at each firmware release (which would make sense given the in.stiks seem to have a durable label/tag attached to them).

I'll contact our local pool mob too, maybe they'll have a suitable in.stik already.

agittins, 2024-09-29:
Gah! Oh no... after taking a **proper** look at the labels on my unit, it turns out I have.... an in.XM1 😱 

[As far as I know](https://www.spaparts.com.au/gecko-aeware-in.touch2-wifi-smartphone-spa-control) these are *not* compatible with the in.touch-2, regardless of firmware. 😭 

This means that for my part at least, my hopes of reverse engineering the in.touch2 protocol are almost certainly dashed. I'm sending an enquiry to Gecko anyway to find out if there's any hope of an update, but I expect it's pretty unlikely.

Hopefully though, the pinout and board analysis will help someone else sniff the i2c data and work out how to emulate it with an esp32 or similar. Given it's just i2c this should be completely possible, given enough time to work out the protocol - but it will have to be done with an existing, working in.touch because there will be master/slave determination as well as address, handshaking etc to work out. Once it's worked out though it should be possible to replicate without any in.touch.

For my part, I'll now switch to reverse engineering the keypad and IR interface instead! I have a crusty old remote for the spa (pretty sure it's [this model here](https://www.spaparts.com.au/aeware-irmt-4-universal-infra-red-floating-remote), now reduced from a bargain AU$440 to a run-out special of AU$110!), so as long as corrosion (both spa water and battery electrolyte) hasn't completely destroyed the board it *should* be fairly easy to record the IR codes out of it.

Getting the IR codes will allow an esp8266 or esp32 to **control** the spa functions (either by sending IR via LEDs or perhaps by direct connection to the IR port), but not monitor the status. It's a start.

Reverse-engineering the in.K600 keypad might allow full two-way control. If anyone has a spare / dead keypad I'll gladly accept ~~victims~~ tributes for the cause (I'm in Australia though, shipping might be exorbitant).

My in.K600 keypad seems to be the "MD" or "Menu-Driven" variant, which has a full dot-matrix display and menu system, versus the in.K600 "SI" or "Streamlined Interface"(!) which has a custom LCD with a series of status icons and a big 7-digit display. The fact that two very different keypads (with the same model number!) exist suggests to me that it's likely the comms is a low-level protocol and all the screen drawing etc is handled at the display unit itself.

agittins, 2024-09-29:
Oh, and just noticed in the techbook for both XM1 and XM2 that the 5v rails seem to be split, one rail for C1/CO and one rail for C2/IO, and each is rated for a max of 125mA.

The 12v rail seems to be common over each port, and the techbook states the total combined load on the 12v rail is 300mA max.

This will be worth keeping in mind when designing a module to hang off one of the ports, as the power budget looks pretty tight. Options might be to just include a decent-sized cap on the power rail if the average load will be low enough, or even include a LiPo cell and controller so that the controller can trickle-charge the cell, or using the 12v rail with an efficient DC-DC module. My esp32 modules look to draw about 110mA on average while idling, and will peak over that during wifi, I am sure. I guess this is why the in.touch-2 only uses the 5v rail for i2c ESD protection, and draws all of its power from a (moderately complex) on-board step-down circuit.

Another option is that the L1 light port is rated at 9.5V **AC** at 1Amp. One could run off a lipo and then get the controller to turn on the light periodically to power a charging circuit! 😅 Beware that isolation etc might be a big issue there - no idea what the internal power supply looks like and what constitutes "ground" in reference to the 9VAC supply. Given it's a human sous vide machine (not sui-cide machine) it will almost certainly be referenced to earth, but still..

Anyway, here's the bit from the techbook about the low-volage load limits:
[image]

Another point of interest is that the IR receiver pin is stated as being available on every low-voltage port except L1 (Light) and RH (Remote Heater), so this might make it very easy to intercept or inject IR signals. Also I've noticed that some keypads have the purple CO connector on them while others have the orange C1/C2 connector. I wonder if the ports are actually identical, or maybe the controller only expects certain device addresses (assuming they're all i2c) on certain ports? For that matter we don't know yet if the controller acts as a master or if it acts as an i2c slave on all or some ports.

rwskinner, 2024-11-25:
I'm not sure if any of this is remotely helpful but I'll share what I have.

I have the original XM1 and it has an accessory device called in.watch that plugs into the CO port, then sends a 433mhz signal to a remote receiver.  The remote receiver then displays water temperature and a few diag codes.  You all may already have this but this is what I found when disassembling the transmitter, I found it's using 4 wires.  5+, Grd, SDA, SCL.  (I2C)
The pin out I have is:
1 - N/C
2 - SCL
3 - SDA
4 - 5vdc
5 - N/C
6 - N/C
7 - N/C
8 - GRD

I think the XM2 has the 485 populated on the board near the CO port, and the XM1 uses only I2C.  The reason I say this is because none of the add-on Wifi modules will work on the XM1 while they work fine on the XM2 and Y series.  When I contacted Gecko they said basically the same thing.

I haven't done any further testing.  I need to get a long PS2 cable to extend this where it's easier to access.  I can try and start simple with a logic probe and protocol analyzer to see if any of the byte data makes sense.  My problem is that I'm already working two full time jobs!

DuckManQuack, 2024-12-02:
I'm also in this same "hot tub" lets say . I have the IN.XM1 and the K600 (With the segment display not menu) would love to see if you guys make any more progress on this! Home Assistant awaits spa control!
Unfortunately I don't have any spares I can donate to the cause but if moral support is accepted, I can supply that in 22gal drums!

rwskinner, 2024-12-02:
To determine which is master or slave I would think on the CO port. the Spa Pack would be the slave and the remote devices would be the master right?

If the remote device  (In.Watch) is powered up and I2C pins monitored, but not connected, wouldn't we see a request sent to the Spa Pack for the requested data, if the remote device was the master  (Start, Address, and Data addresses requested)?  I don't mess with I2C much but I do Modbus and CAN/J1939 a lot so forgive me if it's a dumb question.

>>For that matter we don't know yet if the controller acts as a master or if it acts as an i2c slave on all or some ports.

dccourt, 2025-02-12:
Some really interesting work so far here, and the pinout confirmations are really useful.

When the ink1000 touchscreen on my swim spa started to delaminate, I was able to get a replacement under warranty, and hold on to the old unit.  So I have a C0 (I think) client device that's functional and easy to test with, take photos etc.

I'd been toying around with the interface under the assumption that it was RS485, and getting confused as to why I wasn't seeing any paired signals on any of the pins.  Knowing now that it's probably I2C instead is really helpful.

With it being winter now, it's hard to observe live comms from the spa itself, but I have been powering up the old ink1000 unit in isolation to try to see what it does during boot time.  It certainly does seem to initiate some comms on the data lines, but while looking for RS485 I wasn't getting very far.

Hopefully I'll be able to attach a primitive I2C sniffer in the next couple of weeks, and after that I should have a more sophisticated protocol analyzer arriving, and I should be able to do some proper analysis with sigrock.

Once the basic parameters of the comms are understood, and as the weather gets warmer, it should be a lot easier to really get stuck into some reverse engineering.  I'll report back as soon as I have anything useful.

agittins, 2025-02-12:
@rwskinner my gut instinct would be that the spa pack might be master, and it would scan the bus(es) at boot to init/discover any peripherals it supports. But you could certainly be right, and also right I think that the in.touch alone might reveal itself under some bus sniffing.

@dccourt awesome! Very interested to keep up with your progress!

DuckManQuack, 2025-02-18:
I've just come across this https://github.com/etienne4313/spa_bluetooth_controller I'm not too sure on how this is actually connected to the in.xm though, I couldn't see any wiring diagram.

agittins, 2025-02-18:
It looks like it's wired directly to the buttons on the control pad (or via relays) given that the outputs seem to be mapped to physical pins rather than some sort of serial, and the functions that turns the main relay on and off are identical, and work by pulsing the output for 500ms.

That's just from looking at the code though, i could easily be misinterpreting what I'm looking at.

dccourt, 2025-02-24:
I finally got a logic analyser hooked up to the in.k1000, unplugged from the spa pack.

It's looking a lot like this is an I2C slave - from power up until "there's an error communicating with the spa pack, check your connections" I see only the following brief activity on SDA and SCL, the rest of the time they're high:

[image]

dccourt, 2025-02-24:
Bah.  Verified that my esp8266-based i2c scanner works via the logic analyser, but it doesn't find anything at addresses 1-127 on the i2c lines on the in.k1000.

I suspect there's some additional handshake on some of the other pins, but I'm not seeing anything particularly obvious being initiated by the in.k1000, beyond the startup pulse:

[image]

At this point, I think I need to wait until I can sniff the live lines, which might be a month or two.

tommievanlent, 2025-02-27:
I just stumbled upon this great thread after having the same idea. I have an older Gecko M-Class spa.
It has an I2C connector, but after reading this thread, I think the best approach is to read the lines between the mainboard and the controller display. I hope the weather is okay this weekend so I can connect my logic analyzer to some pins.

I emailed Gecko support and kindly asked if they could provide information on interfacing with the I2C, but they politely declined my request.

I will keep you guys posted on my progress.

tommievanlent, 2025-02-28:
Well. i got my mainboard out of the spa and on my workbench, i see some continuous data but cannot make any scense of it..

[image]

nebhead, 2025-03-02:
Watching this thread eagerly.  I have a Gecko Yj controller in my spa:

[image]

It appears to have CO port as well, so I'm wondering if this will lead to something that I can also use.

[image]

I've just done some continuity testing between the two ports and it seems that they at least share some of the same pins.

[image]

Top is CO (as marked on the silk screen), bottom is C. 

This is the result of my continuity testing: 

[image]

Seems to line up with the pinout in the image above?

dala318, 2025-03-03:
Thanks for your analysis, and i hope you are correct since it would mean there is no need for break-out y-cables to sniff or inject data between the control-board and key-pad.
Now pin-numbering is a head-ache and whether you look at the pins or the connector. But what worries me that something is wrong in your aligning of bins and connector in last image is that the GND (8) pin is not shared. Also I see something that look like pin-numbering printed on the board which would invert some of the aligning.

[image]

dala318, 2025-03-03:
> Well. i got my mainboard out of the spa and on my workbench, i see some continuous data but cannot make any scense of it..

@tommievanlent, your image of signal-levels look like it's coming from PulseView, did you try running any of the decoders on it? Or do you have the raw data stored and can share for others to try. (I attempted to send your image to ChatGPT for decoding but ran out of quota before an answer was given :-) )

nebhead, 2025-03-03:
> But what worries me that something is wrong in your aligning of pins and connector in last image is that the GND (8) pin is not shared. Also I see something that look like pin-numbering printed on the board which would invert some of the aligning.

Right!  I was worried about that too hence the open question.  But I think you are right about the numbering, here's a slightly less blurry view.  I took a couple of pictures yesterday but have buttoned it all up since.  Can't have it out of commission or I will get in trouble with my partner. :)  The silkscreen definitely seems to indicate the orientation you have pointed out.  

[image]

All that said, here is a picture with the shroud installed that gives some clues to orientation of the connector.  The key bits are on the right side.  

[image]

tommievanlent, 2025-03-13:
> [@tommievanlent](https://github.com/tommievanlent), your image of signal-levels look like it's coming from PulseView, did you try running any of the decoders on it? Or do you have the raw data [...]

I gave up and went the arduino way. I removed the main board and gone fullt DIY

[image]

dccourt, 2025-03-21:
I got a brief opportunity to do some protocol sniffing on my in-situ In.YE controller yesterday.  I had my logic analyser connected to what I believe is the in.touch port (C0?) while I interacted with my in.key controller plugged into the other port (C1?).  Not certain about port names because I was using a cable I'd fitted a few months ago, then re-sealed the controller unit, so I didn't actually look inside the In.YE this time.

Anyway, I didn't see any traffic on any of the 5 non-power/non-GND pins with that set up - so looks like I'll need to create a man-in-the-middle patch cable to sniff on the in.key's line directly.  Conceivably once the protocol is understood, I'll be able to switch back to the other port, although I think it would also be acceptable to continue the man-in-the-middle approach if necessary.

nebhead, 2025-03-21:
> Anyway, I didn't see any traffic on any of the 5 non-power/non-GND pins with that set up - so looks like I'll need to create a man-in-the-middle patch cable to sniff on the in.key's line directly. [...]

Just speaking for myself, totally fine with man-in-the-middle kind of situation.  Not afraid to do some splicing and soldering. :)

rwskinner, 2025-03-21:
From me decoding industrial protocols MIM is typically the best way to go anyways.  I mainly did Modbus, CAN, J1939.  I hope you have some luck.  My spa install is fairly hard to get to and in regular service.  I know enough about I2C to be dangerous (not much)

DuckManQuack, 2025-06-16:
@agittins Did you even manage to get anywhere with your digging? It's time to do my annual maintenance (I have to replace a pump) But I'm going to purchase an IR sensor for the in.xm. (I'm assuming its the same IR sensor you have installed.)

I'm thinking even if I can send the "pump 1 on" IR signal 3 times you could probably assume that the pump should be on and then have that state change in HA. Then the temp feedback could be just a WIFI or Zigbee temp probe?

agittins, 2025-06-19:
No news to report, sorry - haven't got around to trying things out, yet - but I do have one of those little logic analyser thingies sitting on my desk, mocking me (along with several dozen other projects piled on my desk!).

Looking at the manual for the IRMT-4 remote control, it looks like the pump controls are "toggles", ie they use the same button to cycle on or off. Presumably the controller will "debounce" the signal a bit, so perhaps sending a continuous stream of the "pump1" signal will reliably toggle it once, but if you send the signal with any delay between them there's probably some point at which it will consider them separate actions.

But yes, if you can get it to reliably trigger, you should have a solution.

Only downside then is you don't really "know" which state it's in. If you have some form of power monitoring on that circuit available you could infer the state by whether the power draw increases or decreases, or you could put a vibration sensor on the tub or add a microphone to your esp32 and use volume as a hint to whether the pump is running or not.

agittins, 2025-06-19:
Ok, so you inspired me to pull out the old remote and see what codes it sends. This should make it easy for you to set up an esphome device to send codes. You could use it with an IR LED and the receiver, OR.... I reckon you *could* almost certainly wire to the IR line on the connector directly. If it doesn't work (or works weirdly) then toggle the `inverted` option on the remote_transmitter pin definition.

### Gecko/Aeware IR Codes
- Uses NEC codes. Device address is 24966 (0x8661 or 0x6186 depending on endian-ness).
- Table below created using an IRMT-4 remote and esp8266 esphome (post-2024) NEC decoder.
- No known spa functions for shifted press, but included for completeness since it does transmit and has the same "address". Possible that these will be interpreted as digits 0-9, but unknown if spa-pack uses that at all.
- Dashes indicate no code was sent (shift-power, 7_rec) when in SPA mode.
- Codes sent in other "modes" (tv, cd, dvd vcr etc) ignored since they are programmable.

| Button             | Spa Function | Address | Command | Shift |
| ------------------ | ------------ | ------- | ------- | ----- |
| POWER              | ???          | 24966   | 48960   | -     |
| 0_VOL_PLUS_PUMP1   | Pump 1       | 24966   | 47430   | 59670 |
| 1_CH_PLUS_PUMP2    | Pump 2       | 24966   | 65280   | 60690 |
| 2_VOL_MINUS_LIGHT1 | Light        | 24966   | 47175   | 60435 |
| 3_CH_MINUS_BLOWER  | Blower       | 24966   | 64260   | 60945 |
| 4_BACK_PUMP3       | Pump 3       | 24966   | 61965   | 61200 |
| 5_AUX1             | AUX1 ?       | 24966   | 63240   | 61455 |
| 6_FWD_AUX2         | AUX2 ?       | 24966   | 62220   | 59160 |
| 7_REC              | ???          | 24966   | -       | 59415 |
| 8_TVAV_STOP_LIGHT2 | Light 2?     | 24966   | 62985   | 60180 |
| 9_PAUSE_MUTE       | ???          | 24966   | 64005   | 59925 |

So in esphome you'd set up the ir transmitter:

```
remote_transmitter:
  pin: GPIOXX
  carrier_duty_percent: 50%
```

And use something like this on a button action etc:

```
on_...:
  - remote_transmitter.transmit_nec:
      address: 24966
      command: 61965
      command_repeats: 1
```

If you are connecting directly to the spa-pack instead of using IR LEDs, then I suspect you'd want to set the `carrier_duty_percent` to `100%`, as the IR receiver module *probably* does the demodulation to remove the carrier before the signal gets sent on the port - but maybe it doesn't, so try both!

DuckManQuack, 2025-06-19:
Holy moly! That is awesome! I'll be giving this a try as soon as my parts come in.
I really appreciate that input and knowledge.

If only I could inspire my own guys that easy haha!

I like the idea of vibration sensors, I have power monitoring on the whole tub line, but there are a bunch of the same pumps that might make that difficult.

NorthernMan54, 2025-06-24:
Am the developer of this code base ( but not the person who did the original integration ) https://github.com/NorthernMan54/esp32_balboa_spa

One thing I thought I would share was that for balboa spa packs the api on the rs485 interface is the same as the one used on the cloud/wifi side.  So if you can figure out the wire protocol, then the messages may be the same as the WiFi interface.

DuckManQuack, 2025-06-30:
@agittins Well hopes were dashed with that one. I tired 3 different online retailers all whom *in-stock* did not actually have it in-stock and many years discontinued.... It was a good idea!

ccutrer, 2025-07-21:
> Am the developer of this code base ( but not the person who did the original integration ) https://github.com/NorthernMan54/esp32_balboa_spa
>
> One thing I thought I would share was that for balboa [...]

This is exactly what I'm hoping! I _am_ the person that originally started the reverse engineering of the BWA controller, but I just had my spa replaced yesterday and I'm now rocking a Gecko controller. I really don't want to buy an in.touch (though fun fact - I already have one, on my Mr. Steam steam shower generator)! But given the work that has taken place so far is showing that it's I2C instead of RS485 means it's unlikely to be the same protocol as on the IP side, since I2C is typically used as reading/writing registers (similar to ModBus) rather than a stream oriented protocol that TCP is. But I haven't looked too much at the IP side in geckolib yet - maybe it just abstracts the register style protocol over TCP. Anyhow... subscribing to thread, and hopefully being able to contribute soon-ish. I don't care much about automating the jets, but it's killing me already not being able to see or change the temperature without walking outside!

currentsauce, 2025-07-24:
I’m about to get a hot tub with a Gecko in.yj-2 and was looking for a way of remotely controlling it without having to get an in.touch, great that this is topic is currently in discussion!
Since it looks like the I2C comms are unknown right now, if by the time I get it they’re still not reverse engineered, I’ll try to capture some I2C data.
Did I interpret correctly that the I2C lines on the C and CO port are the same bus?
I would have thought the in.yj is the I2C master and the control panel is a slave. Considering that the control panel can get temperature levels, and control the system, should just be a case of figuring out the I2C commands and data - maybe easier said than done though!
What I’m hoping, is that the control panel embedded system does all the processing of display and buttons, and just sends a simple data word back to the controller, rather than the controller sending complex display data etc.

agittins, 2025-07-24:
> I would have thought the in.yj is the I2C master and the control panel is a slave.

It might be, but my gut instinct is that the pack acts as a slave on the co ports. That way they can release new peripherals without having to have the pack updated and containing control code for every variation. 
Similarly to what @ccutrer describes, it makes sense for the control panel to initiate comms when it wants to rather than the pack having to know how to talk to and control all the different peripherals. 

But until we get further along, we're just guessing.

currentsauce, 2025-07-25:
That makes sense, but I’ve seen in the Gecko manuals that various add on devices are daisy chained together, and in the case of the in.touch, this has to go on the end since it has no port for daisy chaining. With that in mind, the only definite device that is present is the main pack, and can’t have multiple masters in I2C. But then again, I suppose it is a given that there is always a control screen! My only concern is that if the screen is a master, then the in.touch will be controlling in a different way, so I’d hope by mimicking the control panel commands the control panel doesn’t throw a wobbly. In that case, we’d have to mimick the commands of an in.touch.
But yes you’re right we’re just guessing here, definitely looking forward to one of us figuring it out! :)

ccutrer, 2025-07-25:
> I’ve seen in the Gecko manuals that various add on devices are daisy chained together. [...] With that in mind, the only definite device that is present is the main pack, and can’t have multiple masters in I2C.

Just because devices are daisy chained doesn't mean it's a simple bus connection internally. I have a device that is ModBus (similar single master, multiple slave topology), and a OEM cloud connection device for it that has a daisy chain port. The cloud connection device (and the diagnostic device that I plug into the daisy chain port) are both the masters; the actual device I'm controlling is the slave. I know for a fact that that port is not connected as a bus internally in the cloud connection device - it's implementing a full proxy so that if it gets a request on its upstream port, it multiplexes it with its own requests on the downstream port, avoiding any communication conflicts.

agittins, 2025-07-26:
> can’t have multiple masters in I2C

i2c supports multiple masters, they just have to co-operate in taking turns. 
https://www.i2c-bus.org/multimaster/

ccutrer, 2025-07-27:
To my knowledge neither of the underlying I2C drivers in ESPHome (Arduino and ESP-IDF) support multi-master. And there's where most of my I2C knowledge comes from.

dccourt, 2025-08-01:
I finally had the opportunity today to sniff some live data from my IN.YE-5.

I built a Y cable that allowed me to tap the C1 line between the controller board and the keypad unit, and sniff with sigrok, capturing all the non-power pins.

I'm struggling to make sense of what I see, though.  The pins we think are SCL and SDA (D1 and D2 in my capture) do seem to look like I2C, but I don't see any anything very meaningful in the traffic.  (To be fair, I'm not used to viewing I2C in Pulseview so I might just be misinterpreting it).

I do also see activity on D0 apparently going from high to being held low during comms, and a brief low-to-high pulse on D4 at the start.

During this capture, there should be any normal background polling activity, plus events from me activating pump 1 at the keypad for a second or two, then deactivating it.  I think my pump commands are probably around the 20 second mark.

Pin mappings, relative to nebhead's plug image earlier in this thread:
D0 : pin 1 ("NC")
D1 : pin 2 (SCL)
D2 : pin 3 (SDA)
D3 : pin 5 ("NC")
D4 : pin 7 ("NC")

FYI, those pin numbers match the P1 connector numbering on the keypad PCB.

The activity on the NC pins (which if I've understood correctly, aren't shared between the C0 and C1 ports), does make me wonder whether there's something special about the C1 port.   I also seem to remember that the connectors for those ports are specially keyed to avoid inserting to the wrong port, so that further suggests that the ports are not interchangeable.  Still, the C0 port is supposed to be functional for the wifi controller, so it might be possible to do something with that port once the protocol is better understood.

[turn on pump1 1MHZ.zip](https://github.com/user-attachments/files/21555125/turn.on.pump1.1MHZ.zip)

Anyway, I've attached the sigrok capture file if anyone else wants to poke at it.

To do more detailed sniffing, I plan to install an ESP32-C3 attached to the Y-cable, running circuitpython for easy remote update, but to do so I think I'll need to build some resistor dividers first, since the ESP32 is only 3v3 tolerant on its inputs, AFAIK.

dccourt, 2025-08-09:
I'll have to check that it really is 5V on the spa connection, but that sounds good - thanks for the info.

I've been fiddling around a bit more in PulseView to try to understand the capture.  I actually suspect that we might be wrong about the I2C pin assignments.

I'm pretty certain that D0 is an /EN -style signal - it goes low for the duration of any interesting activity.
And D2 appears to be SCL - when it's doing stuff, it's running at a little over 100kHz, which would make sense.
However, D1 rarely seems to do anything very interesting.  Most of the time it seems to be almost like a uniform clock signal, counting every 8 pulses on SCL, like here - where I don't think the decoded protocol makes much sense, either:

[image]

But treating D4 as SDA looks much more plausible at that point:
[image]

This shows a very brief exchange at around 16.58 seconds, which still seems to make sense with D4:
[image]

And here's a full sigrok-cli decoding of the capture, assuming SDA=D4:
[sda_is_d4.txt](https://github.com/user-attachments/files/21694444/sda_is_d4.txt)

codewise-nicolas, 2025-09-05:
Ive done more captures and tried to isolate the bus to try to determine what is happening.

[gecko-in.ye5-intouch-captures-2.zip](https://github.com/user-attachments/files/22168437/gecko-in.ye5-intouch-captures-2.zip)

Possible theories/ideas/other things of note
- Multi master bus?
- Non addressed bus? Since the intouch is meant to be daisy chained, its possible the protocol doesnt use addresses but instead just raw data. The pulseview decoder shows commands are address write to 0x17 and the first byte of the data is also 0x17, 0x17 also re-appears further into the data sequence.
- Possibly a "i want to talk" or "i have some settings" protocol as seen with the app command data starting with a short burst then a long burst and a 2nd long burst, versus the keypad version only having 2 long bursts.

Observations
- At startup something (the pack probably) floods the bus with address write 54 (see "start keypad p1 on off p2 lo hi off light on off repeat app wait till no flo at 4min"), because in the capture where sda/scl is not connected the intouch doesnt do much except try to send a few writes sporadically. 

Main takeways
- 12v is not used by the intouch2. Only 5v was needed.
- Only 5v, sda, scl, gnd are needed (no other signaling happening on other pins)
- Keypad actions cause the same data to be sent on the bus as the app/intouch based ones, with the intouch sending some extra pre command.

And the last thing I only thought about as I was checking the captures, Is the keypad on the same i2c bus? is that the majority of the traffic being captured?
I will try again tomorrow to capture without the keypad connected and see if the intouch still works and what traffic is visible.

codewise-nicolas, 2025-09-10:
Please correct me if Im wrong here @dccourt, but I think what @dccourt was asking for @VivekKaloliyaCore was how did you send commands to the spa pack ? 
From your earlier comment I got the impression you not only "saw" the commands on the bus, but you actually sent some commands to the pack and had it respond?

I will try and check C1 port tomorrow as per your request. 
Is it possible that like their older keypads they dont have a proper bus but merely control the led's and such directly? it would explain why the C1 port is dedicated to the keypad.

Any pictures of the board inside the keypad? That would help us try to understand what kind of communication it performs.
Also what pinout from the keypad side is being used.

But alas, like @dccourt mentioned, it makes more sense to imitate the c0/intouch protocol than to try and intercept the keypad bus.

VivekKaloliyaCore, 2025-09-10:
@codewise-nicolas  and @dccourt 
**Please correct me if Im wrong here @dccourt, but I think what @dccourt was asking for @VivekKaloliyaCore was how did you send commands to the spa pack ?**
I attached keypad on C1 port.
I attached in.touch2 gecko spa transmitter (https://geckoalliance.com/products/in-touch-2/) to CO port.
Then I attached SCL and SDA lines of CO port to ESP32 running I2C spoof code.
Now when I pressed light button on keypad (attached to C1 port), I got some sort of bit patterns along with Start/Stop, Read/Write, ACK/NACK bits.
Based on the bit patterns, I have decoded it into the hex format and compare it with the command decoded by @codewise-nicolas. Nearly both are same.

@codewise-nicolas 
**I will try and check C1 port tomorrow as per your request.**
Did you retrieve command sequence from keypad (C1 port)? If yes, can you help me with your pin diagram or pin description of C1 port that you have used.

@codewise-nicolas 
**Any pictures of the board inside the keypad? That would help us try to understand what kind of communication it performs.
Also what pinout from the keypad side is being used.**
I will try to get them if possible. Lemme see how I goes meanwhile.

dccourt, 2025-09-10:
One of my earlier posts in this thread has photos of the keypad circuit board.

There's a 1x8 0.1" header, P1, which connects directly to the C1 port.  Terrible ASCII art here for the connections, but:

C1 port, looking down into the socket:

1234
5678      5=GND, 6=12V, 7=5V.   Others are all connected at the keypad end, but function not yet certain.

P1 header :     12345678  , left-to-right if P1 header orientated towards top of board.
4=5V
6=12V
8=GND

Pin connections between C1 and P1, in the official keypad cable:

P1    C1
1 ->  4
2 ->  8
3 ->  3
4 ->  7
5 ->  2
6 ->  6
7 ->  1
8 ->  5

I hope to get an RPi Pico W wired up and installed on the C0 port this evening, so I can more easily investigate the C0 port in-situ, and with remote OTA code updates.  Looks like CircuitPython's i2ctarget library might be sufficient, which would be convenient.

codewise-nicolas, 2025-09-13:
Below is a 3rd set of captures.
- spa pack turn on without keypad, but with intouch
- c1 keypad port power up

codewise-nicolas, 2025-09-14:
On the C1/keypad front...

Assuming C1 may be a very basic data bus (not i2c), the capture i got is reminiscent of a simple shift register.
Serial in, parallel out.
The data i captured on the supposed sda/scl pins looks like a latch signal and 8 bits of data.

[image]

However there are other places where the supposed "latch" signal is only after a few bits, not 8 as would be needed to be a shift register.

See dccourt's better captures of C1 here https://github.com/esphome/feature-requests/issues/2893#issuecomment-3169684850
They also have what looks like shift register data on d1/d2

The other possibility would be an UART. The "data" line consistently does have 8 bit groupings.
Unsure what the other lines purpose would be.

codewise-nicolas, 2025-09-14:
Thanks @dccourt. The i2c header and the word "serial" sure helps a little. 
If you still have the keypad open, can you see if any of the i2c labled pins go to the large pin header/wire to the spa pack?

Im wondering if there is different protocols for keypads, from the very basic k19 to the advanced one you have k1000.

dccourt, 2025-09-14:
Good Q about the I2C header, I don't know why I never spotted that there before.  Anyway, the spa cable header (P1), I2C header (P5), and PROG header (P6), and ? header (P7) seem to only share GND.  I didn't find any other connected pins between them.

codewise-nicolas, 2025-09-15:
> I think the `serial` box is just a placeholder for a sticker with a serial number here.

Omg yes your probably right. Doh!

zteifel, 2025-12-27:
Hi! I have been following this topic and decided to buy a in.touch3 to be able to reverse engineer the i2c protocol. I have created a solution and repo that works. It has pump1, pump2, light, temperatures and program selection and some other stuff. Feel free to check it out and leave a comment [https://github.com/zteifel/esphome-gecko](https://github.com/zteifel/esphome-gecko)

nebhead, 2025-12-28:
> Hi! I have been following this topic and decided to buy a in.touch3 to be able to reverse engineer the i2c protocol. I have created a solution and repo that works. It has pump1, pump2, light, [...]

Amazing!  If others can report back on success here, that would be wonderful to see.

agittins, 2025-12-28:
>> If it works, I'd like to throw you and @agittins a little cash to repay you for all your work. Can you set up a donation method?

> That is very kind of you. Just trying to give back to the community. I have added a buy me a coffee link in the repo but feel free to not donate and buy something fun for a project instead.

Likewise, I've added my buy me a coffee link to my profile - contributions are not in any way expected, but they are gratefully accepted and will probably be spent on similar toys/projects! 😄

agittins, 2025-12-28:
> I have created a solution and repo that works. It has pump1, pump2, light, temperatures and program selection and some other stuff. Feel free to check it out and leave a comment https://github.com/zteifel/esphome-gecko

This is fantastic, well done! I'm away from our spa for a while but super excited to try this out and see if it can also support older packs like mine.

[6 more replies]