esphome/feature-requests #2896: Implement FPS metter do LVGL

kyvaith, 2024-09-23:
**Describe the problem you have/What new integration you would like**
Show FPS counter in LVGL screens
**Please describe your use case for this integration and alternatives you've tried:**
Optimize screens for best performance

**Additional context**

In raw LVG we can define `#define LV_USE_PERF_MONITOR 1` to have FPS counter shown.
[image]

clydebarrow, 2024-10-30:
You can simply add the define in your yaml:

```
esphome:
  ...
  platformio_options:
    build_flags: ["-DLV_USE_PERF_MONITOR"]  
```

closed 2024-10-30 by clydebarrow

kyvaith, 2024-10-30:
Perfect 
[image]

UserRS232, 2025-01-17:
> Perfect [image]

what software do you use here??