esphome/feature-requests #2900: Support for Yonan A/C Units (KT3F-24GW/X1)

maciekish, 2024-09-26:
**Describe the problem you have/What new integration you would like**
I have 11 Yonan A/C units which i can not replace and are not supported by ESPHome climate. I do have some programming experience but 0 experience with IR/RF, and cannot make any sense of the codes for the life of me. I don't have a problem writing a platform for Yonan but i need help understanding the codes first.

**Please describe your use case for this integration and alternatives you've tried:**
It seems like the remote (RG57A2/BGEF) is the same across multiple brands, including Midea. So i tried the Midea platform but that didnt work, but ESPHome is able to properly decode the commands from my remote with the Midea platform if i set up a receiver... I'm also able to capture and replay codes using remote_transmitter.transmit_toshiba_ac and rc_code_1: 0xB24D9F60A05F, the Yonan reacts properly to that.

This is ESPHome **receiving** a command properly from my original Yonan remote:

[image]

**Additional context**
Here are some codes i captured, but cannot find any pattern in them:

### Temperatures at low fan in cooling mode
| Temp | Hex | Binary |
| --- | --- | --- |
| 17 | 0xB24DBF4000FF | 101100100100110110111111010000000000000011111111 |
| 18 | 0xB24DBF4010EF | 101100100100110110111111010000000001000011101111 |
| 19 | 0xB24DBF4030CF | 101100100100110110111111010000000011000011001111 |
| 20 | 0xB24DBF4020DF | 101100100100110110111111010000000010000011011111 |
| 21 | 0xB24DBF40609F | 101100100100110110111111010000000110000010011111 |
| 22 | 0xB24DBF40708F | 101100100100110110111111010000000111000010001111 |
| 23 | 0xB24DBF4050AF | 101100100100110110111111010000000101000010101111 |
| 24 | 0xB24D9F6040BF | 101100100100110110011111011000000100000010111111 |
| 25 | 0xB24D9F60C03F | 101100100100110110011111011000001100000000111111 |
| 26 | 0xB24D9F60D02F | 101100100100110110011111011000001101000000101111 |
| 27 | 0xB24D9F60906F | 101100100100110110011111011000001001000001101111 |
| 28 | 0xB24D9F60807F | 101100100100110110011111011000001000000001111111 |
| 29 | 0xB24D9F60A05F | 101100100100110110011111011000001010000001011111 |
| 30 | 0xB24D9F60B04F | 101100100100110110011111011000001011000001001111 |

### Fan settings at 25 degrees
| Speed | Hex | Binary |
| --- | --- | --- |
| Aut | 0xB24DBF40C03F | 101100100100110110111111010000001100000000111111 |
| Low | 0xB24D9F60C03F | 101100100100110110011111011000001100000000111111 |
| Med | 0xB24D5FA0C03F | 101100100100110101011111101000001100000000111111 |
| Hgh | 0xB24D3FC0C03F | 101100100100110100111111110000001100000000111111 |

### Modes at 25 degrees
| Mode | Hex | Binary |
| --- | --- | --- |
| Auto	| 0xB24D1FE0C837 | 101100100100110100011111111000001100100000110111 |
| Cool	| 0xB24D9F60C03F | 101100100100110110011111011000001100000000111111 |
| Dry 	| 0xB24D1FE0C43B | 101100100100110100011111111000001100010000111011 |
| Heat	| 0xB24D9F60CC33 | 101100100100110110011111011000001100110000110011 |
| Fan 	| 0xB24D9F60E41B | 101100100100110110011111011000001110010000011011

**Any chance CN2 or CN3 is unlabeled UART?** There is a UART `midea` platform as well...
[image]
[image]
[image]
[image]

maciekish, 2024-09-28:
False alarm. Looks like "Coolix" works -.-

DjordjeMandic, 2024-12-20:
CN3 is for UART needed for midea component