esphome/feature-requests #2902: HX711 via I2C (m5stack)

PineappleEmperor, 2024-09-28:
**Describe the problem you have/What new integration you would like**
Support HX711 sensors via I2C.

**Please describe your use case for this integration and alternatives you've tried:**
I currently use a lot of m5stack ESP32 gear, and would like to use the m5stack HX711 (along with other sensors) which uses I2C. At present, the HX711 sensor requires dout_pin and clk_pin. I've looked to see if I can directly use arduino code, but this is a messy workaround and I'd prefer to keep to native ESPHome if possible.

**Additional context**
I'm new (and very amateur) with ESPHome, so if I'm missing something obvious please let me know!

HeathersZen, 2024-10-19:
Bump. Just got this thing and am getting into making it work.

HeathersZen, 2024-10-19:
FWIW,  I managed to get the M5 Atom stack to return a value by following this video: https://www.youtube.com/watch?v=GlIS2HsSdWY

The only difference I can tell from my original attempt was that I flashed ESP Home to the device using the https://web.esphome.io/ website instead of the HAOS add-in.

I haven't calibrated it yet, so I may yet run into issues, but here's the YAML that is currently working. I haven't bothered to clean it up yet.

substitutions:
  name: esphome-web-a03858
  friendly_name: Aeroponica-Scale1

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:
  port: 80

sensor:
  - platform: hx711
    name: "HX711 Value"
    dout_pin: GPIO32
    clk_pin: GPIO26
    gain: 128
    update_interval: 60s

Eyal51, 2024-10-19:
If that's working for you then you're not using a i2c device.

HeathersZen, 2024-10-19:
I’m using the M5Stack unit. Perhaps it isn’t the one OP is referring to.Scale Kit with Weight Unitshop.m5stack.com

ScottG489, 2024-10-21:
I believe we're referring to this one:
https://shop.m5stack.com/products/mini-scales-unit-hx711

From chatting with the devs, this would require code changes to ESPHome.

PineappleEmperor, 2024-10-21:
> I believe we're referring to this one:
> 
> https://shop.m5stack.com/products/mini-scales-unit-hx711
> 
> 
> 
> From chatting with the devs, this would require code changes to ESPHome.

Yeah that's the one I was referring to. I had assumed the hx711 component would need changing, but similar to other components that can be used with I2C OR direct pins?

DjordjeMandic, 2025-01-23:
[image]

Is this it?

regencape, 2025-01-24:
If you cut a grove cable and put the grove side into your HX711 and the wires into the right side of an Atom Lite (= i2c), it will work. At least, it does with me. :) Yellow=G(PIO)21, white=G(PIO)25, red=5V(olt), black=G(round). 

Use the following code ([see](https://community.home-assistant.io/t/help-needed-with-esphome-config-for-m5stack-mini-scale/755527/6) [and also](https://github.com/m5stack/M5Unit-Miniscale/tree/3e92369d61f346ed42f38a8c3155773a2685735a/src)):

```
# i2c communication
i2c:
  - id: bus_a
    sda: GPIO25
    scl: GPIO21
    scan: true

i2c_device:
   id: i2cdev
   address: 0x26

# Returns actual sensor data (sends data every second)
sensor:
- platform: template
  name: "HX711 weight Sensor"
  update_interval: 1s
  unit_of_measurement: "g"  
  device_class: "weight"
  state_class: "measurement"
  lambda: |-
    uint8_t data[4];
    float c;
    uint8_t *p;
    id(i2cdev).read_bytes(0x10,data,4);
    p = (uint8_t *)&c;
    memcpy(p, data, 4);    
    return c;

# Button in Hassio to "calibrate" (sensor is set ~0)
button:
  - platform: template
    id: SetOffset
    name: "HX711 calibrate"
    on_press:
      then:
        - lambda: |-
            id(i2cdev).write_byte(0x50, 1);

# HX711 button in Hassio (only returns on/off)
binary_sensor:
  - platform: template
    name: "HX711 button"
    icon: "mdi:radiobox-blank"
    lambda: |-
      uint8_t data = 0;
      id(i2cdev).read_bytes(0x20,&data,1);
      return data==0;
```

ScottG489, 2025-02-19:
Thanks, this worked great for me! Only difference was the GPIO for the i2c was 26 and 32 respectively for me. I'm using a grove cable plugged into the grove connector on a Atom Lite and the device.

Tirccir, 2025-05-22:
@regencape This is the only article I can find anywhere relating to using I2C with the M5stack scales. I was wondering if you had any advice on this? I have currently tried your code and I get no errors but I only get a reading of 0 grams.

I am using the M5stack Scales Kit but I have replaces the transmitter with the Unit Weight-I2C which is slightly different to what everyone else on this thread is using. Worth a shot I guess.

Unit Weight-I2C:
https://docs.m5stack.com/en/unit/Unit-Weight%20I2C

Scales Kit:
https://docs.m5stack.com/en/app/scales_kit

Thanks in advance

regencape, 2025-05-24:
@Tirccir I'm not really the expert. Key thing is to have the ports and addresses right. The only thing I can think of that might cause the issue is the following:
Unit Weight-I2C                   SCL (white)      SDA (yellow) 
Scales Kit                             SCL (yellow)      SDA (white) 
Mini Scales Unit (HX711)    SCL (white)      SDA (yellow) 

But this hardly seems the issue since the SCL and SDA in the Unit Weight-I2C is similar to Mini Scales Unit. But maybe something happens I don't understand either. The schematics on https://docs.m5stack.com/en/app/scales_kit and https://docs.m5stack.com/en/unit/Unit-Weight%20I2C might help, but that's way over my head. :)