esphome/feature-requests #2906: Add support to MLX90640 Thermal camera

darbogx, 2024-10-01:
**Describe the problem you have/What new integration you would like**
Add support for MLX90640 as I did not found any easy way to make it works with HA.

**Please describe your use case for this integration and alternatives you've tried:**
The thermal imaging camera can be used to monitor 3D printers, heating and other electrical equipment.

**Additional context**
I am not the first one who are looking for it. Maybe there is possibility to add official support working with ESP and MLX90640, I would be very grateful :).

Jochen754, 2024-10-06:
This might be what you are looking for:
https://github.com/Chill-Division/M5Stack-ESPHome/blob/8da018d6d5eaa73a92276594999f5a4e7c32fb60/Thermal%20Camera%20Unit%20(MLX90640).md

darbogx, 2024-10-07:
I saw this and tried but have trouble with compiling that.

Jochen754, 2024-10-07:
You are right. My linked example is not the current version and won't compile.
The current version that also will compile is:

https://github.com/Chill-Division/M5Stack-ESPHome/blob/main/Thermal%20Camera%20Unit%20(MLX90640).md

It compiles for me but I have no MLX90640 at hand. You can try it and give feedback if it works. :)

darbogx, 2024-10-09:
Ok, I made a few attempts and it works, thanks. I made a silly mistake that caused a compilation error. Temperature sending to HA correctly. Is there any option to send the image to HA? Just like with the other cameras?

Jochen754, 2024-10-10:
Showing the image sould be possible with the card in homeassistant.

Just use http://ipAddressOfDevice/thermal-camera as the URL