esphome/feature-requests #2910: Light component - fix output min_power headaches by adding a minimum brightness control to lights

ojwc, 2024-10-05:
**Describe the problem you have/What new integration you would like**

Add light control parameters to all PWM light components (e.g., monochromatic, CCT, RGBWW etc.) to allow minimum and maximum brightness values to be set - in the light component itself, rather than relying on setting `min_power` in the output component. 

A `minimum_brightness` parameter would tell the light component the lowest value to pass to the output component for a set/constant brightness. When transitioning to or from off, the light component would behave as it currently does. It would also continue to represent the lowest possible brightness setting in Home Assistant/web server as 1% - that is, the brightness range should be automatically recalculated. Users could find the 'correct' value for `minimum_brightness` by finding the lowest brightness setting (percentage) at which the light with default settings is consistently on without flickering, then set `minimum_brightness` to that value.

A visual example:

```
light:
    - platform: rgbww
      id: lightbulb_rgbww
      name: "Lightbulb"

      warm_white: output_warmwhite
      cold_white: output_white
      red: output_red
      green: output_green
      blue: output_blue

      cold_white_color_temperature: 6500 K
      warm_white_color_temperature: 3000 K

      minimum_brightness: 14%
```

**Please describe your use case for this integration and alternatives you've tried:**

I have many RGBWW lightbulbs flashed with ESPHome. Without setting `min_power` in each colour channel's output component, when brightness is set to ~14% or lower, the light simply turns off. If I set `min_power` (and accompanying `zero_means_zero` to ensure the light _can_ turn off), the light will not turn off at low brightness values as before, but between 0% and 14% the brightness simply doesn't change. In addition to this, when toggling the light, setting `min_power` leads to the light transition 'stuttering' around the minimum value. When turning on, the light will instantly turn on to the minimum and then transition as expected. When turning off the reverse happens; it will transition down to the minimum value then instantly turn off. When `min_power` isn't used, the light transitions smoothly on and off, as expected. I suspect this occurs because `min_power` is telling the device that no PWM output value below the set value is permitted (or, with `zero_means_zero` set to true, no value between 0 and the set value is ever permitted), and the light components aren't aware of this and do not account for it.

A different but related issue (more of an undesired or clunky behaviour) is that in either case, the brightness range of the light is never truly represented. Ideally that range would be from 0% (off) to 100% (maximum brightness), with 1% representing minimum brightness. In my case the range is from 14% (off) to 100% (maximum), whether `min_power` is configured or not, and around 15% represents minimum brightness.

If a minimum brightness could be configured in the light component, the output components could be left with default parameters. The light component logic could manage transitions across the minimum value, to or from zero. This could also solve the scaling issue mentioned above.

**Additional context**

I've tried to tackle this issue many times before. There are some examples on reddit and the home assistant community forum of people successfully using template output components to apply lambda filters to PWM values, but I've found this approach to be needlessly difficult as well as inconsistent. Besides these few examples, there are many more of people trying to address undesired light behaviour and being unsatisfied with simply setting `min_power` on the output component.

Setting `gamma_correct` to 1 solves the issue completely (which is fine for monochromatic or CCT lights) but doing this breaks any RGB features - colours just end up not reflecting what is set by the user. It's possible to create separate RGB and CCT light components and then set `gamme_correct` to 1 for the CCT light, but the RGB light will continue to have the issues outlined above. Setting a global `minimum_brightness` value for a light component would be a more straightforward approach.

bmarien, 2024-10-14:
+1

I've been looking for the exact same thing.

bzumik1, 2024-10-25:
I'm so glad I found this issue. This is a total mess; I don’t understand how it made it to production. It seems to be caused by the default gamma of 2.8. Wouldn’t it be possible to calculate the minimum brightness based on the set gamma, so the ‘minimum brightness’ option on the light wouldn’t even be necessary?

ssieb, 2024-10-26:
I'm not understanding what you're doing that's causing these issues.  Why do your outputs have a minimum value?
If you set the minimum power on the output correctly, the light should still work smoothly down to that point because the output remaps the range, so the light still thinks it has the full range.  And the min power has to be on the output because that is independent of any gamma correction.
You're going to have to provide more information.

ojwc, 2024-10-26:
> I'm not understanding what you're doing that's causing these issues. Why do your outputs have a minimum value? If you set the minimum power on the output correctly, the light should still work [...]

The output has a minimum value because without it, when brightness on the light is set below some value (in the case of most of my lights, about 14%) they simply turn off. However when the minimum value is set, the light transitions don’t work correctly when turning on or off. I did state this in my issue.

ssieb, 2024-10-27:
What kind of output has a limitation like that?
What do you mean by "Brightness values between 1% and 14% are the same when min_power is set on the output."?

ojwc, 2024-10-27:
libretiny_pwm
When setting brightness on the light component, either in Home Assistant, the web server interface, or HomeKit (via HK bridge). When min_power is configured on the output to the minimum value that will turn the light on, brightness values between 1% and 14% are identical.

ssieb, 2024-10-27:
What is the config and what is the gamma correct set to?

ojwc, 2024-10-27:
This one uses SM2135 outputs but the results are the same with similar lights using libretiny_pwm. Gamma correct is not used.

```
esphome:
  name: light-genio-i003330-98bddf
  friendly_name: "Lightbulb - Dining Room Lamp - Genio Candle 5W RGBCCT"
  comment: "Mirabella Genio Wi-Fi Dimmable 5W RGB+CCT LED Candle (I003330)"

bk72xx:
  board: wblc5

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  - platform: esphome
    password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Light-Genio-I003330-98C542"
    password: !secret ap_password
captive_portal:
    
light:
  - platform: rgbww
    name: Lightbulb
    
    red: output_red
    green: output_green
    blue: output_blue
    warm_white: output_ww
    cold_white: output_cw

    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    color_interlock: true
    constant_brightness: True
    restore_mode: RESTORE_DEFAULT_ON
    default_transition_length: 800ms

sm2135:
  data_pin: P24
  clock_pin: P26
  cw_current: 30mA
  rgb_current: 30mA

output:
  - platform: sm2135
    id: output_blue
    channel: 0
  - platform: sm2135
    id: output_green
    channel: 1
  - platform: sm2135
    id: output_red
    channel: 2
  - platform: sm2135
    id: output_ww
    channel: 3
    min_power: 0.01
    zero_means_zero: True
  - platform: sm2135
    id: output_cw
    channel: 4
    min_power: 0.01
    zero_means_zero: True
```

ssieb, 2024-10-27:
Unless you change it, the default gamma is 2.8, so yes, it's being used.

ojwc, 2024-10-27:
> Unless you change it, the default gamma is 2.8, so yes, it's being used.

Correct, to clarify, i haven’t changed gamma correct from the default 2.8. The min_power of 0.01 is set due to the way gamma correct is applied. Re documentation for the light component, my target min power for this lightbulb was about 19%, so 0.19^2.8 is roughly 0.01.

ssieb, 2024-10-27:
I would suggest disabling gamma correction to get the correct output level.  I don't understand what a min power of 0.01 is even going to change.  What's the problem if you go lower?  There's almost no lower to go there.  It's only an 8-bit value.

ojwc, 2024-10-27:
I have explained twice now why setting min_power is required, and also explained in my feature request how changing or removing the gamma correction leads to problems with RGB colour accuracy

bzumik1, 2024-11-01:
@ssieb Please check the original post from @ojwc; it explains everything. When lambda correction is not applied, everything seems to work fine, but when it’s left on default, it starts doing strange things. Transitions do not work correctly, etc. Everything is in the original post.

ThisIsMissLiz, 2026-09-10:
+1 same
Basically just looking for a way to remap brightness of the entire light from the "real", say 20-100, firmware side to a UI of "0-100", since my bulbs don't seem to "respect" zero_means_zero that well. That'd be so awesome if doable! :)

ssieb, 2026-09-11:
> a way to remap brightness of the entire light from the "real", say 20-100, firmware side to a UI of "0-100"

That's exactly what `min_power` is for.

> since my bulbs don't seem to "respect" zero_means_zero that well

They have to or something is really broken.  I think you're misunderstanding something.

dbuezas, 2026-10-11:
For the SM2135 case, the 14 % is one chip step: 0.14^2.8 ≈ 1/255. Below that the chip gets 0, and `min_power` can't add steps.

These chips also have a **current** setting, which ESPHome keeps fixed. My external component [esphome-dynamic-range-led-driver](https://github.com/dbuezas/esphome-dynamic-range-led-driver) lowers the current when the light is dim and raises the level to match, so the light goes dimmer, with more steps and no `min_power` needed. With @ojwc's `30mA` that should be about 3 times dimmer (off at about 9 % instead of 14 %).

**Tested only on BP5758D/BP5768.** SM2135, SM2235, SM2335 and BP1658CJ should work too. If you try it, please tell me in an [issue](https://github.com/dbuezas/esphome-dynamic-range-led-driver/issues) whether it worked.

For ESP8266 PWM bulbs, the same repo has `esp8266_stretch_pwm` to extend the minimum brightness (not for `libretiny_pwm`).