esphome/feature-requests #2913: "HTTPS prevents ESPHome from being able to install this on devices connected to this computer" ... instructions unclear?

microfx, 2024-10-06:
The first instructions are super unclear ... it sounds like three paragraphs telling me the very same thing... maybe clarify / rephrase it. 

`
You are not browsing the dashboard over a secure connection (HTTPS). This prevents ESPHome from being able to install this on devices connected to this computer.

You will still be able to install ESPHome by connecting the device to the computer that runs the ESPHome dashboard.

Alternatively, you can use ESPHome Web to prepare a device for being used with ESPHome using this computer.
`

first ... HTTPS is preventing ESPHome from being able to install this on devices connected to this computer ... but I will still be able to install ESPHome by connecting the device to the computer ... alternatively I can use the Webinterface to prepare ...? 

This absolutely doesn't help me ... and I am quite a proficient noob that understands the basics of ESPHome. I am super confused now. 

I would rewrite this as it makes no sense without being one of the lead coders who understand the tiny nuances in ... some of the words I guess.

ssieb, 2024-10-06:
"This computer" is the one you're running the web browser on.
"The computer that runs the esphome dashboard" is probably your HA system. 
"Esphome web" is https://web.esphome.io/.

Feel free to create a PR with better wording.  Or even put it here.

microfx, 2024-10-06:
next thing I came across with "wrong information": 

`
Not all requirements are currently met. The easiest solution is to download your project and do the installation with ESPHome Web. ESPHome Web works 100% in your browser and no data will be shared with the ESPHome project.
`

then I click on the link to ESPHome Web .. and it says: 

`
Dashboard Unavailable
ESPHome Web requires a browser that supports WebSerial. Please open this website on your desktop using Google Chrome or Microsoft Edge.
`

(I'm using Safari) – thanks for the translation so far... still not 100% clear. I guess I gotta install some HTTPS stuff to not use Chrome or something (I don't want to... my browser is Safari)

ssieb, 2024-10-07:
There's no wrong information there.  You just have to use a compatible browser (Chromium also works if you don't want to use Chrome).  Someone is working on a webserial extension for firefox which seems to work, but that's still not Safari.  I actually thought Safari was using the same core as Chrome now.
Even if you setup HTTPS, you still need a compatible browser.  An alternative method is to use the cli esptool directly.

microfx, 2024-10-07:
I don't wanna argue ... but to me
that text is then double wrong. I was working 5h straight to get HTTPS going because the text said "then it'll work!!" ... but it's also not possible with Safari? The text clearly says: "Your browser can do it easily!!" and then "ooops not this browser — even with HTTPS setup!!"

ssieb, 2024-10-07:
If you don't like the text, then make a PR to improve it.
I didn't say to use the esphome cli, although that's what I use because it's so much easier.  I said you could use the esptool cli program to flash the downloaded firmware file.

microfx, 2024-10-07:
It's not about liking a text... it's that it's telling me stuff that is not true. The text let me suggest it checked whether my browser was capable of the action but the browser can't do it. Even when the HTTPS would be working. 

Also ... I love the esphome cli so much ... but it also failed me and cost me hours of my lifetime. Just like HTTPS and the rest of my life 🤪 yes, my life would be better with Chrome probably! Thanks for listening! 👂