esphome/feature-requests #2917: Handle Shared SPI Reset Pins, e.g., for multiple SPI displays

kkilchrist, 2024-10-12:
**Describe the problem you have/What new integration you would like**
I'm working with multiple GC9A01A displays on a shared SPI bus, each with its own CS pin but shared RST, DC, MOSI, SCK, GND, and VCC lines. It's a project inspired by this one: https://github.com/brettdottech/info-orbs . That project, like mine, uses shared DC and RST lines for all SPI devices. 

When trying to initialize the displays, the shared reset pin is toggled upon each initialization (per ili9xxx component). 
Only the last display works correctly, while subsequent displays either mirror the first or stay blank.

I propose a feature to manage shared reset lines more effectively in multi-device SPI setups, avoiding repeated resets that disrupt proper initialization.

**Please describe your use case for this integration and alternatives you've tried:**
Users may have one ESP32 driving multiple info displays. Given that GC9A01A has an internal frame buffer, a very high number of displays could be driven from a single SPI bus, limited only by the CS pin.

Currently, I hacked on a "uses_shared_reset_pin" parameter in the YAML and added a static boolean to track global_shared_reset_has_been_performed, and added appropriate conditionals in the setup sequence. [See here for that source code, db901e1](https://github.com/kkilchrist/esphome/commit/db901e1f707abd52bf5302f84345ae59bfa78064)

However, I think this should be handled as a parameter on the SPI reset pin, perhaps, as part of the pin schema? Or at the SPI device level? I'd appreciate some input / discussion about the best approach. 

**Additional context**
Special thanks to Discord user @pzich for correctly pointing me towards the underlying issue.

battifreund, 2024-10-30:
This is exactly the problem I encountered. I think the solution is great! It saves me a lot of soldering work and prevents some GPIOs from being wasted.

jesserockz, 2024-10-31:
Could you just add a gpio switch on the reset pin, remove it from the display configs and just toggle it twice using `on_boot` before the display is initialized?

kkilchrist, 2024-10-31:
@jesserockz I do think that would work as well

clydebarrow, 2024-10-31:
> Could you just add a gpio switch on the reset pin, remove it from the display configs and just toggle it twice using `on_boot` before the display is initialized?

 Another approach would be to add it to only *one* display and use the `setup_priority` to make sure it gets setup first. That would guarantee the reset happens before any other display config. The ili9xxx default priority is HARDWARE, i.e. 800, so

```
    setup_priority: 810
```

should do it, the others can stay at the default.

clydebarrow, 2024-11-19:
> Can you share an example of the on_boot that would toggle it twice?

No, just add the setup priority line as above to one display config, which also has a reset pin configured, and do _not_ configure a reset pin for the remaining displays. You don't want to do anything in on_boot..

ptparker, 2024-11-19:
@clydebarrow -- That makes sense. 

Seems simple enough. I will test it over the weekend. Beyond the required boot up reset, Is there any situation you can think of where a display component would use the reset pin and not reset via a fill or a device specific SPI command? I don't know how I would even send a reset via the reset pin. I am asking broadly, but in this specific use case I'm using a GC9A01A with the [ILI9xxx TFT LCD Series](https://esphome.io/components/display/ili9xxx.html) component. 

It's a similar project as the OP of this issue. I am repurposing this ESP32 project into an ESPHome based Multifunction Display for HA. https://github.com/brettdottech/info-orbs

clydebarrow, 2024-11-19:
> Is there any situation you can think of where a display component would use the reset pin and not reset via a fill or a device specific SPI command?

Essentially, no. Certainly the ili9xxx driver only ever uses the reset pin on first setup.

closed 2024-11-24 by clydebarrow