esphome/feature-requests #2918: Enable online_image and/or http_request for platform: host

kkilchrist, 2024-10-13:
**Describe the problem you have/What new integration you would like**  
Add support for the `http_request` component on the host platform in ESPHome, or override / bypass the http_request check in the online_image component.

**Please describe your use case for this integration and alternatives you've tried**  
I’m working on an on-device `libwebp` decoder for `online_image`. WebP images are much smaller than PNG (e.g., 2 KB vs 33 KB for a test image), which makes them ideal for low-bandwidth scenarios and fast transmission in IoT environments. Additionally, WebP supports both lossless and lossy compression. 

Enabling the host platform (macOS) would make debugging much faster. The host platform already has networking support (e.g., for API access), so hopefully this is not a huge ask.

**Additional context**

kkilchrist, 2024-10-17:
Made some pretty exciting progress on this today, after some trial and error (trying to use platformio packages, etc etc), turns out,  a wrapper around the local lib curl works great on MacOS. 

I also had to make some updates to the json component (which uses platform flags to allocate a read buffer). 

Anyway, this allows HTTP requests to be on the host platform. I've only tested GET, but it works on my machine for both http and https endpoints. 

For those interested, I’ve added the component to my GitHub fork. 

Here's a sample YAML:
```yaml
esphome:
  name: local-http-request
  platformio_options:
    build_flags: 
      - -lcurl

host: 

external_components:
  - source:
      type: git
      url: "https://github.com/kkilchrist/esphome"
      ref: "http_request_curl"
    components: [ "http_request", "json" ]

http_request:
  id: http_request_component
  timeout: 1s  

interval:
  - interval: 10s  
    then:
      - logger.log: "Making HTTP request"
      - http_request.get:
          url: "https://jsonplaceholder.typicode.com/todos/2"
          capture_response: true  # Capture the response for use in the Lambda function; when set to true, the response data will be captured and placed into the body variable as a std::string for use in lambdas
          on_response:
              - logger.log:
                  format: 'Response status: %d, Duration: %u ms'
                  args:
                    - response->status_code
                    - response->duration_ms
              - lambda: |-
                  ESP_LOGI("main", "Response body: %s", body.c_str());

logger:
  level: INFO  # Enable debug level logging to see detailed logs
```

This wasn't a small task as I had originally hoped, but it's progress!

lanrat, 2024-11-16:
Hello @kkilchrist, I just ran across this issue looking to solve the same problem for implementing gif support for online_image.

How stable is your solution? And how far along is your webp support? If it supports animated webp images, it is likely a better alternative top gif support.

Thanks.

regystro, 2024-12-12:
This feature would be very useful to speed up the testing process. I use a waveshare e-paper display in some projects with LVGL, and compiling, uploading and testing scripts is almost instantaneous using host platform as opposed to the normal way. This allows faster previews for positioning display components, choosing font sizes, an so on.

Any progress in your WIP? What I've tested so far, does not work in Linux/WSL environment.

Cheers.

clydebarrow, 2025-01-06:
```
host:
  mac_address: "62:23:45:AF:B3:DD"

http_request:
  id: http_request_id

external_components:
  - source: github://pr#7908
    components: [image,online_image,animation]
  - source: github://pr#8040
    components: [http_request]
  - source: github://pr#8039
    components: [json]

online_image:
  - id: edge_1
    url: http://localhost/AlphaEdge.png
    resize: 100x100
    format: png
    type: RGB
    update_interval: 60s
    use_transparency: opaque
```

regystro, 2025-01-23:
Hi. I'm getting this error testing the above code:
```
INFO ESPHome 2024.12.4
INFO Reading configuration host-test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing sdl (platform: platformio/native)
------------------------------------------------------------------------------------------------------------------------
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- pngle @ 1.0.2
Compiling .pioenvs/sdl/src/esphome/components/json/json_util.o
Compiling .pioenvs/sdl/src/esphome/core/component.o
Compiling .pioenvs/sdl/src/esphome/core/component_iterator.o
Compiling .pioenvs/sdl/src/esphome/core/controller.o
Compiling .pioenvs/sdl/src/esphome/core/entity_base.o
Compiling .pioenvs/sdl/src/esphome/core/helpers.o
src/esphome/components/json/json_util.cpp: In function ‘std::string esphome::json::build_json(const json_build_t&)’:
src/esphome/components/json/json_util.cpp:17:30: error: ‘const class esphome::RAMAllocator<unsigned char>’ has no member named ‘get_max_free_block_size’
   17 |   auto free_heap = ALLOCATOR.get_max_free_block_size();
      |                              ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/json/json_util.cpp: In function ‘bool esphome::json::parse_json(const std::string&, const json_parse_t&)’:
src/esphome/components/json/json_util.cpp:52:30: error: ‘const class esphome::RAMAllocator<unsigned char>’ has no member named ‘get_max_free_block_size’
   52 |   auto free_heap = ALLOCATOR.get_max_free_block_size();
      |                              ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/sdl/src/esphome/core/log.o
*** [.pioenvs/sdl/src/esphome/components/json/json_util.o] Error 1
============================================== [FAILED] Took 2.74 seconds 
```

clydebarrow, 2025-01-24:
You'll need to be building against the dev branch.

regystro, 2025-01-24:
You're right, thank you. I've been running some tests and here are my results:

External image with HTTP works like a charm:
`url: http://pldh.net/media/pokemon/go/alpha/pikachu.png`

However, when using HTTPS, the SDL window crashes with no error thrown, even specifying `verify_ssl: false` in `http_request:`
`url: https://pldh.net/media/pokemon/go/alpha/pikachu.png`

clydebarrow, 2025-01-24:
Yes, there is a lot left to do yet. Contributions are welcome.

closed 2025-04-28 by jesserockz