esphome/feature-requests #2920: Sensor for modbus device state

Martidjen, 2024-10-13:
**Describe the problem you have/What new integration you would like**

I would like to be able to check the state of a modbus device. Just seeing if the device is online or offline, like ping for webservices. 

**Please describe your use case for this integration and alternatives you've tried:**

I am currently monitoring my solar inverter through modbus. All working well, until the sun goes down and the inverter shuts off. The state of my modbus sensors will keep showing the latest known value received from the inverter, so it looks like the inverter is still delivering very little power all night long. Would be nice if I could either have a sensor for the modbus state of the device which I can use to overwrite the sensor states to 0, or have something like the availability payload options for mqtt sensors in home assistant for each modbus sensor. 

**Additional context**

nagyrobi, 2024-10-13:
Related: https://github.com/esphome/feature-requests/issues/1568, https://github.com/esphome/feature-requests/issues/1694, https://github.com/esphome/issues/issues/6075

Martidjen, 2024-10-14:
Thank you for linking this issue, I did not find this one when searching for related issues. I will try to use the timeout filter for now as mentioned in #6075, this does kind of seem to be what I am looking for, although it is still kind of masking the real issue for offline modbus offline devices.