esphome/feature-requests #2924: UART over USB (CDC/VCP)

ruben-iteng, 2024-10-18:
**Describe the problem you have/What new integration you would like**

I would like to have the option to use the USB hardware port on selected platforms (R2040, ESP32-C3/S2...) as UART bus. Similar to how the `logger` component can use the USB_CDC as hardware port but then for the `uart` component.

**Please describe your use case for this integration and alternatives you've tried:**

With this functionality it is possible to write new/custom components that require UART over USB.
e.g. I could make an AT command component to control the ESPHome device with a host, or the other way around (some solar inverters use ModBus over USB CDC/VCP to communicate).

**Additional context**

-

nagyrobi, 2024-10-18:
@clydebarrow

ruben-iteng, 2024-10-18:
> like this one? [esphome/esphome#7520](https://github.com/esphome/esphome/pull/7520)

I saw this, but I'm nut sure it is the same issue. I'm really just looking to do this:
``` yaml
uart:
  hardware_uart: USB_CDC
```
instead of `rx/tx_pin` or `port`

randybb, 2024-10-18:
should be as for now USB_CDC or USB_SERIAL_JTAG are only for logger. With this you could use it for anything else - as a standard serial port, HID on some never even as USB OTG.

ruben-iteng, 2024-10-18:
> should be as for now USB_CDC or USB_SERIAL_JTAG are only for logger. With this you could use it for anything else - as a standard serial port, HID on some never even as USB OTG.

Yes, thank you. I indeed looked into it a bit, but this also requires changes to the `uart` component I expect. This features request is still valid in that case (independent from how other USB related things will be implemented).
I don't know the code base that well, but it might be a small change to support UART over USB already for the `uart` component as it is already implemented for the `logger`.

clydebarrow, 2024-10-18:
> With this you could use it for anything else - as a standard serial port, HID on some never even as USB OTG.

Not exactly. USB OTG is a host function, can can't be used at the same time as USB client mode, which is what this request relates to. Using the USB-CDC or JTAG serial as a UART shouldn't be hard since it's built into the microcode and would not need changes to the UART component, just need to implement the UART class in a separate component.  HID would need a significant amount of new code.

randybb, 2024-10-18:
Sure not with the host one, but would be possible to make similar for otg... - mentioned that how it is with usb in general. And Tom is already working on HID.

djtef, 2025-01-12:
Hello, 
This improvement would be very useful in my case, my heat pump communicate in serial with a gateway in USB CDC. 
To replace the manufacturer gateway by a DIY one with esphome, I need a bridge USB CD/TTL. With this new feature the esp32 s2 would be sufficient.

ruben-iteng, 2025-01-21:
@clydebarrow could you give some pointers on how to implement this. I might be able to implement it with some help 🙂

clydebarrow, 2025-01-21:
Depends what "this" is. If it's the subject of this feature request, i.e. a USB client as a UART component, then I have not looked at it.

ruben-iteng, 2025-01-21:
> Depends what "this" is. If it's the subject of this feature request, i.e. a USB client as a UART component, then I have not looked at it.

Oh yes sorry for not being clear.
I indeed ment USB client as UART component (what the logger component is using currently).

Danixu, 2025-02-07:
Just I was looking for this. I have a climate device that uses a USB connector as UART and I have bought a ESP32-C3 to try to use the USB as UART, but I see that for now is not possible.

felipecrs, 2025-02-07:
@Danixu, maybe @clydebarrow's [solution](https://github.com/esphome/feature-requests/issues/2944#issuecomment-2538295242) will work for you.

clydebarrow, 2025-02-07:
> ESP32-C3 to try to use the USB as UART, but I see that for now is not possible.

You need an S3 or S2. The C3 does not support USB OTG.

Danixu, 2025-02-07:
Luckily the ESP32 is a cheap hardware 😅 . I'll try to buy some S3 and I'll test it. For now I have soldered a USB cable to the GPIO1 and GPIO3 pins to use the normal UART, but I prefer a cleaner approach.

Thanks!!

felipecrs, 2025-10-15:
Isn't this done already?

https://esphome.io/components/usb_uart/

tomaszduda23, 2025-10-31:
https://github.com/esphome/feature-requests/issues/2924 I started implementation for nrf52.

closed 2025-12-14 by bdraco