esphome/feature-requests #2929: Add support for partial updates on Waveshare 7.5v2

eikowagenknecht, 2024-10-23:
**Describe the problem you have/What new integration you would like**

The Waveshare 7.5v2 supports partial refresh (as seen for example in this table):
[image]

**Please describe your use case for this integration and alternatives you've tried:**

It would be nice if esphome supported this as well for this model, so that less refreshing needs to occur

**Additional context**

For other models, there are alreasy issues here, but not for this one (as far as I can see).

adrione, 2024-11-11:
Please! :)

M-Rapske, 2025-02-27:
Updated and tested. Works well. Thanks a Lot.