esphome/feature-requests #2931: [Modbus_Controller] Modbus Server with dynamic adress

Alexandreao, 2024-10-23:
**Describe the problem you have/What new integration you would like**

Modbus server with and dynamic address tag. Allow a lambda function to modify the address based on a global variable

**Please describe your use case for this integration and alternatives you've tried:**

I'm current using the Modbus integration to build a custom air quality sensor. The sensor runs on esp32-c3 and is equipped with Sen54 and radar ld2410. This works fine through Wi-Fi connection, but the idea is to use Wired connection as a Modbus server. Since I will have multiple of these sensors, instead of hardcoded the address for each one, I would like to have an address that can be store on flash memory and replaced via specific command

**Additional context**

Waveshare was a device (waveshare Modbus_RTU_IO_8CH) where is possible to change the Modbus address using a specific command:

[image]

nagyrobi, 2024-10-23:
This request translates into making the slave address templatable.

Alexandreao, 2024-11-04:
> This request translates into making the slave address templatable.

Can you help me to do this change?