esphome/feature-requests #2933: Pin as a dedicated convenience component?

mariusz-schimke, 2024-10-26:
**Describe the problem you have/What new integration you would like**
When you specify pin details of a component, sometimes the same pin needs to be shared with several components, in which case you set the `allow_other_uses` flag. But as far as I understand, you have to make sure the settings like `mode` or the advanced settings have to be the same in each component that shares the pin. This is prone to mistakes.

What would you say if pin could be a separate component with an id? This way you could define it once and just refer to by ID in the components that need to use it. The `inverted` flag, though, is something that should be overloadable then.

I don't think that I have a broad enough view on everything to be sure this makes sense and won't break anything, but it's just an idea to consider.

**Please describe your use case for this integration and alternatives you've tried:**
```yaml
binary_sensor:
  # normal mode
  - platform: gpio
    id: push_button_normal_mode
    pin:
      number: $push_button_input
      inverted: true
      mode: INPUT_PULLUP
      allow_other_uses: true
    filters:
      - lambda: |-
          // block in the authorized mode
          return id(authorized_mode_switch).state
            ? false
            : x;
    on_multi_click:
      # a set of different timings available in the normal mode of the button

  # authorized mode
  - platform: gpio
    id: push_button_authorized_mode
    pin:
      number: $push_button_input
      inverted: true
      mode: INPUT_PULLUP
      allow_other_uses: true
    filters:
      - lambda: |-
          // block if in normal mode
          return id(authorized_mode_switch).state
            ? x
            : false;
    on_multi_click:
      # a set of different timings available in the authorized mode of the button
```

**Additional context**
This could be expressed in a simpler way if `pin` was a component:

```yaml
pin:
  id: push_button_pin
  number: GPIO5
  inverted: true
  mode: INPUT_PULLUP
  allow_other_uses: true

binary_sensor:
  # normal mode
  - platform: gpio
    id: push_button_normal_mode
    pin:
      id: push_button_pin
    ...

  # authorized mode
  - platform: gpio
    id: push_button_authorized_mode
    pin:
      id: push_button_pin
    ...
```

ssieb, 2024-10-26:
No, you just shouldn't be doing that.  It's not necessary.  See the copy components.

mariusz-schimke, 2024-10-26:
Beautiful! Didn't know that! Many thanks!

closed 2024-10-26 by ssieb

mariusz-schimke, 2024-10-26:
However, there is another case I came across. I created a file that is a package to be shared between different ESP boards. One of its components is used only optionally, but has to have a pin assigned. Since in one of my cases I need to use all available pins of my board, I have to share one of the used pins with that unused component. I know there is the `!remove` keyword for packages, but I can't use it because the component is referred to by ID from lambdas, and it would cause them to be uncompilable.

So, I still see a case where sharing a pin could make sense. What do you think?

mariusz-schimke, 2024-10-26:
@ssieb, or a better example. My Nice gateway has an output that pulsates with different frequencies depending on whether the gate is opening or closing. Also, when it stays in the low state, it indicates the gate is closed. So to determine whether the gate is in motion or open/closed, I had to use two sensors assigned to the same pin:

```yaml
binary_sensor:
  - platform: gpio
    id: gate_open_pulsating_indicator
    pin:
      number: $gate_open_pulsating_indicator_input
      inverted: true
      mode: INPUT
      allow_other_uses: true

sensor:
  - platform: pulse_meter
    id: gate_in_motion_pulse_meter
    pin:
      number: $gate_open_pulsating_indicator_input
      inverted: true
      mode: INPUT
      allow_other_uses: true
    internal_filter: $gate_relay_debounce_period
    internal_filter_mode: EDGE
    timeout: 1500ms
    filters:
    ...
```

ssieb, 2024-10-26:
Your first case is highly likely to cause you issues.  Fix your configs to not be like that.
For the second one, you could probably do it with just the pulse meter on the pin, but that's what `allow_other_uses` is for.  If you want to copy the pin config, then use a yaml anchor.
Make the pin a component would be a large internal change if it was even possible.

mariusz-schimke, 2024-10-27:
Thanks a lot for the input, I appreciate it!

mariusz-schimke, 2024-10-27:
@ssieb, I completely forgot about anchors, even though I came across them in the past. They are a good solution, but just noticed I still have a tiny problem. One of my projects is a bit too large and I thought I'd just split groups of components into separate packages that are then composed together in the main file. Otherwise the could would be hardly maintainable as a single file.

When an anchor is defined in one package and referenced in another one, then even though in the end the output yaml is one complete file, an error is generated as soon as validation starts. I have to move both the components with the anchor to the same file for it to work.

I'm wondering if anchor validation could rather be done in the end, when the packages are merged together to form a complete output yaml file. Just letting you know there might be an issue. Of course, I can handle it, but I thought it could make sense to let you know.

Best regards!