esphome/feature-requests #2944: Add support for USB host mode CDC-ACM / OTG

felipecrs, 2024-11-13:
**Describe the problem you have/What new integration you would like**

Support for [USB host mode CDC/ACM](https://components.espressif.com/components/espressif/usb_host_cdc_acm) / OTG:

https://components.espressif.com/components/espressif/usb_host_cdc_acm

Allowing users to register an USB ACM device as a `uart` interface in ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**

Boards like the ESP32-S3 DevKit C comes two USB-C ports:

[image]

It would be great if cheap boards like these could be used to interface with inverters through USB. In my case, owner of a SRNE HF Series one, it would avoid a more convoluted solution through RS485:

[image]

And instead, I could just connect the ESP32-S3 to the inverter using an USB-B to USB-C cable/adapter. It would prevent a lot of soldering and lower the entry barrier for newcomers too.

**Additional context**

- https://components.espressif.com/components/espressif/usb_host_cdc_acm

- Looks like this project is one example of how USB host mode can be used in ESP devices:
   - https://github.com/ludoux/esp32-nut-server-usbhid

- Refs https://github.com/syssi/esphome-smg-ii/issues/79

clydebarrow, 2024-12-12:
Try this:

```
external_components:
  - source: github://clydebarrow/esphome@power-meter
    components: [modbus, usb_uart, usb_host]
    refresh: 1h

usb_uart:
  - type: CDC_ACM
    vid: 0x1234
    pid: 0x5678
    channels:
      id: usb_uart_id
      baud_rate: 9600
      buffer_size: 1024
      debug: true
      dummy_receiver: true
```

You'll need to set the vid/pid according to your device.

felipecrs, 2024-12-13:
This is **A M A Z I N G**. I will try it ASAP.

felipecrs, 2024-12-13:
I also had to change framework from arduino to esp-idf.

@clydebarrow, do you have any idea what is happening?

```
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=57371 count=1
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=57378 count=1
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=57860 count=2
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=57866 count=4
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=57871 count=2
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=57874 count=1
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=61485 count=2
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=61488 count=1
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=61501 count=1
[12:01:09][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=61503 count=1
```

(There is no other relevant log).

This is my shrunk config:

```yaml
external_components:
  - source: github://clydebarrow/esphome@power-meter
    components: [modbus, usb_uart, usb_host]
    refresh: 1h

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  # for USB UART
  framework:
    type: esp-idf

usb_uart:
  - type: CDC_ACM
    vid: 0x1a86
    pid: 0x7523
    channels:
      id: usb_uart_id
      baud_rate: 9600
      buffer_size: 1024
      debug: true

modbus:
  - id: modbus_0
    uart_id: usb_uart_id
    send_wait_time: 300ms

modbus_controller:
  - id: srne_0
    address: 0x01
    modbus_id: modbus_0
    command_throttle: 300ms
    update_interval: 15s
    
sensor:
  - platform: modbus_controller
    modbus_controller_id: srne_0
    name: "Inverter Battery Voltage"
    id: battery_voltage
    address: 0x101
    unit_of_measurement: "V"
    register_type: holding
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      - multiply: 0.1
    device_class: voltage
```

Do you think I got something wrong?

felipecrs, 2024-12-13:
Ops, hold on. I had log level set to INFO lol.

felipecrs, 2024-12-13:
Unfortunately, it doesn't work. I don't receive any UART logs at all. I can only see these `Duplicate modbus command found` logs and these:

```
[14:00:24][D][modbus_controller:040]: Modbus command to device=1 register=0xF030 no response received - removed from send queue
```

But maybe my configuration is wrong. @clydebarrow I'd appreciate if you can review it.

clydebarrow, 2024-12-13:
When you plug in the USB device you should at least see some log entries from that. You are using a dual serial port S2 or S3 board I hope - i.e. one for for flashing and logs (UART0) and one for the target device (USB.)

felipecrs, 2024-12-13:
Yes, I am a S3. And I'm plugging the USB into the USB port labeled as USB, while I'm using the second USB port labeled as COM to power the device.

[image]

This is my ESP device: https://aliexpress.com/item/1005005051294262.html

I confirmed my inverter USB, cable, and USB-A to C adapter is working by connecting it to a PC and running `lsusb`.

But let me try unplugging and plugging during operation to see if I get some logs.

felipecrs, 2024-12-13:
I tried a lot. It did not output anything at all. I suppose I'm having some hardware issue I don't understand yet.

I truly appreciate the work you have invested in it though, @clydebarrow. Thank you.

---

@syssi, I don't know if you have an ESP device capable of that, or an inverter with a USB port, but I'm pinging you just in case you are curious to make this work (https://github.com/syssi/esphome-smg-ii/issues/79) and wanted to give @clydebarrow's solution a try.

clydebarrow, 2024-12-13:
Did you set hardware_uart in logger to UART0? Does the log show that the components have been configured? You would expect to see something like this:

```
[07:40:38][C][usb_host:267]: USBClient
[07:40:38][C][usb_host:268]:   Vendor id 067B
[07:40:38][C][usb_host:269]:   Product id 2303
[07:40:38][C][usb_uart:169]:   UART Channel 0
[07:40:38][C][usb_uart:170]:     Baud Rate: 9600 baud
[07:40:38][C][usb_uart:171]:     Data Bits: 8
[07:40:38][C][usb_uart:172]:     Parity: NONE
[07:40:38][C][usb_uart:173]:     Stop bits: 0
[07:40:38][C][usb_uart:175]:     Debug: true
[07:40:38][C][usb_uart:177]:     Dummy receiver: true
```

Then when plugging in the USB device you should see at a minimum:

```
[07:41:33][D][usb_host:032]: New device 1
```

And if the device you are trying to talk to does genuinely have the VID and PID you put, that's a CH340, not a CDC_ACM device - use `type: ch340`.

Also try with a different USB device - you should at least see some logs.

felipecrs, 2024-12-13:
This is my logger configuration:

```yaml
logger:
  level: DEBUG
  baud_rate: 0
```

I am not connected to this ESP device via UART, I'm connected to it via web.

Do I need `dummy_receiver: true` and `buffer_size: 1024`? I guessed I did not and removed them from my config too.

> type: ch340

Thanks, that's good.

---

I'll confirm now if I have that section in my logs.

felipecrs, 2024-12-13:
```
[18:25:51][C][usb_host:267]: USBClient
[18:25:51][C][usb_host:268]:   Vendor id 1A86
[18:25:51][C][usb_host:269]:   Product id 7523
[18:25:51][C][usb_uart:169]:   UART Channel 0
[18:25:51][C][usb_uart:170]:     Baud Rate: 9600 baud
[18:25:51][C][usb_uart:171]:     Data Bits: 8
[18:25:51][C][usb_uart:172]:     Parity: NONE
[18:25:51][C][usb_uart:173]:     Stop bits: 0
[18:25:51][C][usb_uart:175]:     Debug: true
[18:25:51][C][usb_uart:177]:     Dummy receiver: true
```

But nothing else related to USB.

But I think I figured out what the issue is. There is no power in the USB output. I tried plugging my CH431 flasher and it didn't light up. That's weird, isn't it? And it's regardless of which port I try.

Maybe I need to power the ESP through the 5V pins, not through USB.

felipecrs, 2024-12-13:
There's this in my board's back:

[image]

I really wonder why there would be such solder points there...

felipecrs, 2024-12-13:
I soldered the two pins. Now there's light!

What a dumb dev board. Why not make it a detachable jumper at least.

felipecrs, 2024-12-13:
My device is in boot loop after flashing this config with this pin soldered. Without this config it works ok. I will continue debugging tomorrow.

syssi, 2024-12-14:
@clydebarrow Awesome job! I would like to see your feature at the ESPHome core some day! :-))

felipecrs, 2024-12-14:
I found what my issue with boot loop was. Basically, you must not flash esp-idf through UART. You must first flash arduino through UART and then you can flash esp-idf on top of it over OTA.

https://community.home-assistant.io/t/esp32-with-esp-idf-framework-running-into-a-boot-loop/567305/11

felipecrs, 2024-12-14:
Alright, USB is now working:

```
[18:56:40][C][usb_host:267]: USBClient
[18:56:40][C][usb_host:268]:   Vendor id 1A86
[18:56:40][C][usb_host:269]:   Product id 7523
[18:56:40][C][usb_uart:169]:   UART Channel 0
[18:56:40][C][usb_uart:170]:     Baud Rate: 9600 baud
[18:56:40][C][usb_uart:171]:     Data Bits: 8
[18:56:40][C][usb_uart:172]:     Parity: NONE
[18:56:40][C][usb_uart:173]:     Stop bits: 0
[18:56:40][C][usb_uart:175]:     Debug: true
```

```
[18:57:48][D][usb_host:032]: New device 1
[18:57:48][D][usb_host:068]: Open device 1
[18:57:48][D][usb_host:074]: Get descriptor device 1
[18:57:48][D][usb_host:080]: Device descriptor: vid 1A86 pid 7523
[18:57:48][D][usb_host:092]: Device connected: Manuf: (unknown); Prod: USB Serial; Serial: (unknown)
```

My config:

```yaml
esphome:
  platformio_options:
    # qio results into a boot loop when using esp-idf and flashing via USB
    # https://github.com/platformio/platform-espressif32/blob/268f561df4793d631d02e4ff5e67b6b11d1be4ec/boards/esp32-s3-devkitc-1.json#L16
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: esp-idf

# Using USB
external_components:
  - source: github://clydebarrow/esphome@power-meter
    components: [modbus, usb_uart, usb_host]
    refresh: 1h

usb_uart:
  - type: ch340
    vid: 0x1A86
    pid: 0x7523
    channels:
      id: usb_uart_id
      baud_rate: 9600
      debug: true

modbus:
  - id: modbus_0
    uart_id: usb_uart_id
    send_wait_time: 300ms

modbus_controller:
  - id: srne_0
    address: 0x01
    modbus_id: modbus_0
    command_throttle: 300ms
    update_interval: 15s

sensor:
  - platform: modbus_controller
    modbus_controller_id: srne_0
    id: grid_frequency
    name: "Inverter Grid Frequency"
    address: 0x215
    unit_of_measurement: "Hz"
    register_type: holding
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      - multiply: 0.01
    device_class: frequency
```

Unfortunately, I was not able to make it ready anything from my inverter.

I suppose I was supposed to see some UART in/out logs with `debug: true`, but there is nothing. Except:

```
[18:58:02][D][modbus_controller:040]: Modbus command to device=1 register=0x215 no response received - removed from send queue
```

ruteclrp, 2024-12-15:
Hi Felipecrs, you may wish to also look at the ESP32-S3 parts of this thread: https://github.com/esphome/feature-requests/issues/2982

It started with the RPI Pico card, but quickly turned to the S3 card. I think we are on a rather similar path.

/Robert

mcarbonneaux, 2025-01-02:
@clydebarrow it's possible to use [usb_host_hid](https://components.espressif.com/components/espressif/usb_host_hid) with esphome in the same maner ?

clydebarrow, 2025-01-02:
> it's possible to use [usb_host_hid](https://components.espressif.com/components/espressif/usb_host_hid) with esphome in the same manner ?

It would be possible, but it's not implemented at the moment in ESPHome.

mcarbonneaux, 2025-01-03:
the implementation of host hid while permit to control Host HID APC (the majority of the last APC on the marcket).
and also to read keyboard click or mouse click or mouvement.
and also eventualy emulate keyboard to remote control pc for example...

closed 2025-07-06 by clydebarrow

clydebarrow, 2025-07-08:
take it to Discord, #general_support channel.