esphome/feature-requests #2945: ESPHome ac-dimmer with MCP23S17 Component

NytoxRex0, 2024-11-15:
### The problem

The "_Gate_Pin_" from _AC_Dimmer_ component is not available to use for pins of the _MCP23S17_. This device should be more than fast enough for this application, please alter the code to allow this configuration.

### Which version of ESPHome has the issue?

2024.8.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.9.1

### What platform are you using?

ESP32

### Board

WROOM-S2

### Component causing the issue

ac-dimmer, MCP23S17

### Example YAML snippet

```yaml
spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13
  miso_pin: GPIO12

mcp23s17:
  - id: 'mcp23s17_hub'
    cs_pin: GPIO02
    deviceaddress: 0

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 
      mcp23xxx: mcp23s17_hub
      # Use pin number 0
      number: 0
      # One of INPUT or INPUT_PULLUP
      mode:
        output: true
      inverted: false
    zero_cross_pin:
      number: GPIO35
      mode:
        input: true
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: Dimmerized Light
```

ssieb, 2025-09-10:
It is not fast enough and can't be used, particularly not in interrupts.

closed 2025-09-10 by ssieb

diemon24, 2025-09-29:
Okay, I'm using this solution to get zero control with MOC3021 without hardware zero control 

[image]

`
mcp23017:
  - id: 'mcp23017_hub_01'
    address: 0x25

binary_sensor:
  - platform: gpio
    pin: 0
    name: "Null control"
    id: null_control
    internal: True
    on_press:
      then:
        - switch.turn_on: switch_control
    on_release: 
      then:
        - switch.turn_off: switch_control

switch:
  - platform: template
    id: switch_control
    name: "Switch null control"
    internal: true
    optimistic: true

  - platform: gpio
    name: "Output channel 1"
    id: out_channel_1
    pin:
      mcp23xxx: mcp23017_hub_01
      number: 8
      mode:
        output: true
      inverted: false
    internal: True
    restore_mode: ALWAYS_OFF

  - platform: template
    optimistic: true
    id: ch_light_1
    name: "Light channel 1"
    on_turn_on: 
      then:
        - switch.control:
            id: switch_control
            state: false
        - if:
            condition:
              - lambda: !lambda return id(switch_control).state == 0;
            then:
              - switch.turn_on: out_channel_1
    on_turn_off: 
      then:
        - switch.turn_off: out_channel_1

`