esphome/feature-requests #2950: Need support for HLK-LD2461

ajsrikanthaj, 2024-11-19:
**Describe the problem you have/What new integration you would like**
I have come across this radar sensor HLK-LD2461 and found to be very good on specs and it improves the already supported models such as LD2410, LD2420 and LD2450 etc.

**Please describe your use case for this integration and alternatives you've tried:**
Usecase here is to get a better and upgraded hardware for human presense detection.

**Additional context**
Device details:
https://hlktech.net/index.php?id=1185
https://drive.google.com/drive/folders/14_KgZpL4Th2LTRuq_W0X_ePjHu_Z6lcS

Chreece, 2024-11-20:
@ajsrikanthaj nice to find you here also :)
It's nice to have the sensor implemented but it actually isn't better than the other sensors. 
It's quite buggy: 
https://docs.screek.io/3a/ld2461-confusing-behavior-ghosting

ajsrikanthaj, 2024-11-20:
@Chreece good to see you too :)
I have seen the above article and I think it's quite old and there are no much evidance to prove it. On papers this sensor has a good numbers (comparing to previous models) and otehr previous models like LD2450 has a good results, hopoing to get similar quality from the manufacturer.I also seen somwehere that HiLink has done a firmware update. 
I wanted to give this sensor a shot with all the above points.

Chreece, 2024-11-20:
I have also tested the sensor with the tool from the [official folder](https://drive.google.com/drive/mobile/folders/14_KgZpL4Th2LTRuq_W0X_ePjHu_Z6lcS) that runs on windows and had similar ghosting experience. 
Can you point me to the firmware update? 
I can't find any official info about one, and the sensor has no bt connectivity to update firmware.
I had also big expectations from the data given by hlk but...