esphome/feature-requests #2961: Support for Pimoroni Badger 2040 W

Crocmagnon, 2024-11-27:
**Describe the problem you have/What new integration you would like**
Full support for Pimoroni Badger 2040 W, including driving the e-paper display.

**Please describe your use case for this integration and alternatives you've tried:**
Alternatives are custom code in Arduino/C or MicroPython. Pimoroni provides libraries in MicroPython and C++:
https://github.com/pimoroni/pimoroni-pico
https://github.com/pimoroni/badger2040

Looks like there's a driver for UC8151, which matches a chipset from adafruit used in a e-paper screen. Not sure it's the same though.
https://github.com/pimoroni/pimoroni-pico/tree/main/drivers/uc8151
https://www.adafruit.com/product/4262

**Additional context**
I tried the following config:
```yaml
esphome:
  name: badger2040-test
  friendly_name: badger2040-test

rp2040:
  board: rpipicow

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO19
  miso_pin: GPIO16

display:
  - platform: waveshare_epaper
    cs_pin: GPIO17
    dc_pin: GPIO20
    busy_pin: GPIO26
    reset_pin: GPIO21
    model: 2.90in-b
    update_interval: 1s
    lambda: |-
      it.line(0, 0, 50, 50);
```

The screen blinks continuously and shows the expected line, sometimes. When increasing the update_interval, like setting it to 10s, the line never appears but the screen blinks on every refresh.
I tried showing the test card but the screen doesn't change.

Models `2.90in-b` and `2.90in-bV3` are the only ones where the screen does something. For all other 2.90 variants, the screen remains in its original state when booting.

andywarburton, 2025-02-20:
I think the Badger is a great device and I have used it for Smart Home projects in the past before moving to ESPHome for all my devices. I would love to pull it out of retirement!

andywarburton, 2025-02-20:
I spent some time looking into things and it seems the display is the `gdew029t5 ` (same as the adafruit magtag!). I have the buttons working too based on some code I found in the esphome forum!

`
substitutions:
  clk_pin:   GPIO18
  mosi_pin:  GPIO19
  miso_pin:  GPIO16
  cs_pin:    GPIO17
  dc_pin:    GPIO20
  busy_pin:  GPIO26
  reset_pin: GPIO21
  button_a: GPIO12
  button_b: GPIO13
  button_c: GPIO14
  button_down: GPIO11
  button_up: GPIO15
  badger_led: GPIO22
  pico_led: GPIO32

esphome:
  name: badger-rp2040
  friendly_name: Badger RP2040

rp2040:
  board: rpipicow

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "ufkAYb7WCVvqHYe20deUknnTb8SSFUb2mSQ1X6y4Lng="

ota:
  - platform: esphome
    password: "9ec6880a365c267827789d294a5b9209"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

text_sensor:
  - platform: homeassistant
    name: "House state From Home Assistant"
    entity_id: sensor.house_state
    id: ha_house_state
    
spi:
  clk_pin: ${clk_pin}
  mosi_pin: ${mosi_pin}
  miso_pin: ${miso_pin}

display:
  - platform: waveshare_epaper
    cs_pin: ${cs_pin}
    dc_pin: ${dc_pin}
    busy_pin: ${busy_pin}
    reset_pin: ${reset_pin}
    model: gdew029t5
    rotation: 270
    update_interval: 5s
    lambda: |-
      it.line(0, 0, 100, 50);
      it.rectangle(5, 20, 30, 42);
      it.filled_rectangle(40, 40, 30, 42);
      it.circle(20, 40, 10);
      it.filled_circle(20, 75, 10);
      it.filled_ring(75, 75, 30, 20);
      it.filled_gauge(75, 75, 30, 20, 80);
      it.triangle(25, 5, 100, 5, 80, 25);
      it.filled_triangle(115, 5, 95, 25, 125, 70);
      it.filled_regular_polygon(170, 45, 20, EDGES_HEXAGON);
      it.regular_polygon(170, 45, 40, EDGES_OCTAGON, VARIATION_FLAT_TOP);
  

binary_sensor:
  - platform: status
    name: "Status"
    id: device_status

  - platform: gpio
    pin:
      number: ${button_a}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button a"
    on_press:
      then:
      - output.turn_on: badger_led
    on_release:
      then:
      - output.turn_off: badger_led
  - platform: gpio
    pin:
      number: ${button_b}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button b"  

  - platform: gpio
    pin:
      number: ${button_c}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button c"  

  - platform: gpio
    pin:
      number: ${button_up}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button up" 

  - platform: gpio
    pin:
      number: ${button_down}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button down"

output:
  - platform: rp2040_pwm
    pin: ${badger_led}
    id: badger_led
  - platform: gpio
    pin: ${pico_led}
    id: pico_led

light:
  - platform: monochromatic
    name: "Badger LED"
    output: badger_led
  - platform: binary
    name: "Pico LED"
    id: picow_led
    output: pico_led

sensor:
  - platform: internal_temperature
    name: "Internal Temperature"
    update_interval: 60s
    id: processor_internal_temperature

button:
  - platform: shutdown
    name: "Shutdown"
  - platform: restart
    name: "Restart"

`

andywarburton, 2025-02-20:
Adding the RTC is pretty easy too.

```
substitutions:
  i2c_sda: GPIO4
  i2c_scl: GPIO5

i2c:
  sda: ${i2c_sda}
  scl: ${i2c_scl}
  scan: true
  id: bus_a
```

and the rest just follows the [standard instructions for the pcf85063](https://esphome.io/components/time/pcf85063.html)

Crocmagnon, 2025-02-20:
Interesting, thanks! I've followed your config (mine below) and it kinda works but the screen takes ~30s to display an image. It very slowly blinks between black and white before displaying something. It looks like the refresh proceeds as usual but at a much lower speed.

<details>
<summary>ESPHome config</summary>

```yaml
esphome:
  name: badger2040-test
  friendly_name: badger2040-test
  on_boot:
    then:
      pcf85063.read_time:

substitutions:
  clk_pin:   GPIO18
  mosi_pin:  GPIO19
  miso_pin:  GPIO16
  cs_pin:    GPIO17
  dc_pin:    GPIO20
  busy_pin:  GPIO26
  reset_pin: GPIO21
  button_a: GPIO12
  button_b: GPIO13
  button_c: GPIO14
  button_down: GPIO11
  button_up: GPIO15
  badger_led: GPIO22
  pico_led: GPIO32
  i2c_sda: GPIO4
  i2c_scl: GPIO5

rp2040:
  board: rpipicow

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "key"

ota:
  - platform: esphome
    password: "pwd"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.99
    gateway: 192.168.0.254
    subnet: 255.255.255.0

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Badger2040-Test Fallback Hotspot"
    password: "pwd"

text_sensor:
  - platform: homeassistant
    name: "House state From Home Assistant"
    entity_id: input_select.house_present_away
    id: ha_house_state
    
spi:
  clk_pin: ${clk_pin}
  mosi_pin: ${mosi_pin}
  miso_pin: ${miso_pin}

i2c:
  sda: ${i2c_sda}
  scl: ${i2c_scl}
  scan: true
  id: bus_a

time:
  - platform: pcf85063
    # repeated synchronization is not necessary unless the external RTC
    # is much more accurate than the internal clock
    update_interval: never
  - platform: homeassistant
    # instead try to synchronize via network repeatedly ...
    on_time_sync:
      then:
        # ... and update the RTC when the synchronization was successful
        pcf85063.write_time:

display:
  - platform: waveshare_epaper
    cs_pin: ${cs_pin}
    dc_pin: ${dc_pin}
    busy_pin: ${busy_pin}
    reset_pin: ${reset_pin}
    model: gdew029t5
    rotation: 270
    update_interval: 5s
    lambda: |-
      it.line(0, 0, 100, 50);

binary_sensor:
  - platform: status
    name: "Status"
    id: device_status

  - platform: gpio
    pin:
      number: ${button_a}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button a"
    on_press:
      then:
      - output.turn_on: badger_led
    on_release:
      then:
      - output.turn_off: badger_led
  - platform: gpio
    pin:
      number: ${button_b}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button b"  

  - platform: gpio
    pin:
      number: ${button_c}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button c"  

  - platform: gpio
    pin:
      number: ${button_up}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button up" 

  - platform: gpio
    pin:
      number: ${button_down}
      inverted: false
      mode:
        input: true
        pulldown: true
    name: "Button down"

output:
  - platform: rp2040_pwm
    pin: ${badger_led}
    id: badger_led
  - platform: gpio
    pin: ${pico_led}
    id: pico_led

light:
  - platform: monochromatic
    name: "Badger LED"
    output: badger_led
  - platform: binary
    name: "Pico LED"
    id: picow_led
    output: pico_led

sensor:
  - platform: internal_temperature
    name: "Internal Temperature"
    update_interval: 60s
    id: processor_internal_temperature

button:
  - platform: shutdown
    name: "Shutdown"
  - platform: restart
    name: "Restart"
```
</details>

andywarburton, 2025-02-20:
With e-ink displays I don't typically refresh them more than once a minute at most, but from the moment of refresh to displaying the image it typically just takes a couple of seconds even when displaying multiple elements (at least on mine)

andywarburton, 2025-02-20:
I just saw your video and it definitely looks waaaaay slower than mine!

Crocmagnon, 2025-02-20:
I was compiling with an older version of esphome. Updating to 2025.2.0 seems to have improved the situation. Now with the partial refresh I don't see anything happening. I'll try to display something that changes, like time.

Crocmagnon, 2025-02-20:
Yeah looks like that was the issue. Partial refresh is nearly instant and invisible, and full refresh is very fast as well.
Thank you for your help! :)

andywarburton, 2025-02-20:
I'm so happy we managed to get this thing working - I've had a Badger sat in my drawer doing nothing for AGES!

andywarburton, 2025-02-20:
OOOOh... now I want to get the Inky Frame working!

joelis10, 2025-05-23:
yeah it's a non W, see my edit

andywarburton, 2025-05-23:
That’ll be your problem then. You need WiFi for ESPHome and the standard Badger doesn’t have it.