esphome/feature-requests #2967: Component required to replace a real NTC with a device operated with ESPHome

rhabacker, 2024-11-30:
**Describe the problem you have/What new integration you would like**

I need an esphome component that simulates an NTC resistor, i.e. it receives a temperature and returns the corresponding resistance value.

**Please describe your use case for this integration and alternatives you've tried:**

I have a Buderus heater that uses an NTC resistor to detect the outside temperature. Over time, it turned out that the resistors used were unreliable (I was forced to replace them twice). Since I already have a Zigbee-based outdoor temperature sensor connected to Homeassistant, there is a desire to replace the real NTC with an ESPHome-powered device. For this use case, I need the reverse functionality of the current NTC component. 

**Additional context**

ssieb, 2024-11-30:
This isn't really an esphome feature request since you need hardware.  There's at least one supported digital potentiometer, the x9c.

ssieb, 2024-11-30:
If you've already created the component, why are you making a feature request?

rhabacker, 2024-11-30:
I looked at the NTC sensor and a feature request was also created, which in my opinion is helpful for tracking and describing the use case.

rhabacker, 2024-11-30:
> since you need hardware

The proposed component is independent of the hardware, as it provides for the use of an output component via which the hardware is controlled.

closed 2025-05-01 by clydebarrow