esphome/feature-requests #2970: I hope to have an AC zero crossing trigger component

sorrypqa, 2024-12-01:
Describe the problem you have/What new integration you would like

I hope to have an AC zero crossing trigger component.LED lighting is very popular nowadays. It is a capacitive load, and smart switches are easily connected, causing relay contacts to stick together and unable to separate. We hope ESPHome can have a zero crossing trigger switch component, which uses one IO to detect the zero crossing signal provided by chips such as BL0942, BL0906, BL9010, and GS1102. Whether the switch component is turned on or off, it needs to wait for the zero crossing signal to arrive and undergo a custom delay before triggering the relay to turn on or off

Please describe your use case for this integration and alternatives you've tried:

At present, no alternative integration can be found. We have tried using binary sensors for zero crossing detection, which generates a large amount of garbage logs and can easily cause ESP32 reset

Additional context

nagyrobi, 2024-12-01:
I replaced the relay with an SSR, which by nature only switches at zero cross by itself.

sorrypqa, 2024-12-02:
> I replaced the relay with an SSR, which by nature only switches at zero cross by itself.

SSR cannot control high-power loads with a small volume

nagyrobi, 2024-12-02:
Try SSR-41FDA, it's good enough for 6A 250VAC. Can cover a serious amount of LED strips PSUs...

sorrypqa, 2024-12-02:
> Reference in
I cannot buy this model of SSR in my location