esphome/feature-requests #2972: Interrups with I2C

marcoatgit, 2024-12-04:
**Describe the problem you have/What new integration you would like**

Make it possible to use the interrupt pin of I2C I/O extensions.

**Please describe your use case for this integration and alternatives you've tried:**

Using a I2C I/O extension in polling mode, makes it slow. When using for instance an input for a counter or a rotary encoder knob, it will miss pulses. When the I2C I/O extensions can report the I/O change to the processor by interrupt, the I2C buss can be read and processed at the moment the change happens.

**Additional context**

Example: An rotary encoder for change settings will generate up to 10 pulse a second. Polling every 1 seconds will loss a lot of pulses.

ssieb, 2024-12-04:
How are you using an IO expander for a rotary encoder?  Where does the 1 second come from?

marcoatgit, 2024-12-04:
It is not specific only for a rotary encoder. A wind speed sensor with a puls output can also be used as example. It is for all inputs ware the puls rate is higher than the polling interval (update_interval)

The two outputs from the rotary encoder are connected to two inputs of a PCF8575 I2C I/O extender.

The 1 second comes from an example from the internet ware the update_interval was set to 10s, i reduced it already to 1 sec.

When connecting a wind sensor the speed can be even more then 10 time a second.

marcoatgit, 2024-12-04:
The interrupt is also useful when several extensions are connected with several signals connected to the inputs. In this case polling requires a lot of I2C actions. With the interrupt(s), only I2C actions are required when the input(s) changes.

ssieb, 2024-12-05:
How are you even using a rotary encoder on an expander.  This is why things like the `rotary_encoder` and `pulse_meter` components require a physical pin, not an expander.  I have no idea what you're actually doing, but you could try using a `binary_sensor`.  That samples at loop speed which is about 16ms.  I don't think you can access i2c from an interrupt.

marcoatgit, 2024-12-05:
Yes you are right, it not is possible to use the interrupt at the moment. That is why the feature request is made.

ssieb, 2024-12-05:
It's just not possible at all.

pilotak, 2024-12-26:
> It's just not possible at all.

Why not, binary sensor will have an interrupt pin config and will only read its state when the IRQ is made. I'm not saying this would be possible with rotary encode due to two pins sync, but as a counter/push button it would

chkr1011, 2025-01-26:
I am also interested in in this feature. I use PCF8574, PCA9555, MAX7311. All of these have a dedicated interrupt port.

When implementing a custom firmware with Platform IO I can check these interrupts in loop to avoid unnecessary polls. This is very important when reading the state of multiple devices (I have up to 6 devices on one bus). Polling a lot of devices produces a noticeable delay or leads to misses (as @ssieb already mentioned).

The code for the PCA9555 component has a  _loop_ method which always loads the pin states (https://github.com/esphome/esphome/blob/fc847c1de8927871e783db1a35e967efabda9f04/esphome/components/pca9554/pca9554.cpp#L40C1-L40C32).

In my opinion there should be a check of the interrupt first (if configured). And when the interrupt is not triggered, the read should be omitted.

ssieb, 2025-01-26:
> Why not, binary sensor will have an interrupt pin config and will only read its state when the IRQ is made.

You can't read the i2c from an interrupt context.

> In my opinion there should be a check of the interrupt first (if configured). And when the interrupt is not triggered, the read should be omitted.

This is possible, but probably not much benefit.  The i2c read is quite quick, although possibly if you have that many devices, it might help.

pilotak, 2025-01-27:
> You can't read the i2c from an interrupt context.

just make a flag and do it in main loop

ssieb, 2025-01-27:
Right, that's the second case that I replied to.  If someone wants to implement that, they could, but I don't see a lot of benefit.