esphome/feature-requests #2974: ESP32 - Sim800L DTMF Tones

DJGLTD, 2024-12-06:
**Describe the problem you have/What new integration you would like**

Add the function to the Sim800L to include DTMF tones so that incoming calls can complete more actions based of a number pressed when a call is in progress. 

**Please describe your use case for this integration and alternatives you've tried:**

No Support for DTMF tones to be received by esphome to then translate and work with them in automations or as triggers. 

**Additional context**

Sim800L has the option within the documentation to support this however i seem to be unable to add this manually or process this in a way that allows me to get this function to work.

glmnet, 2024-12-09:
indeed `AT+DDET` can be used for detecting DTMF

Kind of low priority when SMS already works, but 100% doable

DJGLTD, 2024-12-10:
I Agree low priority but I can't believe more people aren't asking for this, Like Alarm calling out when it's triggered, Obviously currently theres no way to acknowledge a event, you can inject a message into in my case the ESP32 with the onboard Sim800l to say for example, The alarm has been triggered in the garage press one to accept if that doesn't happen and the call disconnected it would know it got voicemail or you didn't hear it and do something else. My specific use case is for Gates, I have two gates and a garage door. I am wanting to have a dedicated quick call in my phone for each without having to pay for a sim card for each individual gate and then the garage door, I currently have two things, a GSM Intercom and a call to open relay, I've been using the app but it means stopping at the gate to open it. Just thought this would be a much simpler way because I can shout at my phone I also share my drive so I need access for other people without giving them the app. I'm just constantly thinking of different ways it could be useful